thesofproject/sof #873: README has to be updated for Kconfig

lyakh, 2019-01-23:
README.md has to be updated to reflect the new Kconfig-based build process.

jajanusz, 2019-01-23:
also doc/README* should be updated but i don't know which one should be there

@singalsu @lgirdwood 
Why there are README-HOST and README.host in doc?

lgirdwood, 2019-01-23:
@singalsu these two READMEs are exactly the same and can be removed in favor of similar detail in sof-docs.

lyakh, 2019-01-24:
Would be good to add rimage build instructions... I cannot seem to rebuild it...

lgirdwood, 2019-01-24:
@jajanusz not clear to me either (even when reading the build all scripts). Any clues ?

jajanusz, 2019-01-24:
@lgirdwood @lyakh 
You don't need this. It's built on demand wtih 'make bin', no effort and always up-to-date. Also now it will be valid for debug builds that require build number insterted by rimage (however need to add build number incrementing to cmake script).

jajanusz, 2019-01-24:
@lgirdwood @lyakh 
Moreover we think about doing it in Python instead of C, to make it more crossplatform and not build it at all.
Also there are some things that need to be done differently like 8 bytes offset for reading modules sections. It works only by accident and can change with different cores. We need to put some time into rewriting rimage (preferably in Python to have crossplatform tool).

lyakh, 2019-01-25:
This is what I had yesterday:
- I had a tree from before the Kconfig switch,  with built binaries in it.
- I git-updated to the new Kconfig-based tree
- Since I didn't have a README at that time, I ran "cmake -D... ." instead of creating a build directory, as you then described in the README
- I ran "make -j..." and "make bin" but the result wasn't correct
- I checked and I think rimage wasn't up to date. I went under "rimage" and ran "make" there, that updated rimage
- Still the build wasn't correct
- I ran "make clean" and then "make ..." again - that fixed the build
So, at the very least there were some problems with dependencies when rebuilding from an older tree, maybe also rimage wasn't rebuilt correctly, but it's also possible that it's just a one-off conversion problem and once converted to Kconfig it'll all work correctly from now on.

jajanusz, 2019-01-25:
It couldn't use system-wide rimage, because it uses specific path to just-built rimage in build dir. More likely you had some leftover from previous buildsystem (like for example config.h in src/include instead of <builddir>/generated/include), what caused wrong build of something.

jajanusz, 2019-01-26:
@lyakh Can I close it?

closed 2019-01-28 by lyakh