thesofproject/sof #874: Qemu boot on byt and cht will random fail

xiulipan, 2019-01-23:
Dig in from https://github.com/thesofproject/sof/pull/860
Test with master HEAD 

https://github.com/thesofproject/sof/commit/1f97d44a3b474664263583e6c0b1349afcf24b0c
It seems the agent has timeout of unknown reason.

CHT will 100% panic and BYT will random 10% panic in my local docker run.
[dump-byt-success.txt](https://github.com/thesofproject/sof/files/2787445/dump-byt-success.txt)
[dump-byt-fail.txt](https://github.com/thesofproject/sof/files/2787447/dump-byt-fail.txt)
[dump-cht.txt](https://github.com/thesofproject/sof/files/2787446/dump-cht.txt)
[qemu-bridge-mbox-io.hex.txt](https://github.com/thesofproject/sof/files/2787502/qemu-bridge-mbox-io.hex.txt)

Re triggered travis-ci and got a byt failed
https://travis-ci.org/thesofproject/sof/jobs/483369689

```bash
shim.io: write 0x40 =   0x0dead009 (old 0x0)    (233492489)     |....|
shim.io: write 0x44 =   0x80000800 (old 0x0)    (2147485696)    |....|
irq: send busy interrupt 0x80000800
bridge-io: msg send: 0 type 2 msg 64 size 16 ret 0
```

```bash
hexdump -C qemu-bridge-mbox-io
00000000  6c 00 00 00 00 00 00 70  00 00 00 00 00 00 00 00  |l......p........|
00000010  00 00 00 00 00 00 00 00  3c 00 00 00 01 00 01 00  |........<.......|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 31 66 39 37 64 00  |..........1f97d.|
00000040  00 00 00 03 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000050  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 a0 00 00 00  |................|
00000070  00 00 00 70 01 00 00 00  06 00 00 00 00 00 00 00  |...p............|
00000080  01 00 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |................|
00000090  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000a0  00 00 00 00 00 04 00 00  00 04 00 00 00 00 00 00  |................|
000000b0  03 00 00 00 00 00 00 00  00 00 00 00 00 01 00 00  |................|
000000c0  00 09 00 00 00 00 00 00  02 00 00 00 00 00 00 00  |................|
000000d0  00 00 00 00 80 03 00 00  00 0c 00 00 00 00 00 00  |................|
000000e0  04 00 00 00 00 00 00 00  00 00 00 00 00 02 00 00  |................|
000000f0  00 0a 00 00 00 00 00 00  06 00 00 00 00 00 00 00  |................|
00000100  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000800  58 00 00 00 00 00 00 00  00 00 00 00 25 03 06 00  |X...........%...|
00000810  f2 bd 2c ff 00 00 00 00  00 00 00 00 d0 0d 2d ff  |..,...........-.|
00000820  ba 25 2d ff 00 00 00 00  00 00 00 00 00 00 00 00  |.%-.............|
00000830  00 00 00 00 20 07 06 00  20 07 06 00 00 00 00 00  |.... ... .......|
00000840  00 00 00 00 00 00 00 00  48 0f 07 00 00 00 00 00  |........H.......|
00000850  08 00 00 00 c0 7d 32 ff  00 00 00 00 00 00 00 00  |.....}2.........|
00000860  2f 68 6f 6d 65 2f 73 6f  66 2f 77 6f 72 6b 2f 73  |/home/sof/work/s|
00000870  6f 66 2e 67 69 74 2f 73  72 63 2f 6c 69 62 2f 61  |of.git/src/lib/a|
00000880  67 65 6e 74 00 00 00 00  00 00 00 00 2f 68 6f 6d  |gent......../hom|
00000890  65 2f 73 6f 66 2f 77 6f  72 6b 2f 73 6f 66 2e 67  |e/sof/work/sof.g|
000008a0  69 74 2f 73 72 63 2f 6c  69 62 2f 61 67 65 6e 74  |it/src/lib/agent|
000008b0  2e 63 00 00 ff ff ff 00  f5 34 00 00 00 00 00 00  |.c.......4......|
000008c0  38 94 00 20 f0 22 2d ff  40 7e 32 ff 20 2c 30 ff  |8.. ."-.@~2. ,0.|
000008d0  c0 00 34 ff 98 69 30 ff  78 69 30 ff 10 00 00 00  |..4..i0.xi0.....|
000008e0  00 00 00 00 01 00 00 00  64 69 30 ff 64 69 30 ff  |........di0.di0.|
000008f0  b4 04 2d ff ba 25 2d 7f  60 7e 32 ff 00 00 00 00  |..-..%-.`~2.....|
00000900  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00001000
```

xiulipan, 2019-01-23:
@lgirdwood 
Tried each shm bridge file and found one with trace
```bash
./tools/logger/sof-logger -l build_cht_gcc/src/arch/xtensa/sof-cht.ldc -i /dev/shm/qemu-bridge-dram-mem
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      2           SA              1525.625000      1525.625000 /home/sof/work/sof.git/src/lib/agent.c:74        validate(), idle longer than timeout, delta = 4276244769
    0      2          MEM         223696213.958333 223694688.000000 /home/sof/work/sof.git/src/lib/alloc.c:802          count 536923620 free -1087586091
    0      2          DMA              1345.989583              nan /home/sof/work/sof.git/src/lib/dma.c:142
        dma_get() ID 16777215 sref = 27555 busy channels 0
    0      2          MEM              1420.052083        74.062500 /home/sof/work/sof.git/src/lib/alloc.c:815       heap: runtime status
```

It seems the work queue timer for the agent on byt and cht is broken.

lgirdwood, 2019-01-23:
@xiulipan can you try latest HEAD, CHT working for me, tried 5 times

xiulipan, 2019-01-23:
@lgirdwood 
I am ussing 1f97d44

> `00000030  00 00 00 00 00 00 00 00  00 00 31 66 39 37 64 00  |..........1f97d.|`

A simple workaround would solve the problem.
```diff
diff --git a/src/lib/agent.c b/src/lib/agent.c
index 5a31a649..58b995e5 100644
--- a/src/lib/agent.c
+++ b/src/lib/agent.c
@@ -93,7 +93,7 @@ void sa_init(struct sof *sof)
        trace_sa("sa_init(), sa->ticks = %u", sa->ticks);

        /* set lst idle time to now to give time for boot completion */
-       sa->last_idle = platform_timer_get(platform_timer) + sa->ticks;
+       //sa->last_idle = platform_timer_get(platform_timer) + sa->ticks;
        work_init(&sa->work, validate, sa, WORK_ASYNC);
        work_schedule_default(&sa->work, PLATFORM_IDLE_TIME);
 }
```
But I think the root cause of this bug is that the first agent `static uint64_t validate(void *data, uint64_t delay)` is coming earlier than we expected due to https://github.com/thesofproject/sof/pull/860/files#r250219820
Notice the argument `uint64_t timeout` is never used here. As the queue_calc_next_timeout 
```c
/* calculate next timeout */
static inline uint64_t queue_calc_next_timeout(struct work_queue *queue,
                                               uint64_t start)
{
        return queue->ticks_per_msec * queue->timeout / 1000 + start;
}
```
will always using `queue->timeout`, which is set to be 
```
       /* TODO: configurable through IPC */
        queue->timeout = PLATFORM_WORKQ_DEFAULT_TIMEOUT;
```
```c
void work_schedule(struct work_queue *queue, struct work *w, uint64_t timeout)
{
        struct work *work;
        struct list_item *wlist;
        uint32_t flags;

        spin_lock_irq(&queue->lock, flags);

        /* check to see if we are already scheduled ? */
        list_for_item(wlist, &queue->work) {
                work = container_of(wlist, struct work, list);

                /* keep original timeout */
                if (work == w)
                        goto out;
        }

        /* convert timeout micro seconds to CPU clock ticks */
        if (w->flags & WORK_SYNC)
                w->timeout =
                        queue_calc_next_timeout(queue, work_get_timer(queue));
        else
                w->timeout =
                        queue_calc_next_timeout(queue,
                                                work_shared_ctx->last_tick);

        /* insert work into list */
        list_item_prepend(&w->list, &queue->work);

        work_set_timer(queue);

out:
        spin_unlock_irq(&queue->lock, flags);
}
```

xiulipan, 2019-01-23:
@lgirdwood 
Can not reproduce after update with e5bc088e046862650f77876e60516aac097852b5
curious about why changes dw-dma will change the behavior or work queue.

tlauda, 2019-01-23:
@xiulipan @lgirdwood Fixed in #875. My mistake.

lgirdwood, 2019-01-23:
@xiulipan wont revert if it's only showing on Qemu only as it's best to wait for confirmation on real HW (as it could be qemu bug). Now merged #875 .

closed 2019-01-23 by lgirdwood