thesofproject/sof #890: Up2: 2 concurrent streams: xrun on one stream cause xrun on the other and DSP panic

plbossart, 2019-01-25:
To check the independence of concurrent streams, I start two terminals.
On Terminal 1, I run a basic aplay on the Hifiberry output.
On Terminal 2, I run the sof_xrun.sh script, which plays for one second on HDMI then artificially suspends for 1s. This creates an underrun by construction.

What is interesting is that the xrun on Terminal2 create xrun for Terminal 1, and eventually a DSP panic.

Terminal 1:
````
root@ubilinux4:~# aplay -Dhw:0,0 chirpl.wav 
Playing WAVE 'chirpl.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
underrun!!! (at least 0.070 ms long)
underrun!!! (at least 0.056 ms long)
underrun!!! (at least 0.057 ms long)
underrun!!! (at least 0.043 ms long)
underrun!!! (at least 0.049 ms long)
aplay: pcm_write:2011: write error: Input/output error
````

Terminal 2:
````

bash sof_xrun.sh 
[sof_xrun.sh.txt](https://github.com/thesofproject/sof/files/2794361/sof_xrun.sh.txt)

test 34
Write error: -5,Input/output error
xrun_recovery failed: -5,Input/output error
Transfer failed: Input/output error
test 35
test 36 << stops before end
````

[dmesg_xrun.log](https://github.com/thesofproject/sof/files/2794364/dmesg_xrun.log)
[logger.txt](https://github.com/thesofproject/sof/files/2794366/logger.txt)

kernel: topic/sof-dev
firmware: 8051fe393c88a3a541e102f83fd9135f6d3dca22
topology: sof-apl-pcm512x.tplg

filing as a firmware issue since no matter how bad the kernel is the DSP panic should not happen.

keyonjie, 2019-01-25:
good try, can you help try if my PRs #833 and #851 help?

wenqingfu, 2019-05-13:
@stevyan @mengdonglin to verify

ClarexZhou, 2019-05-24:
DSP panic then ipc timed out for   0x60050000: GLB_STREAM_MSG: TRIG_STOP  occurred on CFL laptop during S3 with playback test stress test in 4th time.

**Environment**
Kernel: 9453da1 https://github.com/lyakh/linux/tree/ipc-timeout (with PR895)
FW:b527fd7 https://github.com/lyakh/sof/tree/ipc-timeout (with PR1354)
Topology:Same as above FW repo, file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg
kconfig:
unset CONFIG_SND_SOC_SOF_DEBUG_ENABLE_FIRMWARE_TRACE
unset CONFIG_SND_SOC_SOF_DEBUG_ENABLE_DEBUGFS_CACHE

Also add previous 3th logs.
No error logger info. 4th dmesg as below:
```
[ 1227.392627] sof-audio-pci 0000:00:1f.3: periods:4
[ 1227.392660] sof-audio-pci 0000:00:1f.3: stream_tag 1
[ 1227.392664] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1227.392824] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1227.392826] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790528
[ 1227.392877] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[ 1227.392891] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1227.393020] done.
[ 1227.393046] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1227.393050] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[ 1227.393517] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[ 1227.393520] sof-audio-pci 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[ 1227.393525] sof-audio-pci 0000:00:1f.3: unknown ROM status value 00000048
[ 1227.393784] sof-audio-pci 0000:00:1f.3: error: runtime exception
[ 1227.393786] sof-audio-pci 0000:00:1f.3: error: trace point 00000081
[ 1227.393787] sof-audio-pci 0000:00:1f.3: error: panic at :0
[ 1227.393788] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[ 1227.393790] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[ 1227.393791] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x000723e0 PS       0x00060025 SAR     0x00000000
[ 1227.393793] sof-audio-pci 0000:00:1f.3: EPC1     0xbe04e9cf EPC2     0xbe05c54c EPC3     0xbe0596e4 EPC4    0x00000000
[ 1227.393794] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 1227.393795] sof-audio-pci 0000:00:1f.3: EPS2     0x00060720 EPS3     0x00060b22 EPS4     0x00000000 EPS5    0x00000000
[ 1227.393796] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[ 1227.393798] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe2ec9c0
[ 1227.393800] sof-audio-pci 0000:00:1f.3: 0xbe2ec9c0: be04e9cf 00400000 00000000 00000000
[ 1227.393801] sof-audio-pci 0000:00:1f.3: 0xbe2ec9c4: be075508 ffffffff 00000000 9e0a32cc
[ 1227.393803] sof-audio-pci 0000:00:1f.3: 0xbe2ec9c8: 41d6c100 c59dcd4d e5d7d828 ffffa401
[ 1227.393804] sof-audio-pci 0000:00:1f.3: 0xbe2ec9cc: e60df400 ffffa401 dfcb4c00 ffffa401
[ 1227.393806] sof-audio-pci 0000:00:1f.3: 0xbe2ec9d0: dfcb4c00 ffffa401 a58b70a2 ffffffff
[ 1227.393807] sof-audio-pci 0000:00:1f.3: 0xbe2ec9d4: 42823be8 ffffbffe e9788000 ffffa401
[ 1227.393809] sof-audio-pci 0000:00:1f.3: 0xbe2ec9d8: dfc69700 ffffa401 00000000 00000000
[ 1227.393810] sof-audio-pci 0000:00:1f.3: 0xbe2ec9dc: 00000000 00000000 00000000 00000000
[ 1227.463778] PM: suspend exit
[ 1229.111109] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[ 1229.111129] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1229.414829] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[ 1229.414846] sof-audio-pci 0000:00:1f.3: unknown ROM status value 00000048
[ 1229.415153] sof-audio-pci 0000:00:1f.3: error: runtime exception
[ 1229.415157] sof-audio-pci 0000:00:1f.3: error: trace point 00000081
[ 1229.415160] sof-audio-pci 0000:00:1f.3: error: panic at :0
[ 1229.415164] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[ 1229.415168] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[ 1229.415174] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x000723e0 PS       0x00060025 SAR     0x00000000
[ 1229.415178] sof-audio-pci 0000:00:1f.3: EPC1     0xbe04e9cf EPC2     0xbe05c54c EPC3     0xbe0596e4 EPC4    0x00000000
[ 1229.415182] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 1229.415186] sof-audio-pci 0000:00:1f.3: EPS2     0x00060720 EPS3     0x00060b22 EPS4     0x00000000 EPS5    0x00000000
[ 1229.415190] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[ 1229.415194] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe2ec9c0
[ 1229.415200] sof-audio-pci 0000:00:1f.3: 0xbe2ec9c0: be04e9cf 00400000 00000000 00000000
[ 1229.415206] sof-audio-pci 0000:00:1f.3: 0xbe2ec9c4: be075508 ffffffff 00
```
[test_3.log](https://github.com/thesofproject/sof/files/3215316/test_3.log)
[test_4.log](https://github.com/thesofproject/sof/files/3215317/test_4.log)

lgirdwood, 2019-05-27:
@ClarexZhou your comment is not related to this issue. Please assign your comment to correct issue or create a new one.

tlauda, 2019-06-06:
@plbossart Please reverify if the issue still exists.

jajanusz, 2019-06-17:
@plbossart @stevyan @mengdonglin Do you still see this issue?

Jiangxinx, 2019-07-22:
This issue can not be reproduced on daily test.
***Environment***
Firmware: a513ac72 Master
Kernel: 44c31df1 on topic/sof-dev
Topology: file: sof-apl-pcm512x.tplg
platform: apl-up2

closed 2019-07-22 by Jiangxinx