thesofproject/sof #919: Directories for firmware releases?

plbossart, 2019-01-30:
This is a joint Linux/firmware issue but filing as SOF issue to get feedback from the firmware team.

The current solution for Linux is to have all the binary .ri files in /lib/firmware/intel. This is fine for development and a limited set of products. There will be a point where we will have SOF releases, and multiple binaries provided for all platforms supported.

In Linux firmware circles, new releases are viewed as incremental, and the maintainers make use of symlinks. that means that ALL releases are part of the firmware tree.

see for example the KBL case with the closed-source firmware

````
lib/firmware/intel$ ls -l dsp_fw_kbl*
lrwxrwxrwx 1 root root     20 Dec 14 07:23 dsp_fw_kbl.bin -> dsp_fw_kbl_v3266.bin
-rw-r--r-- 1 root root 238920 Nov 17  2017 dsp_fw_kbl_v1037.bin
-rw-r--r-- 1 root root 238920 Dec 14 07:09 dsp_fw_kbl_v2042.bin
-rw-r--r-- 1 root root 243016 Dec 14 07:09 dsp_fw_kbl_v2630.bin
-rw-r--r-- 1 root root 243016 Dec 14 07:09 dsp_fw_kbl_v3266.bin
-rw-r--r-- 1 root root 238920 Nov 17  2017 dsp_fw_kbl_v701.bin
````

If we release firmware in one shot on multiple platforms, this will become a bit complicated. Maybe we should really add one level of indirection to the SOF firmware files, so that the kernel looks for  files in /lib/firmware/sof, and this sof directory can itself be a symlink to sof-1.x, sof-2.x, etc. There can be multiple ways of releases and symlinks can also be used at the file level if there are hot fixes.

we have a short window of time to change the location of firmware files, so asking @lgirdwood @lbetlej @michalgrodzicki @mmaka1 @wenqingfu to chime in. if we don't change the location then we will have to make sure all files can be identified with a release number and multiple symlinks updated at one, compare to one for a directory.

lgirdwood, 2019-01-30:
I'm fine for /lib/firmware/sof with symlinks for versioning binaries.

plbossart, 2019-01-30:
> I'm fine for /lib/firmware/sof with symlinks for versioning binaries.
>
That's an interesting one. My initial idea was to have 
/lib/firmware/intel/sof, and each contributor company/entity generates 
their own binaries.

If we have a single /lib/firmware/sof, there would be a need to 
coordinate binary generation across multiple companies. Not necessarily 
simple

And there might be some parts that aren't shared so that even on the 
same hardware you could have different binaries.

I would be more comfortable with /lib/firmware/<company>/sof.

dbaluta, 2019-01-30:
I'm in favor of /lib/firmware/<company/product>/sof. At NXP we use /lib/firmware/imx/dsp

plbossart, 2019-02-05:
No one from the firmware team chimed in so I guess there is no objection. I will file a patch this week to the alsa-devel mailing list.

closed 2019-02-05 by plbossart