thesofproject/sof #977: Deduplicate contents of platform headers

jajanusz, 2019-02-07:
There are a lot of common defines for cavs platforms that are duplicated in apollolake, cannonlake, icelake, suecreek.
I think it's better to move for example `<platform>/interupt.h`'s:
```c
/* IRQ numbers - wrt Tensilica DSP */
#define IRQ_NUM_SOFTWARE0	0	/* level 1 */
#define IRQ_NUM_TIMER1		1	/* level 1 */
#define IRQ_NUM_EXT_LEVEL1	2	/* level 1 */
#define IRQ_NUM_SOFTWARE1	3	/* level 1 */
```
to `cavs/interrupt.h` and in platform specific headers just `#include <cavs/interrupt.h>`.

jajanusz, 2019-02-07:
@mmaka1 @lgirdwood 
Please comment

lgirdwood, 2019-02-07:
Common files can be de-duplicated, but files with significant differences should remain in platform/ otherwise it get's too messy.

lbetlej, 2019-03-27:
done

closed 2019-03-27 by lbetlej