thesofproject/sof #984: CTX_SAVE ipc failure during runtime suspend

ranj063, 2019-02-08:
The CTX_SAVE ipc seems to fail always before the system enters runtime suspend:
kernel: https://github.com/ranj063/linux/tree/cros-sof-v4.19-ww05
fw: master ebf4f4b6bffe40fcc2c2bb22119492cf1758070a
topology: sof-cml-rt5682 (with PR https://github.com/thesofproject/sof/pull/976)

Will post sof-logger trace soon.

dmesg logs: 
[   41.384293] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000

[   42.739631] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000

[   42.740056] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x40010000 size 0xc

[   42.740094] sof-audio-pci 0000:00:1f.3: error: ctx_save ipc error during suspend -1

[   42.740164] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful

[   42.741247] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful

[   42.741269] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f

[   42.741455] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044

lgirdwood, 2019-02-08:
@lbetlej fyi, I assume this is calling the PM IPC reply assembly code ?

lbetlej, 2019-02-13:
@ranj063 the fw: master [ebf4f4b] includes PR#962 (@akloniex) and was verified by @emilchudzik , could you confirm that driver is not accessing memory window on CTX_SAVE reply ?

lgirdwood, 2019-02-13:
@ranj063 it's probably worth addig a BUG() macro in platform ops for  mailbox read/write() if DSP is in D3. This way we catch out any abusers.

ranj063, 2019-02-13:
@lbetlej @lgirdwood yes we are accessing the CTX_SAVE reply indeed. @bardliao will be covering for me for the CML release. Bard, could you please take a look at this?

ranj063, 2019-02-15:
@lgirdwood @lbetlej out of curiosity. Whats different in the D3 flow in CNL that causes in problem? I ask because we dont seem to have this problem on GLK

mengdonglin, 2019-02-25:
Kernel PR https://github.com/thesofproject/linux/pull/651/ is to fix this issue.

emilchudzik, 2019-02-25:
Agree. Verified on:
Platform: WHL RVP
kernel: cros-sof-v4.19 / [8492020d5967](https://github.com/thesofproject/linux/commit/8492020d5967) + PR#651 applied
FW: cml-002-drop-stable / [dcd825b](https://github.com/thesofproject/sof/commit/dcd825b923a680f63adde1b392638a14463ead63)
tplg for hda: sof-hda-generic.tplg

closed 2019-02-27 by plbossart