thesofproject/sof #998: Disassembly of rimage output

jajanusz, 2019-02-12:
In tools/README the is something like:
```
rimage can also convert kernel firmware formats to flat binaries formats to
assist in debugging :-

e.g. convert to flat binary, then ELF then dissasemble

rimage -i /lib/firmware/intel/reef-byt.ri -o image.bin -b -m byt

xtensa-byt-elf-objcopy -I binary -O elf32-xtensa-le -B xtensa image.bin image.bin.elf

xtensa-byt-elf-objdump -D image.bin.elf > image.dis.txt
```

This option are no longer in rimage - why?
Should we get this feature back, or maybe it's not needed?

I tried to disassembly like:
```
xtensa-apl-elf-objcopy -I binary -O elf32-xtensa-le -B xtensa src/arch/xtensa/sof-apl.ri.uns img.elf
xtensa-apl-elf-objdump -D img.elf > disassembly.txt
```
It looks okish, but I'm not sure it's 100% correct, rimage inserts manifest in the middle of file and it can make disassembler fail/produce wrong output in some parts of code, or am I wrong?

If reversing manifest insertion etc actions is needed for proper disassembly we should add this to rimage.

lgirdwood, 2019-02-13:
@jajanusz it's no longer needed and can be removed.

jajanusz, 2019-03-18:
This feature is no longer in rimage, neither in readme

closed 2019-03-18 by jajanusz