thesofproject/sof #1010: Merge CML and WHL with RT5682 topologies

ranj063, 2019-02-14:
the only difference is the SSP port num. Add a macro to differenciate which platform the tplg is being built for and choose the port accordingly.

bardliao, 2019-02-15:
I have an idea. Currently, sof_rt5682.c supports one SSP port only. So it is not necessary to identify which ssp port is used on the dai link name. So we can use the same topology for different platforms. ie. we can call it SSP-Codec in both SSP0 and SSP1 port.

plbossart, 2019-02-15:
On 2/15/19 12:53 AM, Bard Liao wrote:
>
> I have an idea. Currently, sof_rt5682.c supports one SSP port only. So 
> it is not necessary to identify which ssp port is used on the dai link 
> name. So we can use the same topology for different platforms. ie. we 
> can call it SSP-Codec in both SSP0 and SSP1 port.
>
It might work but you also need to change the cpu_dai_name, the 
following code will not be enough

if (is_legacy_cpu) {
         links[id].cpu_dai_name = devm_kasprintf(dev, GFP_KERNEL,
                             "ssp%d-port", ssp_port);
     } else {
         links[id].cpu_dai_name = devm_kasprintf(dev, GFP_KERNEL,
                             "SSP%d Pin", ssp_port);
     }

ranj063, 2019-02-15:
@bardliao this will work only when there's one SSP port in use. What happens if there are more than one?
and the ports are switched?

ranj063, 2019-02-15:
@bardliao thinking more about it. this will only fix the cpu dai name issue. WHat about the SSP index required in the dai_config. We'd still need to specify that?

bardliao, 2019-02-18:
@plbossart 
> It might work but you also need to change the cpu_dai_name

No, we can't change the cpu_dai_name, which should be the same as the name defined in the platform driver.

@ranj063 Yes, I assume there is only one SSP port used. The idea is to provide a simple and general use of machine driver and topology. We will still need different topology if there is more than one SSP port are used since we can't register two dai link with the same name.
> thinking more about it. this will only fix the cpu dai name issue. WHat about the SSP index required in the dai_config. We'd still need to specify that?

I was not trying to fix the cpu dai name issue, but the dai link name issue.
So, the only thing we have to do in the m4 file is to modify the name in the DAI_CONFIG().
For example, below is verified on byt.
```
DAI_CONFIG(SSP, 2, 0, SSP-Codec,
           SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
                      SSP_CLOCK(bclk, 2400000, codec_slave),
                      SSP_CLOCK(fsync, 48000, codec_slave),
                      SSP_TDM(2, 25, 3, 3),
                      SSP_CONFIG_DATA(SSP, 2, 24)))
```

plbossart, 2019-02-19:
On 2/17/19 9:32 PM, Bard Liao wrote:
>
> @plbossart <https://github.com/plbossart>
>
>     It might work but you also need to change the cpu_dai_name
>
> No, we can't change the cpu_dai_name, which should be the same as the 
> name defined in the platform driver.
>
It needs to be "ssp0-port", "ssp1-port", or "SSP0 Pin", which means the 
port has to be found with a quirk. You are already doing that but it'd 
need to be expanded so that you use SSP5 on Up2.

bardliao, 2019-02-20:
> It needs to be "ssp0-port", "ssp1-port", or "SSP0 Pin", which means the port has to be found with a quirk. You are already doing that but it'd need to be expanded so that you use SSP5 on Up2.

@plbossart Oh we may talk about different things :) I am not going to remove the quirk of ssp port. My idea is to set dai link name without port information but still keep it in the cpu_dai_name.
```
        /* SSP */
        links[id].name = devm_kasprintf(dev, GFP_KERNEL,
-                                       "SSP%d-Codec", ssp_port);
+                                       "SSP-Codec", ssp_port);
        links[id].id = id;
```

ranj063, 2019-02-28:
@plbossart @bardliao im wondering if I should go ahead and do this now or not. We will be updating the CML topology with the speaker pipeline soon at which point these two files will have to be separated. What do you think?

bardliao, 2019-02-28:
@ranj063 In my opinion, we can have one generic topology for development platforms such as RVPs. And customized topology for products. I also have a idea that maybe we can export all ssp ports in the machine driver and let developers decide what port do they use.

plbossart, 2019-02-28:
@ranj063 For CML, if you are referring to additional speaker amplifiers, they will have to be handled by a dedicated machine driver and the matching quirk to select it (as done for all Chromebooks). If there is only RT5682 connected we can share the same implementation across platforms.
@bardliao it's complicated to add all ports since we wouldn't know which ones are connected to external components such as codecs or amps.

bardliao, 2019-03-01:
@plbossart I know that is crazy. But I meant we use rt5682 in all ssp ports. The benefit is we can easily test all ssp ports as long as we connect it to a rt5682 codec. e.g. Currently, we have to modify the quirk table and rebuild the kernel if we want to test different ssp ports in the same RVP.
Also we don't have to specific a  ssp port in the quirk table if we export all of them.

plbossart, 2019-03-01:
On 3/1/19 12:29 AM, Bard Liao wrote:
> @plbossart <https://github.com/plbossart> I know that is crazy. But I
> meant we use rt5682 in all ssp ports. The benefit is we can easily
> test all ssp ports as long as we connect it to a rt5682 [...]
I doubt we are able to expose all SSPs on any board. There are pin-mux 
issues, e.g. the DMIC takes out 1 SSP, etc.

closed 2019-03-15 by ranj063