thesofproject/sof #1011: [CML] suspend fails at the second time

bardliao, 2019-02-15:
suspend_stress_test will fail at 2nd iteration.
kernel: topic/cros-sof-v4.19 branch  [8492020](https://github.com/thesofproject/linux/commit/8492020d596763defb4b627caa0c32d3347fc374)
fw: master https://github.com/thesofproject/sof/commit/84834803817632a0eed86fc1402cae83597e0c16
topology: sof-cml-rt5682 (with PR #976)

In fact, there are some error at the 1st resume:
```
[ 8221.557905] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000010 timedout
[ 8221.557920] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[ 8221.557932] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[ 8221.557960] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[ 8221.557983] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[ 8221.558688] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 8221.558701] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[ 8221.558719] sof-audio-pci 0000:00:1f.3: error: Error code=0xffffffff: FW status=0xffffffff
[ 8221.558739] sof-audio-pci 0000:00:1f.3: error: iteration 0 of Core En/ROM load failed: -62
[ 8221.558763] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[ 8221.558778] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[ 8221.558787] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[ 8221.558799] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[ 8221.559673] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[ 8221.560674] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010e0e successful
[ 8221.560770] sof-audio-pci 0000:00:1f.3: error: rx list empty but received 0x0
[ 8221.560784] sof-audio-pci 0000:00:1f.3: error: can't find message header 0x0
```

And similar error at the 2nd suspend, finally leads to suspend fail.
```
[ 8261.402649] sof-audio-pci 0000:00:1f.3: error: rx list empty but received 0x0
[ 8261.402667] sof-audio-pci 0000:00:1f.3: error: can't find message header 0x0
[ 8261.472783] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000010 timedout
[ 8261.472806] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[ 8261.472819] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[ 8261.472841] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[ 8261.472870] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[ 8261.472897] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 8261.472917] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[ 8261.472939] sof-audio-pci 0000:00:1f.3: error: Error code=0xffffffff: FW status=0xffffffff
[ 8261.472953] sof-audio-pci 0000:00:1f.3: error: iteration 0 of Core En/ROM load failed: -62
[ 8261.473552] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[ 8261.473577] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[ 8261.473592] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[ 8261.473614] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[ 8261.474544] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[ 8261.474554] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010e0e successful
[ 8261.474628] sof-audio-pci 0000:00:1f.3: error: rx list empty but received 0x0
[ 8261.474640] sof-audio-pci 0000:00:1f.3: error: can't find message header 0x0
[ 8261.508129] nvme 0000:01:00.0: pci_pm_suspend+0x0/0x1a1 returned 0 after 120266 usecs
[ 8261.544624] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000010 timedout
[ 8261.544646] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[ 8261.544658] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[ 8261.544676] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[ 8261.544690] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[ 8261.545503] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 8261.545513] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[ 8261.545544] sof-audio-pci 0000:00:1f.3: error: Error code=0xffffffff: FW status=0xffffffff
[ 8261.545561] sof-audio-pci 0000:00:1f.3: error: iteration 1 of Core En/ROM load failed: -62
[ 8261.545588] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[ 8261.545614] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[ 8261.545624] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[ 8261.545647] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[ 8261.546500] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[ 8261.546516] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010e0e successful
[ 8261.546580] sof-audio-pci 0000:00:1f.3: error: rx list empty but received 0x0
[ 8261.546592] sof-audio-pci 0000:00:1f.3: error: can't find message header 0x0
[ 8261.616459] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000010 timedout
[ 8261.616475] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[ 8261.616498] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[ 8261.616517] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[ 8261.616549] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[ 8261.616569] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 8261.616588] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[ 8261.616609] sof-audio-pci 0000:00:1f.3: error: Error code=0xffffffff: FW status=0xffffffff
[ 8261.616622] sof-audio-pci 0000:00:1f.3: error: iteration 2 of Core En/ROM load failed: -62
[ 8261.616638] sof-audio-pci 0000:00:1f.3: error: dsp init failed after 3 attempts with err: -62
[ 8261.616691] sof-audio-pci 0000:00:1f.3: unknown ROM status value ffffffff
[ 8261.616715] sof-audio-pci 0000:00:1f.3: error: status = 0xffffffff panic = 0xffffffff
[ 8261.616740] sof-audio-pci 0000:00:1f.3: unknown ROM status value ffffffff
[ 8261.616751] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[ 8261.616765] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware after resume -62
[ 8261.616862] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000
[ 8261.923451] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 0x4c
[ 8261.923474] sof-audio-pci 0000:00:1f.3: unknown ROM status value 01009701
[ 8261.923489] sof-audio-pci 0000:00:1f.3: error: status = 0x001d003c panic = 0x00000000
[ 8261.923506] sof-audio-pci 0000:00:1f.3: unknown ROM status value 01009701
[ 8261.923537] sof-audio-pci 0000:00:1f.3: error: ctx_save ipc error during suspend -110
[ 8261.973456] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1d003c timedout
[ 8261.973469] sof-audio-pci 0000:00:1f.3: error: reset enter failed: core_mask f adspcs 0x1d003c
[ 8261.973486] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask f
[ 8261.973500] sof-audio-pci 0000:00:1f.3: error: failed to power down core during suspend
[ 8261.973514] sof-audio-pci 0000:00:1f.3: error: suspending dsp
[ 8261.973531] sof-audio-pci 0000:00:1f.3: error: failed to power down DSP during suspend -5
```
The issue won't be solved by https://github.com/thesofproject/linux/pull/651 which will prevent reading reply of CTX_SAVE

[dmesg_1.txt](https://github.com/thesofproject/sof/files/2868023/dmesg_1.txt)
[dmesg_2.txt](https://github.com/thesofproject/sof/files/2868024/dmesg_2.txt)

mengdonglin, 2019-02-18:
@xun2z please help to reproduce this issue.

emilchudzik, 2019-02-18:
Hi,
I try to reproduce issue on WHL + I2S, but I have issue with entering suspend: https://github.com/thesofproject/sof/issues/1019

In spite of that FW and tplg loaded properly, aplay works.
kernel: topic/sof-dev: branch 69a41dab
FW: 2a86314
topology: sof-cnl-nocodec.tplg

On this kernel on dmesg there is no msg from CTX_SAVE/CTX_RESTORE. Is it excpected?

bardliao, 2019-02-19:
@emilchudzik 
> On this kernel on dmesg there is no msg from CTX_SAVE/CTX_RESTORE. Is it excpected?

You will see something like "ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE" if CONFIG_SND_SOC_SOF_DEBUG_VERBOSE_IPC is set in kernel config.
You can set it from make menuconfig
```
Device Drivers  ---> <*> Sound card support  ---><*>   Advanced Linux Sound Architecture  ---><M>   ALSA for SoC audio support  --->[*] SOF verbose IPC logs 
```

xun2z, 2019-02-19:
```bash
[   34.067417] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[   34.067933] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[   34.067981] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[   34.068464] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[   34.068509] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[   34.068968] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[   34.069015] sof-audio sof-audio: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE
[   34.069516] sof-audio sof-audio: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[   34.069542] sof-audio sof-audio: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[   34.069709] sof-audio sof-audio: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[   34.069806] serial 00:06: activated
[   34.218978] ata3: SATA link down (SStatus 4 SControl 300)
[   34.300774] atkbd serio0: Failed to deactivate keyboard on isa0060/serio0
[   34.349107] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[   34.349110] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
[   34.349547] acpi LNXPOWER:08: Turning OFF
[   34.350155] acpi LNXPOWER:07: Turning OFF
[   34.351819] OOM killer enabled.
[   34.351820] Restarting tasks ... done.
[   34.478003] video LNXVIDEO:00: Restoring backlight state
[   34.478013] PM: suspend exit
[   34.566699] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
[   34.566802] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   34.684789] atkbd serio0: Failed to enable keyboard on isa0060/serio0
[   35.066587] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
[   35.066668] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   35.124794] sof-audio sof-audio: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   35.428804] sof-audio sof-audio: error: ipc timed out for 0x40010000 size 0x4c
[   35.428818] sof-audio sof-audio: status: fw entered - code 00000005
[   35.429594] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[   35.429603] sof-audio sof-audio: error: ctx_save ipc error during suspend -110
[   35.429632] sof-audio sof-audio: FW Poll Status: reg=0x1010f0f successful
[   35.430658] sof-audio sof-audio: FW Poll Status: reg=0xf0f successful
[   35.430663] sof-audio sof-audio: DSP core(s) enabled? 0 : core_mask f
[   35.430714] sof-audio sof-audio: Debug PCIR: 00000000 at  00000044
[   35.430721] sof-audio sof-audio: Debug PCIW: 00000010 at  00000044
[   36.007936] e1000e: enp0s31f6 NIC Link is Up 100 Mbps Full Duplex, Flow Control: None
[   36.007945] e1000e 0000:00:1f.6 enp0s31f6: 10/100 speed: disabling TSO
```

Test env:
sof cml-001-drop-stable: 1f5a965711
kernel topic/cros-sof-v4.19: f69d61898a 
tplg: sof-cml-rt5682.tplg

Though ctx_save error occurs, whl will not be blocked.

[dmesg.log](https://github.com/thesofproject/sof/files/2878538/dmesg.log)

emilchudzik, 2019-02-19:
> @emilchudzik
>
>
> You will see something like "ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE" if CONFIG_SND_SOC_SOF_DEBUG_VERBOSE_IPC is set in kernel config.
> You can set it from make menuconfig
>
> ```
> Device [...]

I have  this option enabled.

emilchudzik, 2019-02-19:
Due to reboot issue [#662](https://github.com/thesofproject/linux/issues/662) on WHL RVP+I2S instead of suspend I use another WHL to find a repro.

On WHL RVP + HDA reworks I can playback on HDA (codec ALC700) and enter suspend/resume in loop. It works fine on each iteration.

kernel: cros-sof-v4.19 / [8492020d5967](https://github.com/thesofproject/linux/commit/8492020d5967)
FW: cml-002-drop-stable / dcd825b9
tplg for hda: sof-hda-generic.tplg

But I see in dmesg error with ctx:

> [ 5894.581642] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
> [ 5894.582030] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x40010000 size 0xc
> [ 5894.582044] sof-audio-pci 0000:00:1f.3: error: ctx_save ipc error during suspend -1

@bardliao could you check what happend in this function:
error: ipc error for 0x40010000 size 0xc

log from last iteration: playback/suspend/resume
[dmesg3.txt](https://github.com/thesofproject/sof/files/2880438/dmesg3.txt)

bardliao, 2019-02-20:
> @bardliao could you check what happend in this function:
error: ipc error for 0x40010000 size 0xc

@emilchudzik Please see #984 And it can be fixed by https://github.com/thesofproject/linux/pull/651

emilchudzik, 2019-02-20:
I have made check with patch thesofproject/linux#651. And playback and suspend works fine in loop. There is no error message with it.

kernel: cros-sof-v4.19 / [8492020d5967](https://github.com/thesofproject/linux/commit/8492020d5967)
FW: cml-002-drop-stable / [dcd825b](https://github.com/thesofproject/sof/commit/dcd825b923a680f63adde1b392638a14463ead63)
tplg for hda: sof-hda-generic.tplg

```
[  179.540605] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  179.540897] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  179.540928] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[  179.541952] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[  179.541958] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[  179.541997] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
--
[  187.684856] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  187.684864] sof-audio-pci 0000:00:1f.3: tplg: complete pipeline PIPELINE.1.HDA0.OUT id 3
[  187.684879] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  187.685016] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  187.685056] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  187.685206] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE

```

I also try to check as Bard ask - on the same platform using nocodec mode.
I prepare suitable kernel and topology. Then trying to go suspend cause OS freezing all the time. 
G3 on platform restore booting system again.

plbossart, 2019-02-20:
@emilchudzik can you please make sure that if you select the nocodec mode the HDAudio options are removed. There is a known issue, see https://github.com/thesofproject/linux/issues/512

emilchudzik, 2019-02-21:
@plbossart I have the same as in the linux#512.
Status is always "active" from:
`# cat /sys/devices/pci0000\:00/0000\:00\:1f.3/power/runtime_status`
on WHL with HDA support and with nocodec mode enabled:

```
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofnocodec [sof-nocodec], device 0: Port0 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 1: Port1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 2: Port2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

plbossart, 2019-02-21:
@emilchudzik can you unset CONFIG_SND_SOC_SOF_HDA_LINK and CONFIG_SND_SOC_SOF_HDA_AUDIO_CODEC from your configuration and rerun the test?

emilchudzik, 2019-02-25:
After unset this two options runtime suspend return "suspended" for a while after end playback:

```
audiodsp@lab201:~$ aplay -vv -Dplughw:0,0 SOF/doreimi-48k-s16le-stereo.wav
Playing WAVE 'SOF/doreimi-48k-s16le-stereo.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Plug PCM: Hardware PCM card 0 'sof-nocodec' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16320
  period_size  : 4032
  period_time  : 84000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4032
  period_event : 0
  start_threshold  : 16320
  stop_threshold   : 16320
  silence_threshold: 0
  silence_size : 0
  boundary     : 9187343239835811840
  appl_ptr     : 0
  hw_ptr       : 0
########     +                                     | 25%^C
Aborted by signal Interrupt...
#####        +                                     | 25%
audiodsp@lab201:~$ cat /sys/devices/pci0000\:00/0000\:00\:1f.3/power/runtime_status
active
audiodsp@lab201:~$ cat /sys/devices/pci0000\:00/0000\:00\:1f.3/power/runtime_status
active
audiodsp@lab201:~$ cat /sys/devices/pci0000\:00/0000\:00\:1f.3/power/runtime_status
active
audiodsp@lab201:~$ cat /sys/devices/pci0000\:00/0000\:00\:1f.3/power/runtime_status
suspended
audiodsp@lab201:~$ cat /sys/devices/pci0000\:00/0000\:00\:1f.3/power/runtime_status
suspended
audiodsp@lab201:~$ cat /sys/devices/pci0000\:00/0000\:00\:1f.3/power/runtime_status
suspended

```

ranj063, 2019-02-28:
@plbossart @emilchudzik this simple change in the kernel fixes the issue for me but it was simply a trial and error on my part. The value of 70ms timeout was based on the skylake driver and it worked in GLK.
I could use some feedback on how to determine the correct value to be set.

diff --git a/sound/soc/sof/intel/hda.h b/sound/soc/sof/intel/hda.h
index 54b817358e31..ee4817abc251 100644
--- a/sound/soc/sof/intel/hda.h
+++ b/sound/soc/sof/intel/hda.h
@@ -198,7 +198,7 @@
 #define HDA_DSP_RESET_TIMEOUT          50
 #define HDA_DSP_BASEFW_TIMEOUT         3000
 #define HDA_DSP_INIT_TIMEOUT           500
-#define HDA_ROM_INIT_TIMEOUT           70
+#define HDA_ROM_INIT_TIMEOUT           100
 #define HDA_DSP_CTRL_RESET_TIMEOUT             100
 #define HDA_DSP_WAIT_TIMEOUT           500     /* 500 msec */
 #define HDA_DSP_REG_POLL_INTERVAL_US           500     /* 0.5 msec */

ranj063, 2019-02-28:
@lbetlej @mmaka1 could you please comment on the value for the ROM init timeout value?

lbetlej, 2019-03-04:
I suggest at least 300 ms for HDA_ROM_INIT_TIMEOUT.

emilchudzik, 2019-03-05:
I have just checked flow from this issue topic.

Platform: Hatch
kernel: cros-sof-v4.19 / 8492020d5967 + thesofproject/linux#651.
FW: cml-002-drop-stable / dcd825b 
tplg: cml-002-drop-stable / dcd825b  [sof-cml-rt5682.zip](https://github.com/thesofproject/sof/files/2930066/sof-cml-rt5682.zip)

Commands "suspend_stress_test" in my opion return fail to suspend (see below screenshot).
When I use more than 2 iter (I mean paramneter: -c2) -> unable to enter suspend.
When I use -c1 several times -> platform trying to go suspend: screen is off, but ssh is still active. So it seems like script "suspend_stress_test" didn't enter platform to suspend even in first iterartion.

Is there any options to go suspend in natural way? e.g. pressing button on platform to double check if it is platform or script issue?
Short pressing "pwr" button try to power off.

```
localhost ~ # suspend_stress_test -c 1
Running 1 iterations with:
  suspend: 5-10 seconds
  wake: 5-10 seconds
  backup_rtc: false
  errors_fatal: true
  fw_errors_fatal: true
  bugs fatal:  true
  warnings fatal:  false
  crcs fatal:  true
  suspend command: powerd_dbus_suspend --delay=0
  wakeup source to ignore: none
  record_dmesg_dir:
  pm_print_times:  true
powerd start/running, process 3340
tlsdated stop/waiting
Suspend     1 of 1: sleep for  7 seconds...Suspend failed, s0ix count did not increment from 0)

powerd start/running, process 3647
tlsdated start/running, process 3674

Finished 1 iterations.
Suspend failures: 0
Wakealarm errors: 0
Firmware log errors: 0
s0ix errors: 1
```

emilchudzik, 2019-03-05:
I didn't see that thinks or WHL RVP. 
Here I use: rtcwake command or short pressing power button.
Both method properly entering suspend WHL RVP.

ranj063, 2019-03-06:
> I suggest at least 300 ms for HDA_ROM_INIT_TIMEOUT.

@lbetlej we used to have this value set to 300ms before. But the suspend stress test failed and after following the analysis done fro skylake, it was decided to use 70ms. I might leave it at 70 for GLK and only change it to 300ms for CML.

ranj063, 2019-03-08:
https://github.com/thesofproject/linux/pull/704 fixes this issue.
@emilchudzik could you please check if you can run a stress test with this change in the kernel

suspend_stress_test -c 500

lgirdwood, 2019-03-11:
@emilchudzik did https://github.com/thesofproject/linux/pull/704 fix it for you ? If so we should close.

emilchudzik, 2019-03-13:
on Hatch this patch doesn't help. I have still issue with first suspend, suspending doesn't work -> screen is off for a while, but logs return:
Suspend     1 of 1: sleep for  7 seconds...**Suspend failed**, s0ix count did not increment from 0)

emilchudzik, 2019-03-13:
I suggest to check repro on CML and if doesn't exists then closed it.
I can rise new issue that on Hatch (CML CHromebook) suspending doesn't work at whole using scripts:
suspend_stress_test

ranj063, 2019-03-13:
@emilchudzik I'm going to send you the latest coreboot that is needed for suspend stress test to work. Please give it a try

closed 2019-03-14 by ranj063

Tom-0505, 2020-09-23:
> @emilchudzik I'm going to send you the latest coreboot that is needed for suspend stress test to work. Please give it a try

@ranj063  can you send me too, I have met the same problem as @emilchudzik too.
THX

Tom-0505, 2020-09-23:
> on Hatch this patch doesn't help. I have still issue with first suspend, suspending doesn't work -> screen is off for a while, but logs return:
> Suspend 1 of 1: sleep for 7 seconds...**Suspend failed**, s0ix count did not increment from 0)

@emilchudzik Is this issue solved? I've met the same problem, and I've turned on HIPM,DIPM,DEVSLP, still can't solved the problem. Please kindly teach me how to do.
THX