thesofproject/sof #1020: platform_interrupt_{un,}mask() argument meaning

lyakh, 2019-02-18:
The second argument of platform_interrupt_{un,}mask() functions is "mask." It is ignored on cAVS and HSW, it's only used on BYT. There it is used as:
```
void platform_interrupt_mask(uint32_t irq, uint32_t mask)
{
	...
		shim_write(SHIM_PIMR, mask << BIT);
}
void platform_interrupt_unmask(uint32_t irq, uint32_t mask)
{
	...
		shim_write(SHIM_PIMR, shim_read(SHIM_PIMR) & ~(mask << BIT));
}
```
where BIT is a constant shift, depending on the irq. In some locations these functions are called with mask=1, which makes sense - a bit is cleared or set in a register. But in many cases both those functions are called with mask=0. Isn't that a bug?

RanderWang, 2019-02-19:
@lyakh  I only found mask = 0 in irq_disable_child which is not for BYT.
On BYT, HSW, BDW, there are only root IRQs or parent IRQs, no child IRQs.

lyakh, 2019-02-19:
@RanderWang hm, looks like you're right, all those cases are either for cAVS-specific drivers, or for SMP, or for child interrupts, e.i. none of them is relevant to BYT. But in any case, the only valid value for mask, where it is indeed used (on BYT), is 1, so, it's a constant, i.e. it isn't needed as a parameter. I'll prepare a patch to remove that parameter. Thanks. Then I'll also reference and close this issue.

closed 2019-03-11 by lgirdwood