thesofproject/sof #1040: DMA usage explanations

dbaluta, 2019-02-23:
Hi,

I hope this is the right place to place some questions about DMA as I am a little bit confused on how things are working  now in SOF FW and how should I implement the specific components for i.MX.

1. In src/audio/host.c there is this define CONFIG_DMA_GW. What is this exactly used for?

As far as I could understand there are two types of DMA controllers supported DW-DMA and HDA DMA. Is there any relation between these two types of controllers and CONFIG_DMA_GW?

2. Assuming that we have a simple scenario like this:

Host - Buffer - DAI

shouldn't we wait for data to reach DAI before calling ipc_stream_send_position in host_dma_cb?

It looks like this happened in the past but was removed with commit bfcf15ef41962 ("host: remove host side buffer check and fix for capture").

3. Looking at this simple playback pipeline:

    Host - Buffer1 - Volume - Buffer2 -  DAI -> codec

is it correct to say that:

There is one DMA channel that copies data from host CPU to buffer1 and another DMA channel
that copies from buffer 2 to DAI FIFOs?

4. It looks like complete field of host_data (src/audio/host.c) is not used anymore. And the call to:

       wait_completed(&hd->complete);

in host_dma_cb should be removed. Am I missing something?

tlauda, 2019-02-25:
@dbaluta 
1. CONFIG_DMA_GW means HD-DMA.
2. It takes position from DAI component, so it doesn't really matter. Anyway, it will only send the position after transferring specific number of bytes passed from kernel (host_period_bytes).
3. Yes.
5. Host component should be generic layer and should not care about your specific HW. Currently it's not 100% the case and that's why we can see defines and other stuff. You probably should implement new kind of driver rather than new implementation of the component.

lgirdwood, 2019-02-25:
@dbaluta as @tlauda says host.c is meant to be generic to any host DMA engine, but we do have some ifdefs that need to be removed. So your i.MX DMA wont need a full FW driver (I assume there is a host driver), but there should be a mechanism for the FW to get the DMA position from i.MX host.

dbaluta, 2019-02-25:
@tlauda Thanks for explanations! 

1,2, 3 - OK.
4 - sent a patch. thanks for review!

5 - Indeed host component should be generic, but it is DMA oriented and I'm struggling to image how it should work if the data can already be accessed from DSP side (it is in a shared SRAM).

What kind of new driver do you mean? A driver that implements a similar interface with DW DMA/HDA driver, but which instead of a DMA copy uses simple transfer (memcpy) operation?

In the following scenario:

Host - Buffer1 - Volume - Buffer2 -  DAI -> codec

data that host component should manage is directly accessible and we do not even need a copy to Buffer1 we should tell that buffer1 start address is the allocated by the Linux kernel.

We indeed use DMA between buffer2 and DAI which I think this maps perfectly with SOF FW has now.

dbaluta, 2019-02-25:
@lgirdwood Thanks for your answer.

Indeed makes sense that host.c should be generic to any host DMA engine. The question now is that should it be generic enough to allow non-DMA data access? Meaning that the data is already accessible by the Host component (no need to start copying it from the Linux host side to the DSP side).

By FW driver do you mean something similar with DW DMA/HDA DMA?
By host driver do you mean a Linux host driver?
By host here do you mean the core where the Linux runs on?

Let me have draw a small picture:

```
[Linux host CPU: allocates memory for audio buffer] <----> [DSP side: host component - buffer component - DAI component]
```
So, the Linux host allocates audio buffer  using snd_pcm_lib_malloc_pages and then it lets DSP know about the physical addresses of this pages using the compressed page table.

Then, on Intel platform  this pages are copied using DMA to DSP. Once the DSP knows the addresses of the pages, it uses again DMA to copy the content of the pages. Component host each time
sends a notification about the DMA pointer position to Linux side, which instead allows sending the next audio buffer data.

On, i.MX there is no need for DMA for Linux host - DSP FW transfer. The addresses of the pages and the content of the pages are directly accesbile through a SDRAM shared memory.

Now the question is how to make Host component (on the FW side) generic enough to handle this situation. On the easiest implementation (although not the optimal one) the Host component will implement simple copy operation (like for example memcpy).

So, Host component will copy data from SDRAM to whatever memory is allocated for component buffer and then send notification about "the DMA" pointer so that on Linux side the ALSA state machine can send next frames.

keyonjie, 2019-02-26:
@dbaluta The issue you met here could be a good motivation to refine our host component part, maybe we can spit it
host.c(platform independent part, e.g. only the component driver/ops common part kept ) + host_copier.c(type: dw_dma/hda_dma/mem_cy... the platform specific part put here as a private ops callback to generic host.c).

dbaluta, 2019-02-26:
@keyonjie sound like a good idea. will try to move into this direction.

lgirdwood, 2019-02-26:
@dbaluta  @keyonjie for the record I've created this https://github.com/thesofproject/sof/issues/1052

closed 2019-06-06 by tlauda