thesofproject/sof #1046: Compilation error when TRACEE = 0

dbaluta, 2019-02-25:
With the following change:

```
--- a/src/include/sof/trace.h
+++ b/src/include/sof/trace.h
-#define TRACEE 1
+#define TRACEE 0
```

compilation fails on baytrail for example:

```
[ 55%] Building C object CMakeFiles/sof.dir/src/audio/dai.c.o
/work/repos/sof/xsof/src/audio/dai.c: In function 'dai_buffer_process':
/work/repos/sof/xsof/src/audio/dai.c:67:2: error: implicit declaration of function 'trace_error_with_ids'; did you mean 'trace_dai_error_with_ids'? [-Werror=implicit-function-declaration]
  trace_error_with_ids(TRACE_CLASS_DAI,  \
  ^~~~~~~~~~~~~~~~~~~~
/work/repos/sof/xsof/src/audio/dai.c:111:4: note: in expansion of macro 'trace_dai_error_with_ids'
    trace_dai_error_with_ids(dev, "dai_buffer_process() "
    ^~~~~~~~~~~~~~~~~~~~~~~~
```

I have tried fixing it by adding an empty definition for _trace_error_with_ids but I get other errors instead.

If anyone care to have a look, otherwise I will look when have some time.

dragosht, 2019-02-27:
Also changing `#define TRACE 0` in src/include/sof/trace.h ends up with the following error on my apollolake build:

    [46%] Building C object CMakeFiles/sof.dir/src/arch/xtensa/task.c.o
    /caddy/sof-latest/sof/src/arch/xtensa/task.c: In function 'task_set_data':
    /caddy/sof-latest/sof/src/arch/xtensa/task.c:105:3: error: implicit declaration of function 'trace_error'; 
    did you mean 'trace_irq'? [-Werror=implicit-function-declaration]
    trace_error(TRACE_CLASS_IRQ,
    ^~~~~~~~~~~
    trace_irq

dbaluta, 2019-02-27:
@dragosht i have some patches prepared will send them asap.

closed 2019-03-01 by mmaka1