thesofproject/sof #1048: CLK:M/N divider support for BCLK

zhuyingjiang, 2019-02-26:
Currently the M/N divider for BCLK is not enabled, on some chip such as MAX98357 "BCLK must be 32, 48, or 64X of LRCLK", so BCLK must be 1.536/2.304/3.072M, and on WHL the MCLK is 24M, can not be dividable by BCLK. 
For platform to work, we need support M/N divider in the SSP config.

plbossart, 2019-02-26:
> Currently the M/N divider for BCLK is not enabled, on some chip such 
> as MAX98357 "BCLK must be 32, 48, or 64X of LRCLK", so BCLK must be 
> 1.536/2.304/3.072M, and on WHL the MCLK is 24M, can not be dividable 
> by BCLK.
> For platform to work, we need support M/N divider in the SSP config.
>
You've got to be very careful here. The M/N dividers do not provide a 
clock which has a 50% duty-cycle so enabling them can have really nasty 
side effects that we've seen in previous programs. Getting a good audio 
output when the M/N dividers are used depends to a large part on the 
clock recovery IP on the amplifier side. That's the main reason why we 
never enabled those M/N dividers. The alternative is an SRC conversion 
to 50kHz which as I understand it should already be functional.

sathyap-chrome, 2019-02-26:
i wanted to add 1 more info. on GLK platform, the Mclk / bclk is 12.5 ( MCLK 19.2 Mhz  and BCLK 1.536 Mhz) Can you help understand how that is possible. Can the same solution be moved to CML ?

plbossart, 2019-02-26:
> i wanted to add 1 more info. on GLK platform, the Mclk / bclk is 12.5 
> ( MCLK 19.2 Mhz and BCLK 1.536 Mhz) Can you help understand how that 
> is possible. Can the same solution be moved to CML ?
>
No, that's incorrect. on APL/GLK you can select independently the clock 
source for MCLK and BCLK.

For the amplifier we use 24.576 as source for the BCLK. The 19.2 MCLK is 
not connected to the amplifier so it's irrelevant.

For the codec we use the 19.2 MHz source for MCLK and BCLK.

We cannot use these settings on CML where neither 19.2 nor 24.576 MHz 
are provided to the audio IP.

#SSP 5 (ID: 0) with 19.2 MHz mclk with MCLK_ID 0 (unused), 1.536 MHz blck
DAI_CONFIG(SSP, 5, 0, SSP5-Codec,
     SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
         SSP_CLOCK(bclk, 1536000, codec_slave),
         SSP_CLOCK(fsync, 48000, codec_slave),
         SSP_TDM(2, 16, 3, 3),
         SSP_CONFIG_DATA(SSP, 5, 16, 0)))

#SSP 1 (ID: 1) with 19.2 MHz mclk with MCLK_ID 0, 1.92 MHz bclk
DAI_CONFIG(SSP, 1, 1, SSP1-Codec,
     SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
         SSP_CLOCK(bclk, 1920000, codec_slave),
         SSP_CLOCK(fsync, 48000, codec_slave),
         SSP_TDM(2, 20, 3, 3),
         SSP_CONFIG_DATA(SSP, 1, 16, 0)))

zhuyingjiang, 2019-02-27:
> The alternative is an SRC conversion to 50kHz which as I understand it should already be functional.

@plbossart Do you mean we set the DAI run at 50K Hz Frame sync, and use SRC convert from 48K to 50K?

@singalsu Does this workable? I check we have sof-apl-src-pcm512x.m4, but don't know if we can use SRC from 48K to 50K?
a DAI config like following?
DAI_CONFIG(SSP, 1, 0, SSP1-Codec,
	SSP_CONFIG(I2S, SSP_CLOCK(mclk, 24000000, codec_mclk_in),
		SSP_CLOCK(bclk, 2400000, codec_slave),
		SSP_CLOCK(fsync, 50000, codec_slave),
		SSP_TDM(2, 24, 3, 3),
		SSP_CONFIG_DATA(SSP, 1, 24))

juimonen, 2019-02-27:
@zhuyingjiang, please check https://github.com/thesofproject/sof/pull/842. There's a topology for 50kHz pipeline (we've tried it with this PR and it works fine). I'm not sure will it work without the other stuff in the PR. The PR hasn't been merged, because it is quite intrusive and @tlauda is working on a pipeline refinement/scheduling PR which is overlapping with this PR heavily.

singalsu, 2019-02-27:
@zhuyingjiang Yes, SRC supports the 48 kHz to 50 kHz conversion.

zhuyingjiang, 2019-02-28:
> @zhuyingjiang, please check #842. There's a topology for 50kHz pipeline (we've tried it with this PR and it works fine). I'm not sure will it work without the other stuff in the PR. The PR hasn't [...]

@juimonen with Pierre's suggestion we can config a SRC pipeline, and I have tested on rt5682 with codec SRC disabled. The only change is in tplg.

zhuyingjiang, 2019-03-01:
We use the SRC solution for this now, track in #1063.
This enhancement issue will be closed for now, and will be reopen if the FW team resource is OK.

closed 2019-03-01 by zhuyingjiang