thesofproject/sof #1050: [Stress test] IPC times out when running multiple pipelines

keqiaozhang, 2019-02-26:
This is more like a regression. We met this issue before, but fixed on glk branch.

How to reproduce:
1. run test scripts under tools/test/qa/tests/stability_test/:
   ./multiplex_pipeline-test.sh pcm0p pcm1p
2. or use audiofuntest tool, test script as below:
```
#!/bin/bash

int=0
while [ $int -lt 100000 ]
do
        echo "Test $int"
        audiofuntest -P "aplay -D hw:0,0 -r 48000 -f s16 -t raw -c 2 -B 0 -" -R "arecord -D hw:0,99 -r 48000 -f s16 -t raw -c 4 -B 0 -" -r 48000 -T 5 -a 0 -m "0,1" -c 4 -g 100
        if [ $? != 0 ]; then
                echo "test failed at $int."
                exit 1
        fi
        sleep 0.2
        echo $int
        let int++
done
```
error message：
aplay: pcm_write:2051: write error: Input/output error

dmesg:
```
[  303.909613] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 99 dir 1 cmd 0
[  303.909678] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000
[  304.212216] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60050000 size 0xc
[  304.212267] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  304.212439] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  304.212444] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  304.212477]  DMIC01: ASoC: trigger FE failed -110
[  304.212527] sof-audio-pci 0000:00:0e.0: pcm: free stream 99 dir 1
[  304.516128] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60030000 size 0xc
[  304.516158] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  304.516328] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  304.516334] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  304.517561] sof-audio-pci 0000:00:0e.0: pcm: close stream 99 dir 1
[  305.131853] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 0
[  305.435650] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60050000 size 0xc
[  305.435681] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  305.435850] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  305.435856] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  305.435889]  Speakers: ASoC: trigger FE failed -110
[  305.435941] sof-audio-pci 0000:00:0e.0: pcm: free stream 0 dir 0
[  305.739373] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60030000 size 0xc
[  305.739401] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  305.739591] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  305.739594] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  305.751792] sof-audio-pci 0000:00:0e.0: pcm: close stream 0 dir 0
[  308.058380] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x40010000 size 0x4c
[  308.058431] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  308.058606] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[  308.058632] sof-audio-pci 0000:00:0e.0: error: ctx_save ipc error during suspend -110
[  308.058692] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
[  308.059772] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[  308.059791] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 3
[  308.059827] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000010 at  00000044
```
Trace: 
No obverses trace error, trace log is attached.
 
Test recipe:
sof-master: 82d29e099cd
kernel: topic/cros-sof-v4.14-rebase-b61733c

[dmesg.log](https://github.com/thesofproject/sof/files/2904488/dmesg.log)
[logger-t.log](https://github.com/thesofproject/sof/files/2904490/logger-t.log)

Jiangxinx, 2019-02-26:
his issue can also be reproduced on UP2 with PCM512x.
No dmesg error trace, but sof-logger have error trace.

sof-logger:
```
test@cd-apl-up2-pcm512x-02:~$ sudo sof-logger -l /etc/sof/sof-apl.ldc
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          DMA         109193447.343750 109193448.000000 intel/cavs/hda-dma.c:346    hda-dmac: 5 preload timeout
    0      1         HOST         109193453.802083         6.458333     src/audio/host.c:726    host_copy_int() error: dma_set_config() or dma_start() failed, ret = 4294967234
    0      1          IPC         109193459.791667         5.989583 src/audio/pipeline.c:623    pipeline_copy() error: ret = -62, dev->comp.id = 16, dir = 1
    0      1         PIPE 3.17    109193466.145833         6.354167 src/audio/pipeline.c:739    pipeline_xrun_recover()
    0      1         COMP         109193478.281250        12.135417 rc/audio/component.c:192    comp_set_state() error: wrong state = 5, COMP_TRIGGER_PREPARE
    0      1         PIPE         109193483.802083         5.520833 src/audio/pipeline.c:350    pipeline_prepare() error: ret = -22,dev->comp.id = 12
    0      1         PIPE 3.17    109193489.687500         5.885417 src/audio/pipeline.c:746    pipeline_xrun_recover() error: pipeline_prepare() failed, ret = -22
    0      1         PIPE 3.17    109193495.000000         5.312500 src/audio/pipeline.c:821    pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1          DAI 3.16    109195775.416667      2280.416748      src/audio/dai.c:114    dai_buffer_process() error: Insufficient bytes for next period. comp_underrun()
```

step:
1. ./multiplex_pipeline-test.sh pcm0p pcm2p

Test env:
sof master: 82d29e09
kernel sof-dev: ee19d24e
tplg: sof-apl-pcm512x.tplg

wwittbrx, 2019-02-26:
I think this commit 3dbd1fb59 reintroduced this, please check. @slawblauciak may have more info on this.

keyonjie, 2019-02-26:
@RanderWang here the BF bit not set error we met:

`
    0      1          DMA         109193447.343750 109193448.000000 intel/cavs/hda-dma.c:346    hda-dmac: 5 preload timeout
`

plbossart, 2019-02-26:
Can we please make sure this test is run by the QA team on a regular basis? Between broken scripts and regressions something is not quite right here.

keqiaozhang, 2019-02-27:
@plbossart 
We will add this test to our daily test basis and @stevyan is working on the broken scripts under tools/test/qa/tests/.

lgirdwood, 2019-03-11:
@keqiaozhang there have been a few recent component and pipeline updates. Can you re-check.

keqiaozhang, 2019-03-12:
@lgirdwood 
This issue still can be reproduced with latest code base.  Error trace logs are same.
```
    0      1          DMA          41334491.614583       224.479172 intel/cavs/hda-dma.c:344    hda-dmac: 5 preload timeout
    0      1         HOST          41334497.812500         6.197917     src/audio/host.c:766    host_copy() error: dma_set_config() or dma_start() failed, ret = 4294967234
    0      1         PIPE          41334503.593750         5.781250 src/audio/pipeline.c:658    pipeline_copy() error: ret = -62, start->comp.id = 5, dir = 1
    0      1         PIPE 1.6      41334510.208333         6.614583 src/audio/pipeline.c:766    pipeline_xrun_recover()
    0      2         PIPE 1.6      41334514.843750         4.635417 src/audio/pipeline.c:345    pipeline_prepare()
    0      2         HOST          41334518.697917         3.854167     src/audio/host.c:617    host_prepare()
    0      1         COMP          41334522.395833         3.697917 rc/audio/component.c:192    comp_set_state() error: wrong state = 5, COMP_TRIGGER_PREPARE
    0      1         PIPE          41334528.177083         5.781250 src/audio/pipeline.c:352    pipeline_prepare() error: ret = -22,dev->comp.id = 1
    0      1         PIPE 1.6      41334533.906250         5.729167 src/audio/pipeline.c:773    pipeline_xrun_recover() error: pipeline_prepare() failed, ret = -22
    0      1         PIPE 1.6      41334538.958333         5.052083 src/audio/pipeline.c:842    pipeline_task(): xrun recover failed! pipeline will be stopped!
```
Test recipe:
sof master: 2dbe69
kernel cros-sof-v4.14-rebase: b178744d

ranj063, 2019-03-12:
@keqiaozhang would it be possible to try this with linux PR#695?

lgirdwood, 2019-03-12:
@dabekjakub looks like a HDA DMA timeout is being cascaded into other timeouts.

lgirdwood, 2019-03-12:
@slawblauciak looks like @RanderWang is also working on this within kernel too. https://github.com/thesofproject/linux/pull/705. Can you guys align as this may need kernel and FW fix. Thanks

keqiaozhang, 2019-03-12:
@ranj063 
There're some conflicts when applying PR#695 to cros-sof-v4.14-rebase.
So I applied this PR to sof-dev branch and tested this issue on GLK chromebook with UbuntuOS. 
This PR can reduce the probability of this issue but can't not be completely fixed. Before the probability  is about 15%, now is reduce to 1%.

ranj063, 2019-03-12:
@keqiaozhang thanks for checking

RanderWang, 2019-03-20:
fixed by ASoC:sof:hda: fix hda preload issue #705

closed 2019-03-20 by RanderWang