thesofproject/sof #1052: host.c: remove DMA specific logic

lgirdwood, 2019-02-26:
host.c has host HDA GW and DW DMA specific logic surrounded by ifdefs. This logic should be abstracted and pushed down into the DMA drivers (using new DMA ops). host.c should also support a buffer interface direct to host memories .

dbaluta, 2019-03-04:
Hi @mmaka1,

Any idea in commit b39b17ddf3e ("comp: host: hda/dw-dma flow unification") why host and local sgl are no longer used for HDA DMA?

tlauda, 2019-04-12:
@dbaluta When using HDA DMA for host transfers we don't have access to the host buffer from DSP side, so we don't need host sgl. Local sgl is just the pointer to the config sgl.

dbaluta, 2019-04-18:
@tlauda if we don't have acess to host buffer from DSP how do we copy the data?

lgirdwood, 2019-04-18:
@dbaluta we only have access to the host buffer after copying to DSP with HDA DMA. The Intel HDA DMA is programmed by both FW and driver; driver programs host address, FW programs DSP address, I dont remember who enables the copy though.

dbaluta, 2019-04-18:
@lgirdwood thanks, make sense.

dbaluta, 2019-04-18:
> host.c has host HDA GW and DW DMA specific logic surrounded by ifdefs. This logic should be abstracted and pushed down into the DMA drivers (using new DMA ops). host.c should also support a buffer interface direct to host memories .

Here I am working on creating a new object host_copier which will expose mostly the same operations as host object (params, trigger, copy, prepare, etc). But I still cannot wrap my mind around on how to split the functionality. 

For the moment it makes sense that 
* dma_buffer member of host_data to stay in host.c
* dma pointer to be pushed into specific host_copier instance
* dma_sg_config config memober of host_data it should be pushed into specific host_copier.
... together with host and local members of host_data.

I've thought about pushing the specific ops as you said in DMA drivers but I think it makes more sense to create a special object host_copier which will contain an instance to a dma channel (if we use DMA).

lgirdwood, 2019-04-19:
@dbaluta yes, I think special object is probably the way to go atm. I guess your DSP can memcpy() directly from host -> DSP and schedule this based on the pipeline copy() callback. Another alternative is a dummy DMA driver that does just memcpy() ?

dbaluta, 2019-04-21:
@lgirdwood yes, I can directly memcpy from host -> DSP. For the moment I'm going with the dummy DMA driver that just does memcpy().

lgirdwood, 2019-06-07:
@michalgrodzicki @mengdonglin lets make this P1 for v1.4 as it cleans up our DMA logic a lot. @tlauda has already made a start, but lets track with this feature.

PR #1565 (2019-06-25): host: unify component for all DMA types
Unifies host component for all DMA types.
We don't need to branch on CONFIG_DMA_GW anymore.

Closes #1052.
closed 2019-06-25 by jajanusz