thesofproject/sof #1054: Check aplay function failed on daily test.

Jiangxinx, 2019-02-27:
Summary:
aplay failed with 'hw params uninstall (-19)' reason.

Step:
1. aplay -Dhw:0,0 -f s16_le -c 2 -r 48000 /dev/zero

Error output:
```
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 84000
PERIOD_SIZE: 4032
PERIOD_BYTES: 16128
PERIODS: (4 5)
BUFFER_TIME: 340000
BUFFER_SIZE: 16320
BUFFER_BYTES: 65280
TICK_TIME: 0
[INFO]: aplay FAILED via PCM: Port5
```

Dmesg:
```
[  136.056814 <    0.003133>] sof-audio-pci 0000:00:0e.0: pcm: open stream 0 dir 0
[  136.056819 <    0.000005>] sof-audio-pci 0000:00:0e.0: period min 192 max 16384 bytes
[  136.056822 <    0.000003>] sof-audio-pci 0000:00:0e.0: period count 2 max 16
[  136.056824 <    0.000002>] sof-audio-pci 0000:00:0e.0: buffer max 65536 bytes
[  136.057251 <    0.000427>] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[  136.057254 <    0.000003>] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[  136.057258 <    0.000004>] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[  136.057260 <    0.000002>] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[  136.057270 <    0.000010>] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[  136.057271 <    0.000001>] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[  136.057276 <    0.000005>] sof-audio-pci 0000:00:0e.0: pcm: hw params stream 0 dir 0
[  136.057280 <    0.000004>] sof-audio-pci 0000:00:0e.0: generating page table for 000000005e1bbd3c size 0xff00 pages 16
[  136.057301 <    0.000021>] sof-audio-pci 0000:00:0e.0: period_bytes:0xfc0
[  136.057303 <    0.000002>] sof-audio-pci 0000:00:0e.0: periods:16
[  136.057311 <    0.000008>] sof-audio-pci 0000:00:0e.0: stream_tag 1
[  136.057341 <    0.000030>] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  136.057497 <    0.000156>] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60010000 size 0x14
[  136.057509 <    0.000012>] sof-audio-pci 0000:00:0e.0: error: hw params ipc failed for stream 1
[  136.057513 <    0.000004>] sof-audio-pci 0000:00:0e.0: ASoC: 0000:00:0e.0 hw params failed: -19
[  136.057517 <    0.000004>]  Port5: ASoC: hw_params FE failed -19
[  136.060418 <    0.002901>] sof-audio-pci 0000:00:0e.0: pcm: free stream 0 dir 0
[  136.060446 <    0.000028>] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  136.060571 <    0.000125>] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60030000 size 0xc
[  136.061930 <    0.001359>] sof-audio-pci 0000:00:0e.0: pcm: close stream 0 dir 0
[  136.541234 <    0.479304>] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[  136.541257 <    0.000023>] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  137.041208 <    0.499951>] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[  137.041229 <    0.000021>] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  138.311255 <    1.270026>] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  138.311382 <    0.000127>] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  138.311404 <    0.000022>] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
[  138.312468 <    0.001064>] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
```

Sof-logger:
```
0      2          MEM             50134.375000         2.552083      src/lib/alloc.c:823       count 4 free 4
    0      2          IPC             50213.906250        79.531250    src/ipc/handler.c:1004   ipc: pipe 17 -> complete
    0      2         PIPE 3.17        50225.989583        12.083333 src/audio/pipeline.c:174    pipeline_complete()
    0      2          IPC             50304.531250        78.541664    src/ipc/handler.c:1004   ipc: pipe 11 -> complete
    0      2         PIPE 2.11        50313.020833         8.489583 src/audio/pipeline.c:174    pipeline_complete()
    0      2          IPC             50391.562500        78.541664    src/ipc/handler.c:1004   ipc: pipe 5 -> complete
    0      2         PIPE 1.5         50396.822917         5.260417 src/audio/pipeline.c:174    pipeline_complete()
    0      2          IPC             50484.635417        87.812500    src/ipc/handler.c:856    ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC             50487.656250         3.020833    src/ipc/handler.c:858    ipc: comp 19 -> cmd 0
    0      2       VOLUME             50496.770833         9.114583   src/audio/volume.c:433    volume_cmd()
    0      2       VOLUME             50499.843750         3.072917   src/audio/volume.c:326    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 19
    0      2       VOLUME             50502.708333         2.864583   src/audio/volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 0, value = 65536
    0      2       VOLUME             50505.677083         2.968750   src/audio/volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 1, value = 65536
    0      2          IPC             50666.718750       161.041672    src/ipc/handler.c:856    ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC             50669.687500         2.968750    src/ipc/handler.c:858    ipc: comp 13 -> cmd 0
    0      2       VOLUME             50676.718750         7.031250   src/audio/volume.c:433    volume_cmd()
    0      2       VOLUME             50679.635417         2.916667   src/audio/volume.c:326    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 13
    0      2       VOLUME             50684.375000         4.739583   src/audio/volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 0, value = 65536
    0      2       VOLUME             50687.031250         2.656250   src/audio/volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 1, value = 65536
    0      2          IPC             50835.572917       148.541672    src/ipc/handler.c:856    ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC             50838.385417         2.812500    src/ipc/handler.c:858    ipc: comp 7 -> cmd 0
    0      2       VOLUME             50843.437500         5.052083   src/audio/volume.c:433    volume_cmd()
    0      2       VOLUME             50846.041667         2.604167   src/audio/volume.c:326    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 7
    0      2       VOLUME             50848.645833         2.604167   src/audio/volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 0, value = 65536
    0      2       VOLUME             50851.250000         2.604167   src/audio/volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 1, value = 65536
    0      2          IPC             51049.687500       198.437500    src/ipc/handler.c:856    ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC             51052.447917         2.760417    src/ipc/handler.c:858    ipc: comp 1 -> cmd 0
    0      2       VOLUME             51055.781250         3.333333   src/audio/volume.c:433    volume_cmd()
    0      2       VOLUME             51058.541667         2.760417   src/audio/volume.c:326    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 1
    0      2       VOLUME             51061.041667         2.500000   src/audio/volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 0, value = 41349
    0      2       VOLUME             51063.645833         2.604167   src/audio/volume.c:331    volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 1, value = 41349
    0      2          IPC             51245.468750       181.822922    src/ipc/handler.c:643    ipc: pm -> restore
    0      2          IPC             57958.854167      6713.385254    src/ipc/handler.c:239    ipc: comp -1 -> params
    0      1          IPC             57969.114583        10.260417    src/ipc/handler.c:244    ipc: comp -1 not found
    0      2          IPC             58073.125000       104.010414    src/ipc/handler.c:367    ipc: comp -1 -> free
```

Env:
sof master: e38db1a6
kernel sof-dev: ee19d24e
tplg: sof-byt-nocodec.tplg

Log files:
[check_aplay_function.sh-190227_104821.zip](https://github.com/thesofproject/sof/files/2908386/check_aplay_function.sh-190227_104821.zip)

ranj063, 2019-02-27:
This is due to the recently merged topology patch https://github.com/thesofproject/sof/pull/1047
but the unmerged kernel PR https://github.com/thesofproject/linux/pull/675

Jiangxinx, 2019-03-01:
Fix by thesofproject/linux#675 and  #1047.

closed 2019-03-01 by Jiangxinx