thesofproject/sof #1062: Register card failed on byt RT5651.

Jiangxinx, 2019-03-01:
Summary:
Register card failed with 'devm_snd_soc_register_card failed (-19)' reason.
sof-logger is empty. sof-logger -t log is attached below.

Step:
1. aplay -l

Error output:
aplay: device_list:270: no soundcards found...

Dmesg:
```
[   14.110713] sof-audio-acpi 80860F28:01: tplg: load pcm Passthrough 0
[   14.110729] sof-audio-acpi 80860F28:01: sink BUF1.0 control none source PCM0P
[   14.110744] sof-audio-acpi 80860F28:01: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.110820] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.110834] sof-audio-acpi 80860F28:01: sink PGA1.0 control none source BUF1.0
[   14.110848] sof-audio-acpi 80860F28:01: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.110910] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.110921] sof-audio-acpi 80860F28:01: sink BUF1.1 control none source PGA1.0
[   14.110935] sof-audio-acpi 80860F28:01: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.110996] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111009] sof-audio-acpi 80860F28:01: sink PCM0C control none source BUF2.0
[   14.111022] sof-audio-acpi 80860F28:01: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111084] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111095] sof-audio-acpi 80860F28:01: sink BUF2.0 control none source PGA2.0
[   14.111109] sof-audio-acpi 80860F28:01: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111170] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111181] sof-audio-acpi 80860F28:01: sink PGA2.0 control none source BUF2.1
[   14.111194] sof-audio-acpi 80860F28:01: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111257] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111268] sof-audio-acpi 80860F28:01: sink SSP2.OUT control none source BUF1.1
[   14.111281] sof-audio-acpi 80860F28:01: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111339] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111350] sof-audio-acpi 80860F28:01: sink BUF2.1 control none source SSP2.IN
[   14.111363] sof-audio-acpi 80860F28:01: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111422] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[   14.111440] sof-audio-acpi 80860F28:01: tplg: config SSP2 fmt 0x4001 mclk 19200000 bclk 1920000 fclk 48000 width (16)20 slots 2 mclk id 0
[   14.111475] sof-audio-acpi 80860F28:01: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[   14.111561] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[   14.111571] sof-audio-acpi 80860F28:01: ASoC: Parent card not yet available, widget card binding deferred
[   14.111575] sof-audio-acpi 80860F28:01: tplg: complete pipeline PIPELINE.2.SSP2.IN id 13
[   14.111587] sof-audio-acpi 80860F28:01: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[   14.111676] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[   14.111684] sof-audio-acpi 80860F28:01: tplg: complete pipeline PIPELINE.1.SSP2.OUT id 6
[   14.111696] sof-audio-acpi 80860F28:01: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[   14.111758] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[   14.113429] rt5651 i2c-10EC5651:00: Warning: Invalid over-current-threshold-microamp value: 0, defaulting to 2000uA
[   14.114097] bytcr_rt5651 bytcr_rt5651: ASoC: no sink widget found for ssp2 Tx
[   14.114110] bytcr_rt5651 bytcr_rt5651: ASoC: Failed to add route codec_out0 -> direct -> ssp2 Tx
[   14.114121] bytcr_rt5651 bytcr_rt5651: ASoC: no sink widget found for ssp2 Tx
[   14.114128] bytcr_rt5651 bytcr_rt5651: ASoC: Failed to add route codec_out1 -> direct -> ssp2 Tx
[   14.114139] bytcr_rt5651 bytcr_rt5651: ASoC: no source widget found for ssp2 Rx
[   14.114146] bytcr_rt5651 bytcr_rt5651: ASoC: Failed to add route ssp2 Rx -> direct -> codec_in0
[   14.114156] bytcr_rt5651 bytcr_rt5651: ASoC: no source widget found for ssp2 Rx
[   14.114164] bytcr_rt5651 bytcr_rt5651: ASoC: Failed to add route ssp2 Rx -> direct -> codec_in1
[   14.114174] bytcr_rt5651 bytcr_rt5651: ASoC: no source widget found for ssp2 Tx
[   14.114182] bytcr_rt5651 bytcr_rt5651: ASoC: Failed to add route ssp2 Tx -> direct -> AIF1 Playback
[   14.114192] bytcr_rt5651 bytcr_rt5651: ASoC: no sink widget found for ssp2 Rx
[   14.114199] bytcr_rt5651 bytcr_rt5651: ASoC: Failed to add route AIF1 Capture -> direct -> ssp2 Rx
[   14.114207] bytcr_rt5651 bytcr_rt5651: ASoC: failed to init SSP2-Codec: -19
[   14.114213] bytcr_rt5651 bytcr_rt5651: ASoC: failed to instantiate card -19
[   14.114304] bytcr_rt5651 bytcr_rt5651: Topology forgot to remove link Passthrough?
[   14.114323] bytcr_rt5651 bytcr_rt5651: devm_snd_soc_register_card failed -19
[   14.382752] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[   14.382838] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   14.705041] systemd-udevd (1480) used greatest stack depth: 12040 bytes left
```

Test env:
sof master: 58e90d90
kernel sof-dev: 65df8efe
tplg: sof-byt-rt5651.tplg

Log:
[logger.log](https://github.com/thesofproject/sof/files/2918108/logger.log)
[dmesg.log](https://github.com/thesofproject/sof/files/2918109/dmesg.log)

ranj063, 2019-03-01:
@Jiangxinx could you please check if this change in topology helps fix this?
diff --git a/tools/topology/sof-byt-rt5651.m4 b/tools/topology/sof-byt-rt5651.m4
index c64aaa48..4f8ebef2 100644
--- a/tools/topology/sof-byt-rt5651.m4
+++ b/tools/topology/sof-byt-rt5651.m4
@@ -87,3 +87,7 @@ DAI_CONFIG(SSP, 2, 0, SSP2-Codec,
                      SSP_CLOCK(fsync, 48000, codec_slave),
                      SSP_TDM(2, 25, 3, 3),
                      SSP_CONFIG_DATA(SSP, 2, 24)))
+
+VIRTUAL_WIDGET(ssp2 Rx, out_drv, 1)
+VIRTUAL_WIDGET(ssp2 Tx, out_drv, 2)

Jiangxinx, 2019-03-01:
@ranj063 I tried your change, can not build 
sof-byt-rt5651.tplg file.

build log：
```
0x00136c/004972 - route 'BUF1.1 ->  -> SSP5.OUT': write 132 bytes
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
ALSA lib conf.c:1877:(snd_config_load1) _toplevel_:1795:0:Unexpected end of file
ALSA lib parser.c:250:(tplg_load_config) error: could not load configuration file sof-byt-rt5651.conf
ALSA lib parser.c:328:(snd_tplg_build_file) error: failed to load topology file sof-byt-rt5651.conf

failed to compile context sof-byt-rt5651.conf
[ 60%] Generating sof-apl-rt298.tplg
topology/CMakeFiles/topology_sof-byt-rt5651.dir/build.make:60: recipe for target 'topology/sof-byt-rt5651.tplg' failed
make[2]: *** [topology/sof-byt-rt5651.tplg] Error 1
0x000000/000000 - header index 0 type 8 count 1 size 0x70/112 vendor 0 version 0
0x000024/000036 -manifest : write 112 bytes
0x000094/000148 -manifest : write 0 priv bytes
```

ranj063, 2019-03-01:
@Jiangxinx thanks, I'll take a look at it today.

plbossart, 2019-03-01:
|ETE |
> |[ 14.111684] sof-audio-acpi 80860F28:01: tplg: complete pipeline
> PIPELINE.1.SSP2.OUT id 6 [ 14.111696] sof-audio-acpi 80860F28:01: ipc
> tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE [ 14.111758] [...]
|looks like a mix of topology issues and possibly a failed codec 
interaction. Are you sure the codec was properly detected?|
> |[ 14.114207] bytcr_rt5651 bytcr_rt5651: ASoC: failed to init
> SSP2-Codec: -19 [ 14.114213] bytcr_rt5651 bytcr_rt5651: ASoC: failed
> to instantiate card -19 [ 14.114304] bytcr_rt5651 bytcr_rt5651: [...]
|so the card creation fails but the sof-audio-acpi continues. To me this 
is a 'feature' of the device model that we need to work around. The 
errors which happen during the probe of a platform driver are not 
propagated to the caller, so maybe we need a callback invoked from the 
machine drivers to notify the acpi/pci drivers things didn't pan out.|
>
>
>
>

ranj063, 2019-03-05:
@juimonen could you please take a look at this bug since you are the platform owner for BYT :)

wenqingfu, 2019-03-06:
If there's a working (card at least register) combo of linux and sof commit id on BYT RT5651, maybe a bisect can find the culprit since this bug is still a new regression?  add @juimonen as owner. 

It's a P1 since it's a showstopper on one supported platform

@Jiangxinx @stevyan  can you help with bisect, too?

Jiangxinx, 2019-03-07:
I have done bisect, find it is caused by a kernel commit: 
ASoC: SOF: remove dai capabilities from dai definitions   a044708ed

Here is the bisect results:
1. 
env:
    sof master: 58e90d90
    kernel sof-dev: f147f56b
    tplg: sof-byt-rt5651.tplg
result:  
    passed

2. 
env:
    sof master: 58e90d90
    kernel sof-dev: a044708ed
    tplg: sof-byt-rt5651.tplg
result:  
    failed

juimonen, 2019-03-07:
@Jiangxinx I don't see that commit in the tree at all. That same place is annotated to some code from @plbossart... Is this some squash/merge problem?

xiulipan, 2019-03-08:
@juimonen 
I think we have some 4.20.0 to 5.0.0 switch, and I do remember I saw this PR set 
https://github.com/thesofproject/linux/pull/675 and https://github.com/thesofproject/sof/pull/1047

@ranj063 could you have a look if that means we need some tplg change special for legacy platforms.
I believe some mismatch may be not be found from test since this change is some related to both tplg and driver.

ranj063, 2019-03-09:
@xiulipan my change that I mentioned in an earlier comment should fix it. I will check it today and submit a PR

ranj063, 2019-03-10:
@juimonen looks like the ssp2 TX/RX that were removed in ASoC: SOF: remove dai capabilities from dai definitions a044708ed caused the card registration failure. 

But reverting this commit only gives me an hw_params ipc error. needs more investigation

ranj063, 2019-03-10:
https://github.com/thesofproject/sof/pull/1108 fixes this issue
https://github.com/thesofproject/linux/pull/706 is also needed for BYT RT5651 to prevent hw_params failure

ranj063, 2019-03-15:
@Jiangxinx can this be closed?

xiulipan, 2019-03-15:
@ranj063
I think we are testing with test topology. Could you help to check with test topology or we should to abandon to use test topology?
I did not think we have full support for the mixer now? Or we should simplify the topology we are using for byt?

ranj063, 2019-03-16:
@xiulipan i think we should only use sof-byt-rt5651 and report any failures with it. 
It is our only topology with a mixer based pipeline so lets use it

Jiangxinx, 2019-03-19:
@ranj063 I have used sof-byt-rt5651 instead of test topology, and can register card,  but aplay failed. Sof-logger have no obvious error trace.

Dmesg:
```
[14012.011393] sof-audio-acpi 80860F28:01: Copying cached debugfs data
[14012.011741] sof-audio-acpi 80860F28:01: Copying cached debugfs data
[14012.011939] sof-audio-acpi 80860F28:01: Copying cached debugfs data
[14012.012437] sof-audio-acpi 80860F28:01: Copying cached debugfs data
[14012.013400] sof-audio-acpi 80860F28:01: Copying cached debugfs data
[14271.586576] sof-audio-acpi 80860F28:01: pcm: open stream 0 dir 0
[14271.586584] sof-audio-acpi 80860F28:01: period min 192 max 16384 bytes
[14271.586588] sof-audio-acpi 80860F28:01: period count 2 max 16
[14271.586592] sof-audio-acpi 80860F28:01: buffer max 65536 bytes
[14271.586972] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[14271.586978] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[14271.586986] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[14271.586990] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[14271.587002] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[14271.587006] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[14271.587014] sof-audio-acpi 80860F28:01: pcm: hw params stream 0 dir 0
[14271.587021] sof-audio-acpi 80860F28:01: generating page table for 0000000078243706 size 0xff00 pages 16
[14271.587026] sof-audio-acpi 80860F28:01: stream_tag 0
[14271.587037] sof-audio-acpi 80860F28:01: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[14271.888082] sof-audio-acpi 80860F28:01: error: ipc timed out for 0x60010000 size 0x6c
[14271.888163] sof-audio-acpi 80860F28:01: error: unexpected fault 0x00000000 trace 0x60010000
[14271.888176] sof-audio-acpi 80860F28:01: error: waking up any trace sleepers
[14271.888194] sof-audio-acpi 80860F28:01: error: hw params ipc failed for stream 0
[14271.888210] sof-audio-acpi 80860F28:01: ASoC: 80860F28:01 hw params failed: -110
[14271.888226]  Low Latency: ASoC: hw_params FE failed -110
[14271.888257] sof-audio-acpi 80860F28:01: pcm: free stream 0 dir 0
[14271.888271] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[14272.192173] sof-audio-acpi 80860F28:01: error: ipc timed out for 0x60030000 size 0xc
[14272.192253] sof-audio-acpi 80860F28:01: error: unexpected fault 0x00000000 trace 0x60030000
[14272.192266] sof-audio-acpi 80860F28:01: error: waking up any trace sleepers
[14272.203048] sof-audio-acpi 80860F28:01: pcm: close stream 0 dir 0
[14276.178191] sof-audio-acpi 80860F28:01: pcm: open stream 1 dir 0
[14276.178200] sof-audio-acpi 80860F28:01: period min 192 max 262144 bytes
[14276.178204] sof-audio-acpi 80860F28:01: period count 2 max 32
[14276.178207] sof-audio-acpi 80860F28:01: buffer max 8388608 bytes
[14276.178689] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[14276.178695] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[14276.178703] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[14276.178707] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[14276.178719] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[14276.178723] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[14276.178731] sof-audio-acpi 80860F28:01: pcm: hw params stream 1 dir 0
[14276.178753] sof-audio-acpi 80860F28:01: generating page table for 0000000017a5c9e9 size 0x17700 pages 24
[14276.178759] sof-audio-acpi 80860F28:01: stream_tag 0
[14276.178769] sof-audio-acpi 80860F28:01: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[14276.480084] sof-audio-acpi 80860F28:01: error: ipc timed out for 0x60010000 size 0x6c
[14276.480164] sof-audio-acpi 80860F28:01: error: unexpected fault 0x00000000 trace 0x60010000
[14276.480177] sof-audio-acpi 80860F28:01: error: waking up any trace sleepers
[14276.480195] sof-audio-acpi 80860F28:01: error: hw params ipc failed for stream 0
[14276.480211] sof-audio-acpi 80860F28:01: ASoC: 80860F28:01 hw params failed: -110
[14276.480227]  Media Playback 1: ASoC: hw_params FE failed -110
[14276.480258] sof-audio-acpi 80860F28:01: pcm: free stream 1 dir 0
[14276.480272] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[14276.784174] sof-audio-acpi 80860F28:01: error: ipc timed out for 0x60030000 size 0xc
[14276.784284] sof-audio-acpi 80860F28:01: error: unexpected fault 0x00000000 trace 0x60030000
[14276.784295] sof-audio-acpi 80860F28:01: error: waking up any trace sleepers
[14276.787036] sof-audio-acpi 80860F28:01: pcm: close stream 1 dir 0
```

Test env:
sof master: 18165867
kernel sof-dev: 41e4680a
tplg: sof-byt-rt5651.tplg

lgirdwood, 2019-03-19:
@Jiangxinx @ranj063 this looks like a FW issue, I assume there is nothing in the error log ? Might be worth adding more trace here.

ranj063, 2019-03-19:
@lgirdwood @xiulipan you need my linux PR https://github.com/thesofproject/linux/pull/708 to get aplay to work on the minnowboard

stevyan, 2019-03-22:
**Summary:**
This issue has been fixed, and doesn't occurred within daily build based on BYT rt5651.

**Test env:**
sof master: 6e88bdf1
kernel sof-dev: 95b2fab5
tplg: sof-byt-rt5651.tplg

closed 2019-03-22 by stevyan