thesofproject/sof #1063: CLK: MAX amplifier generate only noise on WHL/CML paltform

zhuyingjiang, 2019-03-01:
This issue is from #1048, 
There is only 24M MCLK on WHL/CML platform, but the MAX98357 amplifier need the BCLK to be 32/48/64 X of LRCK. so now there is only noise from the speaker.
There have 2 solutions with this: 1. enable the M/N divider(track by #1048) ; 2. use the SRC pipeline.
This issue track the SRC solution.

lgirdwood, 2019-03-11:
@zhuyingjiang any update or can this be closed ?

zhuyingjiang, 2019-03-12:
@lgirdwood I have test this on platform GLK with MAX98357, with PR#1099, and give the reference for @sathyap-chrome, 16bits is said working on CML with MAX98357, will wait the feedback if 24/32 still works on that.
 Will close this once get the feedback.

michalgrodzicki, 2019-03-26:
@sathyap-chrome - any update?

sathyap-chrome, 2019-03-26:
Hi Michal, sorry for the delay. due to other issues, i couldnt try this (24/32) playback . Will provide an update tomorrow EOD my time.

sathyap-chrome, 2019-03-28:
Hi @lgirdwood  with 16 bits, playback was not pleasant with FSYNC of 50000. Hence we are currently using 46875 FSYNC with 16 bits playback.

With 50000 FSYNC, did you observe playbacking going unpleasant on GKL after sometime ( ~ 2 - 3 mins of playback)

With 24 bit i'm not getting a good FSYNC value . I'll try 24 bit with FSYNC 50000 and check if the playback is fine.

plbossart, 2019-03-28:
@sathyap-chrome why would you use 46875Hz? This isn't a frequency we want to use at all...I know it's a convenient 24MHz/512, but that's not the frequency we resample to. We need to close this issue asap.

sathyap-chrome, 2019-03-28:
@plbossart  This was recommended by maxim team. Please find attached mail discussion (attached as .txt). @bardliao is copied.
[Re BCLK not being right for MAX98357a.txt](https://github.com/thesofproject/sof/files/3018744/Re.BCLK.not.being.right.for.MAX98357a.txt)

zhuyingjiang, 2019-03-29:
link the [PR1138 ](https://github.com/thesofproject/sof/pull/1138)

wenqingfu, 2019-04-04:
> @plbossart This was recommended by maxim team. Please find attached mail discussion (attached as .txt). @bardliao is copied.

@plbossart good or not good to close this issue now?

plbossart, 2019-04-04:
NAK. we don't have a SRC to 46.875kHz so this is not acceptable from an audio quality perspective.

lgirdwood, 2019-04-05:
@zhuyingjiang @sathyap-chrome can you check with your contacts if stereo TDM mode is less restrictive on clock ratios. I had a quick look at the datasheet and it's difficult to say. https://datasheets.maximintegrated.com/en/ds/MAX98357A-MAX98357B.pdf

sathyap-chrome, 2019-04-05:
Hi @lgirdwood , response from maxim.
Hi Sathya,

TDM Mode is always 8 channel, not stereo. What is the issue with clock ratios that you are seeing?

-	From the datasheet.

“The digital audio interface is highly flexible with the MAX98357A supporting I2S data and the MAX98357B supporting left-justified data. Both ICs support 8 channel time division multiplexed (TDM) data.”

Thanks,
Shalin.

zhuyingjiang, 2019-04-24:
@sathyap-chrome Can this be close now?

sathyap-chrome, 2019-04-24:
We can close this as we have a working configuration.
We can keep the discussion on the exact BCLK for customer board configuration -Since we see different configuration being asked by customers.

closed 2019-04-24 by zhuyingjiang