thesofproject/sof #1073: [PM] Failed to aplay source after suspend/resume without playback on APL-nocodec

Jiangxinx, 2019-03-04:
Summary:
Fail to check aplay function after suspend/resume 50 times without plaback on APL-nocodec.
sof-logger is empty. sof-logger -t log is attached below.

Step：
1.sudo ./stress-s3.sh 50 1 0

Dmesg:
```
[  349.430069] r8169 0000:03:00.0 enp3s0: Link is Up - 100Mbps/Full - flow control rx/tx
[  350.582642] PM: suspend entry (deep)
[  350.582646] PM: Syncing filesystems ... done.
[  350.583335] Freezing user space processes ... (elapsed 0.001 seconds) done.
[  350.585322] OOM killer disabled.
[  350.585323] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[  350.587053] printk: Suspending console(s) (use no_console_suspend to debug)
[  350.588174] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  350.588198] r8169 0000:03:00.0 enp3s0: Link is Down
[  350.895170] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x40010000 size 0x4c
[  350.895182] sof-audio-pci 0000:00:0e.0: unknown ROM status value 01006701
[  350.895187] sof-audio-pci 0000:00:0e.0: error: status = 0x001d003c panic = 0x00000000
[  350.895192] sof-audio-pci 0000:00:0e.0: unknown ROM status value 01006701
[  350.895212] sof-audio-pci 0000:00:0e.0: error: ctx_save ipc error during suspend -110
[  350.946112] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1d003c timedout
[  350.946122] sof-audio-pci 0000:00:0e.0: error: reset enter failed: core_mask 3 adspcs 0x1d003c
[  350.946125] sof-audio-pci 0000:00:0e.0: error: dsp core reset failed: core_mask 3
[  350.946128] sof-audio-pci 0000:00:0e.0: error: failed to power down core during suspend
[  350.946131] sof-audio-pci 0000:00:0e.0: error: suspending dsp
[  350.946135] sof-audio-pci 0000:00:0e.0: error: failed to power down DSP during suspend -5
[  350.946149] pci_pm_suspend(): snd_sof_suspend+0x0/0x10 [snd_sof] returns -5
[  350.946158] dpm_run_callback(): pci_pm_suspend+0x0/0x120 returns -5
[  350.946165] PM: Device 0000:00:0e.0 failed to suspend: error -5
[  350.946170] PM: Some devices failed to suspend, or early wake event detected
[  351.057779] r8169 0000:02:00.0 enp2s0: Link is Down
[  351.058014] r8169 0000:03:00.0 enp3s0: Link is Down
[  351.058424] OOM killer enabled.
[  351.058425] Restarting tasks ... done.
[  351.060332] video LNXVIDEO:00: Restoring backlight state
[  351.060345] PM: suspend exit
[  351.257025] ata2: SATA link down (SStatus 4 SControl 300)
[  351.264780] ata1: SATA link down (SStatus 4 SControl 300)
[  353.162866] r8169 0000:03:00.0 enp3s0: Link is Up - 100Mbps/Full - flow control rx/tx
[  353.305383] sof-audio-pci 0000:00:0e.0: pcm: open stream 0 dir 0
[  353.305388] sof-audio-pci 0000:00:0e.0: error: pcm open failed to resume -22
[  353.305395] sof-audio-pci 0000:00:0e.0: ASoC: can't open component 0000:00:0e.0: -22
[  353.305398]  Port0: ASoC: failed to start FE -22
```

Env:
sof master: 24e241b
kernel sof-dev: cf5978ae
tplg: sof-apl-nocodec.tplg

Log:
[dmesg.log](https://github.com/thesofproject/sof/files/2924841/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/2924842/logger.log)

lgirdwood, 2019-03-05:
@Jiangxinx which board ?

Jiangxinx, 2019-03-06:
@lgirdwood APL-UP2-NOCODEC

plbossart, 2019-03-06:
Can you clarify the reference to 'aplay'?

are you 
- running the test and you see a ROM error before running aplay, in which case aplay is really irrelevant?
- or are you done with the test and resuming when aplay starts?

Jiangxinx, 2019-03-06:
@plbossart Attach some details:
1. aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero
   result: passed.
2. sudo ./stress-s3.sh 50 1 0
  result:
```
== start time: 2019-03-06 18:25:52 ==
---- test 1 of 1 ----
System will suspend after 1 seconds ...
system will resume after 2 seconds ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Wed Mar  6 10:25:56 2019
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
aplay: main:788: audio open error: Invalid argument
Failed to aplay source!
== end time: 2019-03-06 18:26:00 ==
```

lgirdwood, 2019-03-11:
@Jiangxinx can you try with this https://github.com/thesofproject/linux/pull/704

ranj063, 2019-03-18:
@zhuyingjiang im starting to suspect if this is a regression at all? Are you sure rtcwake -s 2 -m mem ever worked with nocodec before? It looks like the DSP cannot be powered back on after entering S3.

But if I do "rtcwake -s 2 -m freeze", the system enters S2Idle and resume fine.

stevyan, 2019-03-18:
I've rolled back the build version 20190227 to check this issue, it works well. However, this issue is always exist since issue #1054 occurred (Fixed by thesofproject/linux#675 and #1047) and never disappear. Anyway, the S3 operation is working (suspend/resume), but, the aplay will be failed till now.

lgirdwood, 2019-03-18:
@stevyan, @Jiangxinx  the IPC PM context save is failing, so this looks more like a FW issue.
```
[  350.588174] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  350.588198] r8169 0000:03:00.0 enp3s0: Link is Down
[  350.895170] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x40010000 size 0x4c
[  350.895182] sof-audio-pci 0000:00:0e.0: unknown ROM status value 01006701
[  350.895187] sof-audio-pci 0000:00:0e.0: error: status = 0x001d003c panic = 0x00000000
[  350.895192] sof-audio-pci 0000:00:0e.0: unknown ROM status value 01006701
[  350.895212] sof-audio-pci 0000:00:0e.0: error: ctx_save ipc error during suspend -110
[  350.946112] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1d003c timedout
[  350.946122] sof-audio-pci 0000:00:0e.0: error: reset enter failed: core_mask 3 adspcs 0x1d003c
[  350.946125] sof-audio-pci 0000:00:0e.0: error: dsp core reset failed: core_mask 3
[  350.946128] sof-audio-pci 0000:00:0e.0: error: failed to power down core during suspend
[  350.946131] sof-audio-pci 0000:00:0e.0: error: suspending dsp
[  350.946135] sof-audio-pci 0000:00:0e.0: error: failed to power down DSP during suspend -5
```
The driver expectation is that this and subsequent commands will succeed, which they usually do. This first command that fails is the PM IPC and that then cascades into driver side PM failures.

@ranj063 @plbossart I think it's worthwhile here to see if forcing the device into D3 at the PCI level helps at all since it looks like the SHIM level PM get stuck at this point. I'm not sure if the PM core will set D3 if we return errors like above in the driver.

ranj063, 2019-03-18:
@lgirdwood the log from above is a bit old. We have address the CTX_SAVE failure that happens prior to entering suspend by not reading the mem windo for this IPC.
So now the log looks like this: 
PM: suspend entry (deep)
[40556.846651] PM: Syncing filesystems ... done.
[40556.849149] Freezing user space processes ... (elapsed 0.000 seconds) done.
[40556.850008] OOM killer disabled.
[40556.850010] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[40556.851313] printk: Suspending console(s) (use no_console_suspend to debug)
[40556.854808] r8169 0000:02:00.0 enp2s0: Link is Down
[40556.865326] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000044
[40556.865943] sof-audio-pci 0000:00:0e.0: loading firmware
[40556.865959] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000002 at  00000048
[40556.865964] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000000 at  00000048
[40556.865968] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000044
[40556.865973] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000004 at  00000044
[40556.865975] sof-audio-pci 0000:00:0e.0: booting DSP firmware
[40556.866080] sof-audio-pci 0000:00:0e.0: period_bytes:0x0
[40556.866083] sof-audio-pci 0000:00:0e.0: periods:1
[40556.866129] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3030303 successful
[40556.866134] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3030302 successful
[40556.866138] sof-audio-pci 0000:00:0e.0: unstall/run core: core_mask = 1
[40556.866143] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 1 : core_mask 1
[40556.867278] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000000 successful
[40556.867284] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010202 successful
[40556.884155] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x5000001 successful
[40556.904344] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x5 successful
[40556.904375] sof-audio-pci 0000:00:0e.0: Firmware download successful, booting...
[40556.904430] sof-audio-pci 0000:00:0e.0: ipc rx: 0x70000000: FW_READY
[40556.904436] sof-audio-pci 0000:00:0e.0: ipc: DSP is ready 0x70000000 offset 0x81000
[40556.904441] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x70000000: FW_READY
[40556.904476] sof-audio-pci 0000:00:0e.0: firmware boot complete
[40556.904487] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000048
[40556.904494] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000002 at  00000048
[40556.904501] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000004 at  00000044
[40556.904508] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000000 at  00000044
[40556.904530] sof-audio-pci 0000:00:0e.0: period_bytes:0x0
[40556.904535] sof-audio-pci 0000:00:0e.0: periods:1
[40556.904545] sof-audio-pci 0000:00:0e.0: stream_tag: 1
[40556.904575] sof-audio-pci 0000:00:0e.0: ipc tx: 0x90010000: GLB_TRACE_MSG
[40556.905232] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x90010000: GLB_TRACE_MSG
[40556.905286] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.905452] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.905490] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.905627] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.905660] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[40556.905802] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[40556.905834] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[40556.905930] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[40556.905963] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.906179] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.906224] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[40556.906315] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[40556.906329] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 1 : core_mask 1
[40556.906347] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[40556.906402] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[40556.906432] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.906517] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.906548] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.906630] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.906655] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[40556.906741] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[40556.906766] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[40556.906847] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[40556.906907] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.906996] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[40556.907023] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[40556.907099] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[40556.907111] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 1 : core_mask 1
[... cut ...]
[40560.088434] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1d003c timedout
[40560.088447] sof-audio-pci 0000:00:0e.0: error: reset enter failed: core_mask 3 adspcs 0x1d003c
[40560.088461] sof-audio-pci 0000:00:0e.0: error: dsp core reset failed: core_mask 3
[40560.088468] sof-audio-pci 0000:00:0e.0: error: failed to power down core during suspend
[40560.088476] sof-audio-pci 0000:00:0e.0: error: failed to power down DSP during suspend -5
[40560.088496] sof-audio-pci 0000:00:0e.0: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -5)

lgirdwood, 2019-03-19:
@ranj063 thanks for the update, did we try and extend the ROM timeout ?

ranj063, 2019-03-19:
@lgirdwood yes, for APL it has been increased from 70 to 150ms. And for CML it has been increased to 300 ms

ranj063, 2019-03-20:
> I've rolled back the build version 20190227 to check this issue, it works well. However, this issue is always exist since issue #1054 occurred (Fixed by [...]

@stevyan could you please check something for me with the build version 20190227?
Was the DSP getting powered off and on during the S3 test?

Jiangxinx, 2019-03-20:
@ranj063 I have checked the DSP status with version 20190227 and daily version.
step:
   1.sudo ./stress-s3.sh 50 1 0
   2.sudo cat  /sys/bus/pci/devices/0000\:00\:0e.0/power/runtime_status

Output(20190227):
   active

Dmesg(20190227):
```
[  158.785845] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000044
[  158.785853] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000040 at  00000048
[  158.785859] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000000 at  00000048
[  158.788827] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000048
[  158.788835] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000040 at  00000048
[  158.788842] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000040 at  00000048
[  158.788848] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000000 at  00000048
[  158.788854] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000048
[  158.788860] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000040 at  00000048
[  158.788916] sof-audio-pci 0000:00:0e.0: loading firmware
[  158.788935] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000002 at  00000048
[  158.788939] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000000 at  00000048
[  158.788945] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000044
[  158.788950] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000004 at  00000044
[  158.788953] sof-audio-pci 0000:00:0e.0: booting DSP firmware
[  158.789125] sof-audio-pci 0000:00:0e.0: period_bytes:0x0
[  158.789129] sof-audio-pci 0000:00:0e.0: periods:1
[  158.790242] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3030303 successful
[  158.790251] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3030302 successful
[  158.790256] sof-audio-pci 0000:00:0e.0: unstall/run core: core_mask = 1
[  158.790264] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 1 : core_mask 1
[  158.791414] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000000 successful
[  158.791420] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010202 successful
[  158.815708] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x5000001 successful
[  158.835776] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x5 successful
[  158.835783] sof-audio-pci 0000:00:0e.0: Firmware download successful, booting...
[  158.840532] sof-audio-pci 0000:00:0e.0: ipc rx: 0x70000000: FW_READY
[  158.840535] sof-audio-pci 0000:00:0e.0: ipc: DSP is ready 0x70000000 offset 0x81000
[  158.840540] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x70000000: FW_READY
[  158.840665] sof-audio-pci 0000:00:0e.0: firmware boot complete
[  158.840676] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000048
[  158.840680] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000002 at  00000048
[  158.840685] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000004 at  00000044
[  158.840688] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000000 at  00000044
[  158.840707] sof-audio-pci 0000:00:0e.0: period_bytes:0x0
[  158.840709] sof-audio-pci 0000:00:0e.0: periods:1
[  158.840716] sof-audio-pci 0000:00:0e.0: stream_tag: 1
[  158.840745] sof-audio-pci 0000:00:0e.0: ipc tx: 0x90010000: GLB_TRACE_MSG
[  158.841429] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x90010000: GLB_TRACE_MSG
[  158.841472] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  158.841629] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  158.841673] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  158.841937] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  158.841972] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  158.842226] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  158.842261] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  158.842475] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  158.842500] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  158.842797] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  158.842834] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[  158.842952] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[  158.842969] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 1 : core_mask 1
[  158.842984] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[  158.843215] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[  158.843254] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  158.843363] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  158.843388] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  158.843489] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  158.843508] sof-audio-pci 0000:00:0e.0: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  158.843718] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[... 273 lines cut ...]
[  139.935702] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  140.241036] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x40010000 size 0x4c
[  140.241046] sof-audio-pci 0000:00:0e.0: unknown ROM status value 01006701
[  140.241051] sof-audio-pci 0000:00:0e.0: error: status = 0x001d003c panic = 0x00000000
[... cut ...]
[  140.292007] PM: Some devices failed to suspend, or early wake event detected
[  140.403390] r8169 0000:03:00.0 enp3s0: Link is Down
[  140.404521] OOM killer enabled.
[  140.404523] Restarting tasks ... done.
[  140.407299] video LNXVIDEO:00: Restoring backlight state
[  140.407316] PM: suspend exit
[  140.602822] ata2: SATA link down (SStatus 4 SControl 300)
[  140.602861] ata1: SATA link down (SStatus 4 SControl 300)
[  142.036709] r8169 0000:03:00.0 enp3s0: Link is Up - 100Mbps/Full - flow control rx/tx
```

zhuyingjiang, 2019-03-20:
@ranj063 this issue is caused by commit 9a5e2e4ff4 and 7b12f064, revert these two can make the test passed.
But since 7b12f064 is to make NOCODEC and HDA mutually incompatible. This means NOCODEC is still depend on HDA when do suspend/resume, so there should be some bug during the NOCODEC suspend/resume.
I will go deep with this.

ranj063, 2019-03-20:
@zhuyingjiang could you please paste commit id with a link so its easier to check?

zhuyingjiang, 2019-03-21:
@ranj063 Updated, and I made a PR[#739](https://github.com/thesofproject/linux/pull/739) for this issue, could you review it for me?

Jiangxinx, 2019-03-29:
passed on daily test.

closed 2019-03-29 by Jiangxinx

Jiangxinx, 2019-03-29:
Not reproduce on daily test.
Step：
1. sudo ~/test/stress-s3.sh 50 1 0

Env:
sof master: 742f83ba
kernel sof-dev: e29122d6
tplg: sof-apl-nocodec.tplg

mengdonglin, 2019-04-01:
Fixed by https://github.com/thesofproject/linux/pull/739, same as https://github.com/thesofproject/sof/issues/1072