thesofproject/sof #1080: platform: merge platform.h and plat_cfg.h

lgirdwood, 2019-03-05:
Both files do the same thing so merge them.

tlauda, 2019-03-05:
@lgirdwood platcfg.h is also used in asm code, so it might not be that easy.

lgirdwood, 2019-03-05:
ifdef ASSEMBLY etc should fix assembly usage.

abonislawski, 2019-03-15:
It is more tricky because it will be included, for example, from base_module.c (memory.h->platform.h) where ASSEMBLY def will not work as for .S files

So, it would be necessary to make whole platform.h working (without ifdef ASSEMBLY) which means moving some problematic things from platform.h to another already included files at this stage. Not sure if worth it.

OR instead of #ifdef ASSEMBLY use something like #ifdef sof_target (if it is possible) which will effectively split the file in a needed way but Im not sure if it is a better way than two files.

Maybe someone have a better idea?

lgirdwood, 2019-03-15:
> It is more tricky because it will be included, for example, from base_module.c (memory.h->platform.h) where ASSEMBLY def will not work as for .S files

Not following. All .S files must be built with -DASSEMBLY meaning we can move any C code behind an ifdef. Same goes for linker script usage as we define -DLINKER now too.

i.e. platcfg.h
```
/* all generic macros here */
...

/* All C stuff here */
#if  !defined(ASSEMBLY) && !defined(LINKER)

/* C macros and declarations here */

#endif
```

abonislawski, 2019-03-19:
Yes, thats right. Maybe I have explained this not good enough, but merging this two files with if !defined(ASSEMBLY) is very easy but the problem is to compile it :) Because the includes flow will be different now (in platform.h we have some extra includes comparing to platcfg.h),
You can check working version for APL+CNL+ICL here:
https://github.com/abonislawski/sof/commit/65f83bf818b044ae40fc25a8291ab3750da3f713

Im not sure if the includes should look like this, maybe there will be needed bigger rework?

lgirdwood, 2019-03-19:
The patch looks OK to me, what errors do you see when building ?

abonislawski, 2019-03-20:
This one is working without errors but I had to play a little with includes flow (without that there were errors like implicit declaration or undefined reference, etc.).
In some places I didnt just replaced platcfg.h with platform.h, for example in memory.h platcfg.h was deleted but added platform.h to linker script. In alloc.h I have replaced memory.h with platform.h. Also deleted in cpu.h without adding platform.h
Thats why Im not sure if thats the correct way, please check carefully. If its okay then I will open a PR

lgirdwood, 2019-03-21:
@abonislawski please do send a PR, this way CI will give it a good test too. :)

abonislawski, 2019-03-25:
Done in PR #1171 so can be closed

closed 2019-03-25 by abonislawski