thesofproject/sof #1106: nocodec mode to set LBM in quirks

fredoh9, 2019-03-08:
SSP dai token has quirks which support LBM. Nocodec mode should utilize this to capture PCM data from playback pipeline. This idea came up while discussing https://github.com/thesofproject/linux/issues/640,
https://github.com/thesofproject/linux/pull/682

SOF project is ready to handle this. Only topology file change is required.

>sscr1 |= (ssp->params.quirks & SOF_DAI_INTEL_SSP_QUIRK_LBM) ? SSCR1_LBM : 0;

fredoh9, 2019-03-18:
I'm using UP2 board to test this. Initially scope is very narrow to set quirks bit but it turns out something else is missing.

This is what I’m doing and debugging.
1. Using nocodec mode in UP2. Sound card is enumerated
2. Able to play 48Khz stereo wav file in PCM0P with aplay and capture PCM samples in PCM0C with arecord
3. The Captured wav file has only silence
4. Set LBM bit in config->ssp.quirks, sof_link_ssp_load(), rebuilt kernel (Eventually this will be set from the topology file later)
5. Test same testing above again
6. But only silence packets are captured

I printed quirks from kernel, it is 64=0x40 in sof_link_ssp_load()
> #define SOF_DAI_INTEL_SSP_QUIRK_LBM                         (1 << 6)
> 
> [    4.003976] sof-audio-pci 0000:00:0e.0: tplg: config SSP0 fmt 0x4001 mclk 24576000 bclk 1536000 fclk 48000 width (16)16 slots 2 mclk id 0 quirks 64

And I checked register values in DSP too. sscr1 = 0xd0700004. This means SSCR1 set LBM bit correctly.
> #define SSCR1_LBM       (1 << 2)
> 
>     0      2          SSP             52440.156250         3.645833 ers/intel/cavs/ssp.c:184            ssp_set_config(), config->format = 0x4001
>     0      2          SSP             52447.708333         7.552083 ers/intel/cavs/ssp.c:769            ssp_set_config(), sscr0 = 0xc1c00f7f, sscr1 = 0xd0700004, ssto = 0x00000000, sspsp = 0x2100000

fredoh9, 2019-03-19:
Interesting results. Same change works in today's tip.
Linux : 41e4680a035ec574a170947b79a93c6db3a85749 + Fred Hard corded change
SOF : 181658671f8a60d011bf9de9a5b4a06c2800c096
I will submit a PR with topology changes soon.

wenqingfu, 2019-03-27:
fixed by #1151.

@fredoh9  looks like you need to put the "fixes" magic word in the PR description post for it to automatically close fixed issue. Thanks!

closed 2019-03-27 by wenqingfu