thesofproject/sof #1113: [S3] Fail to aplay source after suspend/resume 42 times on APL-UP2-pcm512x

Jiangxinx, 2019-03-12:
Summary:
Fail to check aplay function after suspend/resume 50 times on APL-UP2-pcm512x.
Dmesg have no abvious error trace.

Step：
1.aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero& sudo ~/test/stress-s3.sh 50 1 1

Output:
```
---- test 42 of 50 ----
System will suspend after 2 seconds ...
system will resume after 4 seconds ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Mar 12 08:44:34 2019
Suspended. Trying resume. Done.
aplay: pcm_write:2011: write error: Input/output error
aplay doesn't exist!
```

Sof-logger:
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA             79677.864583     79677.867188 intel/cavs/hda-dma.c:344 	hda-dmac: 5 preload timeout
    0      1         HOST             79684.114583         6.250000     src/audio/host.c:766 	host_copy() error: dma_set_config() or dma_start() failed, ret = 4294967234
    0      1         PIPE             79690.208333         6.093750 src/audio/pipeline.c:658 	pipeline_copy() error: ret = -62, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5         79696.875000         6.666667 src/audio/pipeline.c:766 	pipeline_xrun_recover()
    0      1         COMP             79709.531250        12.656250 rc/audio/component.c:192 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_PREPARE
    0      1         PIPE             79715.208333         5.677083 src/audio/pipeline.c:352 	pipeline_prepare() error: ret = -22,dev->comp.id = 0
    0      1         PIPE 1.5         79721.041667         5.833333 src/audio/pipeline.c:773 	pipeline_xrun_recover() error: pipeline_prepare() failed, ret = -22
    0      1         PIPE 1.5         79726.354167         5.312500 src/audio/pipeline.c:842 	pipeline_task(): xrun recover failed! pipeline will be stopped!
```

Env:
sof master: 2dbe69e2
kernel sof-dev: facd448b
tplg: sof-apl-pcm512x.tplg

Log file:
[dmesg.log](https://github.com/thesofproject/sof/files/2955860/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/2955862/logger.log)

lgirdwood, 2019-03-12:
@Jiangxinx this is very similar to #1050, the log looks very similar in that HDA DAM timeout failure cascades into other errors.

Jiangxinx, 2019-03-18:
This issue can still be reproduced on daily test.

Env:
sof master: eb8be5be
kernel sof-dev: 41e4680a
tplg: sof-apl-pcm512x.tplg

wenqingfu, 2019-03-20:
#1050 was closed by @RanderWang 's patch thesofproject/linux/#705 which merged several days ago. If @Jiangxinx 's report above still fails at the moment, it's probably not the same issue

mengdonglin, 2019-03-28:
@Jiangxinx can you confirm if you can still reproduce this issue? @stevyan

Jiangxinx, 2019-03-28:
This issue can still be reproduced. Fail to check aplay function after suspend/resume 29 times on APL-UP2-pcm512x.
Sof-logger:
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA             71718.489583     71718.492188 src/drivers/dw/dma.c:500 	dw-dma: 0 channel 0 not ready
    0      1          DMA             71724.739583         6.250000 src/drivers/dw/dma.c:504 	 ena 0x0 cfglow 0x80000223 status 0x5
    0      1          IPC             71731.614583         6.875000 src/audio/pipeline.c:548 	pipeline_trigger() error: ret = -16, host->comp.id = 0, cmd = 1
    0      1          IPC             71738.125000         6.510417    src/ipc/handler.c:498 	ipc: comp 0 trigger 0x40000 failed -16
```

Env:
sof master: 44705dfc
kernel sof-dev: 03670025
tplg: sof-apl-pcm512x.tplg

mengdonglin, 2019-03-28:
@Jiangxinx please check the sof-apl-pcm512x.tplg is the latest as issue https://github.com/thesofproject/sof/issues/1114

tlauda, 2019-03-28:
@Jiangxinx Can you attach the full FW log? The above error means that the component hasn't been set to COMP_STATE_PREPARE after the last playback. Maybe we didn't stop the pipeline?

Jiangxinx, 2019-03-29:
Passed on daily test.

closed 2019-03-29 by Jiangxinx

mengdonglin, 2019-03-29:
@Jiangxinx Please share the test recipe of today's daily test.

Jiangxinx, 2019-03-29:
Not reproduce on daily test.
Step：
1.aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero& sudo ~/test/stress-s3.sh 50 1 1

Env:
sof master: 742f83ba
kernel sof-dev: e29122d6
tplg: sof-apl-pcm512x.tplg