thesofproject/sof #1128: allocator: unify alloc API.

lgirdwood, 2019-03-14:
Unify the allocator API so that rballoc becomes a static function that is called internally by rmalloc. rmmaloc() will make a judgement on calling rballoc depending on request size. i.e.
1. rmalloc uses current block for small requests.
2. rmallloc uses rballoc for large requests.

This also allows the block sizes to be shrunk below the rballoc block size to make the allocations more efficient.

RanderWang, 2019-03-20:
@lgirdwood  One question: there is no lock to protect buffer heap used by rballoc which is only called at setup time. And other heaps are divided into small heaps by core number, so there is no need to protect these heaps.  to add lock to buffer heap ?  Do you have any advice? Thanks!

RanderWang, 2019-03-20:
@lgirdwood  sorry, I made a mistake: there is a lock to protect buffer heap.

RanderWang, 2023-04-07:
it is out of date and was done in fw

closed 2023-04-07 by RanderWang