thesofproject/sof #1147: Broken playback on Up^2 with HifiBerry

lyakh, 2019-03-19:
Playback on Up^2 with a HifiBerry codec and the sof-apl-pcm512x topology currently produces strongly metallic sound. Bisection and a manual verification identified https://github.com/thesofproject/sof/pull/1045/commits/aebe3cef50424fc46915ed6dfa8d18622d22eaa1 from @tlauda as the offending commit.

tlauda, 2019-03-19:
@lyakh Can you upload FW logs?

ranj063, 2019-03-19:
@lyakh @tlauda I think this is the same issue as https://github.com/thesofproject/sof/issues/1107
on the Up2

tlauda, 2019-03-19:
@ranj063 On APL we are using Host DMA, so the flow is a little bit different. We don't observe any playback issues on APL, so that's why I'm asking if it's HDMI or SSP and if there is any bit conversion in between.

ranj063, 2019-03-19:
@tlauda since Guennadi mentions the hifiberry I have to assume it is the SSP. For HDMI the hifiberry has no role to play.

ranj063, 2019-03-19:
@xiulipan @lyakh @tlauda I checked with the latest master (commit: 6e88bdf1) and there is no noise with the hifiberry.

But #1107 still remains.

lyakh, 2019-03-19:
I think this is different from #1107 since that one also mentions a different commit as the culprit. Yes, SSP. The audio format is "Float 32 bit Little Endian, Rate 8000 Hz, Stereo" Log attached, I added a couple of empty lines between initialisation and playback.
[metal.txt](https://github.com/thesofproject/sof/files/2984476/metal.txt)

lyakh, 2019-03-19:
@ranj063 I just double-checked with the same HEAD commit and the distortion is still there.

tlauda, 2019-03-19:
@lyakh 8 kHz means that the SSP is also set to 8 kHz or is there SRC in the pipeline?

ranj063, 2019-03-19:
@lyakh what topology are you using?
sof-apl-pcm512x.tplg without the SRC works without noise at my end.

lyakh, 2019-03-19:
@ranj063 it turned out I was using the topology with the src, sorry for confusion. And it used to work until that commit. I tried with the "no src" topology and it's working now. Is there any reason why the "src" topology shouldn't work for playback?

tlauda, 2019-03-19:
@lyakh @singalsu I haven't touched SRC copy, because Seppo said it's already not limited by copying period size. Maybe we should recheck it.

ranj063, 2019-03-19:
@lyakh the issue is still valid for SRC playback. @singalsu can you please check src copy too?

tlauda, 2019-03-21:
@lyakh I cannot reproduce it with the following pipeline: [host] - [src] - [vol] - [dai (48 kHz 24 bit)] and stream params: 8 kHz, 32 bit.
I'm using current HEAD: cd4633c53e14f722e3e97a17e24a75b07a45868b.
Could you share the full log and screenshot of recorded sine wave .

ranj063, 2019-03-21:
@lyakh #1107 was fixed with the recent changes on master. Can you please check if this issue still persists?

tlauda, 2019-03-21:
@ranj063 My fixes are only applicable to non-cAVS platforms. This APL issue is caused by something else.

singalsu, 2019-03-25:
SRC will need some updates to behave more like other components with timer based scheduling. It supports the circular component buffers but the copy function logic to decide how many frames to process must be updated. I will start soon with needed updates.

lgirdwood, 2019-03-25:
@singalsu scheduling time domain can come from a timer or DMA IRQ so SRC must be able to work with both. There is a PR merging that will also allow external time domains to be defined too (e.g. for ASRC)

singalsu, 2019-03-25:
Also I just confirmed I have this issue with UP2+HiFiBerry so I can test the fix. I've assigned this bug to myself.

lyakh, 2019-03-28:
I'm not sure why @xiulipan considers this to "not be an error," but I can now confirm his statement - I'm getting similar distorted "metallic" sound on MinnowBoard with rt5651 too now.

xiulipan, 2019-03-29:
@lyakh 
I mean there is no error log or error dmesg.

So you mean you have all thing run right, but get noise when play?

lyakh, 2019-03-29:
@xiulipan The noise can be eliminated be reducing the volume, @singalsu thinks, there might be an off-by-one error (in a volume configuration?) somewhere

xiulipan, 2019-03-29:
@lyakh 
Maybe the volume copy function need some check.

singalsu, 2019-03-29:
@tlauda I'm at the moment updating SRC work with current pipeline. It supports circular buffers already but the processing size is pre-determined.

singalsu, 2019-03-29:
@xiulipan @lyakh I've checked that volume nor SRC does not overflow. The overflow happens in PCM512x codec. It can be avoided by setting gain "Digital" to -1. With 0 dB the full scale peaks in music can do what's seen in this issue #619 that was filed by @juimonen.

tlauda, 2019-03-29:
@singalsu Processing size shouldn't be a problem as long as it isn't 0, because that will trigger fail in the following components e.g. volume. But since SRC can sometimes produce 0 samples, we need the mechanism for stopping pipeline processing in that situation. Returning PPL_STATUS_PATH_STOP from src_copy will handle the stopping part, but render ppl scheduled on DMA won't get rescheduled, since we haven't yet started the DAI. We need to carefully think about the right approach and implementation.

singalsu, 2019-03-29:
As long as DAI in playback doesn't starve (xrun), returning zero samples in other pipeline should not be an error. SRC can output 0 samples in complex 44.1 <-> 48 kHz family conversions especially in the beginning when the internal buffers are not filled enough.

When converting within 48 kHz family it can operate with constant number of frames every copy.

tlauda, 2019-03-29:
Yes, agree, but it doesn't make sense to perform copy on the components after SRC, if it produced 0 samples during preload phase. We would output zeroes anyway and also end up with mixed up DMA pointers as it was before. In the end it comes to the right implementation. I think that @bkokoszx is working on such pipeline flow, where components due to the different processing frames can produce even 0 bytes.

lyakh, 2019-03-29:
Looks like I was getting that corrupted sound with pulseaudio running and a few days old kernel with CONFIG_SND_SOC_SOF_DEBUG_FORCE_IPC_POSITION set. Without pulseaudio the audio quality seemed ok, although on maximum volume it was difficult to be sure. I then replaced the kernel with a newer one without that option set, and it was able to work properly also with pulseaudio enabled. However, then I tried using the SRC topology and this time I got no sound at all and I've got a DSP panic. But I have to double-check that, it's possible that I have some incompatible parts installed... And I'll also have to check a current kernel with "FORCE_IPC_POSITION" enabled, since IIRC it should be fixed by now.

lyakh, 2019-03-29:
I rebuilt both the kernel modules and the firmware, no panic this time, but the sound is bad. Attaching the result.
[corrupted-sound.tar.gz](https://github.com/thesofproject/sof/files/3022507/corrupted-sound.tar.gz)

singalsu, 2019-03-29:
@lyakh Yep, that's the issue I'm trying to fix now. Sounds the same in my tests.

singalsu, 2019-04-03:
PR #1219 fixes for me the playback quality with SRC.

singalsu, 2019-04-05:
@lyakh Merged now, can you please check there too.

closed 2019-04-08 by singalsu