thesofproject/sof #1165: Topology: quit building topology if any of the debug macros fail

ranj063, 2019-03-21:
there are some debug macros defined in m4/debug.m4 that help determine if the topology has valid arguments for some macros. In case of a failure, today we continue building the topology anyway. Quit building tplg and report the error instead using m4exit() in case of errors

kv2019i, 2019-04-05:
@ranj063 
I went through the debug.m4 macros today and upon closer look, the DEBUG macros do not check anything, but rather they add informative comments on how the values should match to those defined in driver code.

The check in DEBUG_PCM_ADD() only relates to direct input data, but this seems really painful to do in m4. I think a separate validity checker for the conf-file would be much better (e.g. a feature to raise error if duplicate PCM ids are added).

So with that (plus the upcoming eclipse plugin), I propose to close this as won't-fix.

ranj063, 2019-04-05:
@kv2019i sounds good. Thanks!

closed 2019-04-05 by ranj063