thesofproject/sof #1166: Component registration seems to violate checkpatch

cujomalainey, 2019-03-21:
I am attempting to write a new effect component and I am currently hitting a checkpatch error that I am wondering if I should ignore since there is a reason that the rest of the components don't follow it or if I should send a pull request to fix the other components.

Currently component registration is done as follows.
```
void sys_comp_dai_init(void);

void sys_comp_dai_init(void)
{
        comp_register(&comp_dai);
}

DECLARE_COMPONENT(sys_comp_dai_init);
```

When you try and submit a patch currently with this code you get
```
WARNING: externs should be avoided in .c files
#9: FILE: src/audio/dcblock.c:590:
+void sys_comp_dcblock_init(void);
```

Removing the declaration and using static fixes the error and the code runs fine. I was wondering if there is something about the memory map that static doesn't work well with across platforms or was this check introduced after the components were written.

Regardless of the answer to my question, I can fix anything that needs patching to help.

ranj063, 2019-03-21:
@cujomalainey checkpatch got accidentally removed when the shift to cmake happened so this was likely missed when the self registration was introduced.

cujomalainey, 2019-03-21:
Ok, I'll draft a PR.

abonislawski, 2019-03-22:
Yes, thats because checkpatch was accidentally disabled.

Anyway this checkpatch warning is incorrect with kernel coding style:
https://www.kernel.org/doc/html/v4.11/process/coding-style.html#functions
(EXPORT just after a function body)

Or maybe there is a need to remove this empty line between function body and DECLARE_COMPONENT?

jajanusz, 2019-03-22:
I think we should decide if we really want them to be static., then we should fix checkpatch to treat DECLARE_COMPONENT just like EXPORT_SYMBOL and then start fixing code according to these decisions. Some more info also there: https://github.com/thesofproject/sof/pull/1167#issuecomment-475581866

lgirdwood, 2019-03-22:
ok, path of least resistance it to use EXPORT_SYMBOL instead. Saves changing/testing checkpatch.

jajanusz, 2019-03-22:
> @jajanusz ok, path of least resistance it to use EXPORT_SYMBOL instead. Saves changing/testing checkpatch.

I think it'd be better to not have misleading name. Export sounds like something that is globally exported not just added to container section, where it's not even accessible it with some kind of identifier.

I checked checkpatch and it's not that bad. It does just 2 things for EXPORT_SYMBOL:
1. Adds exception which allows it to be used right after another declaration without blank line - fortunately for us it does it also for all macros starting with DECLARE, so we have it.
2. Checks if there is no blank line before exported struct and EXPORT_SYMBOL macro - we don't have this one, but I think it's not the big deal and I prefer to have better name instead of this check.

lgirdwood, 2019-03-22:
@jajanusz ok, and IIRC we are not really exporting the symbols but we are only exporting/declaring components. i.e. this can only be used with components and not other symbols. So lets go with checkpatch fixes :)

PR #1167 (2019-03-24): Fix checkpatch errors
The init functions are declared in a way that upsets checkpatch, this fixes those errors.

Fixes #1166
closed 2019-03-24 by lgirdwood