thesofproject/sof #1173: [WHL] pulseaudio have no sound

Jiangxinx, 2019-03-22:
Summary:
pulseaudio have no sound on WHL-hda platform. Sof-logger have no obvious error trace.

Step:
1.paplay Music/California_Dreaming_48k_16bit.wav

Dmesg:
```
[   15.344080] sof-audio-pci 0000:00:1f.3: PCM: stream 0 dir 0 DMA position 5328 DAI position 467352
[   15.344128] sof-audio-pci 0000:00:1f.3: PCM: stream 0 dir 0 DMA position 5328 DAI position 467352
[   15.344139] sof-audio-pci 0000:00:1f.3: PCM: stream 0 dir 0 DMA position 5328 DAI position 467352
[   15.370082] sof-audio-pci 0000:00:1f.3: ipc rx: 0x600a0000: GLB_STREAM_MSG: POSITION
[   15.370154] sof-audio-pci 0000:00:1f.3: posn mailbox: posn offset is 0xc104c
[   15.370160] sof-audio-pci 0000:00:1f.3: posn : host 0x9240 dai 0x1d0460 wall 0x715768a
[   15.370166] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x600a0000: GLB_STREAM_MSG: POSITION
[   15.381909] sof-audio-pci 0000:00:1f.3: ipc rx: 0x600a0007: GLB_STREAM_MSG: POSITION
[   15.381982] sof-audio-pci 0000:00:1f.3: posn mailbox: posn offset is 0xc1000
[   15.381987] sof-audio-pci 0000:00:1f.3: posn : host 0x9300 dai 0x1d0b78 wall 0x7175094
[   15.381992] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x600a0007: GLB_STREAM_MSG: POSITION
[   15.422768] sof-audio-pci 0000:00:1f.3: PCM: stream 0 dir 0 DMA position 9360 DAI position 475416
[   15.422798] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[   15.422819] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   15.423033] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   15.423057] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[   15.423130] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[   15.423138] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   15.423253] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   15.423430] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[   15.425182] sof-audio-pci 0000:00:1f.3: PCM: stream 0 dir 1 DMA position 9408 DAI position 475870
[   15.425212] sof-audio-pci 0000:00:1f.3: PCM: stream 0 dir 1 DMA position 9408 DAI position 475870
[   15.425223] sof-audio-pci 0000:00:1f.3: PCM: stream 0 dir 1 DMA position 9408 DAI position 475870
[   15.425244] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[   15.425260] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   15.425395] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   15.425416] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[   15.425463] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 1
[   15.425470] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   15.425593] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   15.425706] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 1
[   16.125938] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[   16.125967] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   17.955049] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   17.955235] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   17.955294] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[   17.955303] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   17.955311] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[   17.955356] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   17.955364] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000010 at  00000044
[   65.613373] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   65.613378] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[   65.613380] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   65.616331] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[   65.616335] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[   65.616337] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[   65.616339] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   65.616341] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[   65.616343] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[   65.616400] sof-audio-pci 0000:00:1f.3: loading firmware
[   65.616402] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[   65.616404] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   65.616406] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   65.616408] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[   65.616409] sof-audio-pci 0000:00:1f.3: booting DSP firmware
[   65.616484] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[   65.616484] sof-audio-pci 0000:00:1f.3: periods:1
[   65.617313] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[   65.617319] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[   65.617321] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[   65.617323] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[   65.618309] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[   65.618312] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010e0e successful
[   65.618327] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   65.688544] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
[   68.693861] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 timedout
[   68.693875] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -62
[   68.693910] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[   68.693918] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[   68.693927] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[   68.693934] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   68.693939] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware after resume -62
[   68.693966] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[   68.693969] sof-audio-pci 0000:00:1f.3: error: pcm open failed to resume -22
[   68.693975] sof-audio-pci 0000:00:1f.3: ASoC: can't open component 0000:00:1f.3: -22
[   68.693982]  HDA Analog: ASoC: failed to start FE -22
```

Env:
sof cml-003-drop-stable: 55ae67b
kernel sof-dev: 2293bc78
tplg: sof-hda-generic.tplg

Log:
[logger.log](https://github.com/thesofproject/sof/files/2995915/logger.log)

plbossart, 2019-03-22:
@Jiangxinx Can you clarify what this has to do with pulse audio? the boot issue should happen with aplay as well.

emilchudzik, 2019-03-25:
@Jiangxinx why did you use sof-hda-generic.tplg? did you get fw from cml-003 branch and topology from master branch?

You should use topology from the same branch as fw.

Please rebuild and try again.

tlauda, 2019-03-28:
@Jiangxinx It looks like problem with FW booting. Does it also reproduce with aplay? This message is weird: Connected to device alsa_output.pci-0000_00_1f.3-platform-**skl_hda_dsp_generic**.stereo-fallback (index: 0, suspended: yes).

Jiangxinx, 2019-03-28:
Update some details:  aplay is also failed after open  pulseaudio service.  By the way, this issue can not be reproduced after close pulseaudio service: 
```
sudo mv /usr/bin/pulseaudio /usr/bin/pulseaudio.bak
```
Env:
sof cml-003-drop-stable: 55ae67b
sof master: 742f83ba
kernel sof-dev: 03670025
tplg: sof-hda-generic.tplg & sof-hda-intel-common.tplg
HW: WHL (HD-A mode) with ALC700

Attach the dmesg info of system booting up after open  pulseaudio service :
```
[    4.530639] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi1 <-> iDisp1 Pin mapping ok
[    4.530648] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi2 <-> iDisp2 Pin mapping ok
[    4.530656] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi3 <-> iDisp3 Pin mapping ok
[    4.530664] skl_hda_dsp_generic skl_hda_dsp_generic: Analog Codec DAI <-> Analog CPU DAI mapping ok
[    4.530672] skl_hda_dsp_generic skl_hda_dsp_generic: Digital Codec DAI <-> Digital CPU DAI mapping ok
[    4.530690] sof-audio-pci 0000:00:1f.3: creating new PCM HDA Analog
[    4.530692] sof-audio-pci 0000:00:1f.3: spcm: allocate Passthrough Playback 0 playback DMA buffer size 0x10000 max 0x10000
[    4.530727] sof-audio-pci 0000:00:1f.3: spcm: allocate Passthrough Capture 0 capture DMA buffer size 0x10000 max 0x10000
[    4.530733] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDA Analog 0 mapping ok
[    4.530746] sof-audio-pci 0000:00:1f.3: creating new PCM HDA Digital
[    4.530748] sof-audio-pci 0000:00:1f.3: spcm: allocate Passthrough Playback 1 playback DMA buffer size 0x10000 max 0x10000
[    4.530754] sof-audio-pci 0000:00:1f.3: spcm: allocate Passthrough Capture 1 capture DMA buffer size 0x10000 max 0x10000
[    4.530758] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDA Digital 1 mapping ok
[    4.530770] sof-audio-pci 0000:00:1f.3: creating new PCM HDMI1
[    4.530772] sof-audio-pci 0000:00:1f.3: spcm: allocate Passthrough Playback 3 playback DMA buffer size 0x10000 max 0x10000
[    4.530777] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI1 3 mapping ok
[    4.530787] sof-audio-pci 0000:00:1f.3: creating new PCM HDMI2
[    4.530788] sof-audio-pci 0000:00:1f.3: spcm: allocate Passthrough Playback 4 playback DMA buffer size 0x10000 max 0x10000
[    4.530794] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI2 4 mapping ok
[    4.530803] sof-audio-pci 0000:00:1f.3: creating new PCM HDMI3
[    4.530805] sof-audio-pci 0000:00:1f.3: spcm: allocate Passthrough Playback 5 playback DMA buffer size 0x10000 max 0x10000
[    4.530811] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI3 5 mapping ok
[    4.530844] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp3 Tx
[    4.530848] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp3 Tx -> direct -> hifi3
[    4.530851] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no sink widget found for iDisp3 Tx
[    4.530852] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp3_out -> direct -> iDisp3 Tx
[    4.530855] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp2 Tx
[    4.530857] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp2 Tx -> direct -> hifi2
[    4.530859] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no sink widget found for iDisp2 Tx
[    4.530861] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp2_out -> direct -> iDisp2 Tx
[    4.530863] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for iDisp1 Tx
[    4.530865] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp1 Tx -> direct -> hifi1
[    4.530867] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no sink widget found for iDisp1 Tx
[    4.530869] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route iDisp1_out -> direct -> iDisp1 Tx
[    4.530875] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for Analog CPU Playback
[    4.530876] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route Analog CPU Playback -> direct -> Analog Codec Playback
[    4.530879] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no sink widget found for Analog CPU Playback
[    4.530881] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route codec0_out -> direct -> Analog CPU Playback
[    4.530883] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for Digital CPU Playback
[    4.530885] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route Digital CPU Playback -> direct -> Digital Codec Playback
[    4.530888] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no sink widget found for Digital CPU Playback
[    4.530889] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route codec1_out -> direct -> Digital CPU Playback
[    4.530892] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no source widget found for Alt Analog CPU Playback
[    4.530894] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route Alt Analog CPU Playback -> direct -> Alt Analog Codec Playback
[    4.530897] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: no sink widget found for Alt Analog CPU Playback
[    4.530898] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Failed to add route codec2_out -> direct -> Alt Analog CPU Playback
[... cut ...]
[    4.531244] input: sof-skl_hda_card Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input9
[    4.531278] input: sof-skl_hda_card HDMI/DP, pcm=7 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input10
[    4.531316] input: sof-skl_hda_card HDMI/DP, pcm=8 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input11
[    4.531349] input: sof-skl_hda_card HDMI/DP, pcm=9 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input12
```

lgirdwood, 2019-03-28:
@Jiangxinx usually pulseaudio will "probe" the capabilities of each PCM device and this is where most of any breakage will occur. So I would really expect to see IPC timeout or FW/driver crash here, but in this case it appears PM/boot related.  Can you confirm this is only an issue with WHL and pulseaudio works with the same topology on other platforms.

RanderWang, 2019-03-29:
According to Michal: cml-0003 is not for HDA, please validate it on master.
I tested on master and got sound, but the sound is not enjoyable.

RanderWang, 2019-03-29:
sorry，  I find that I disable PM on WHL.

mengdonglin, 2019-03-29:
@Jiangxinx @Jiangxinx please help to use master branch to test WHL RVP in HD-A mode, and update the status.

Jiangxinx, 2019-03-29:
I have checked the master branch on daily version, pulseaudio still have no sound and aplay also failed after open pulseaudio service.
Env:
sof master: 742f83b
kernel sof-dev: e29122d6
tplg: sof-hda-generic.tplg
HW: WHL (HD-A mode) with ALC700

emilchudzik, 2019-03-29:
I have the same as @Jiangxinx on linux machine when I start aplay:
[image]

On our python tests in windows HDA works without issues. (FW: master/[a10e69d2](https://github.com/thesofproject/sof/commit/a10e69d2))

linux Env:
FW: master/[a10e69d2](https://github.com/thesofproject/sof/commit/a10e69d2)
tplg: master/[a10e69d2](https://github.com/thesofproject/sof/commit/a10e69d2) -> sof-hda-generic.tplg 
kernel: sof-dev/[e29122d](https://github.com/thesofproject/linux/commit/e29122d)
HW: WHL (HD-A mode) with ALC700

Dmesg log during booting seems to be ok, although I have some erros at the beggining - I don't know if it is important:
[image]

emilchudzik, 2019-03-29:
This is my steps to build kernel:
- fresh kconfig repo
- fresh linux repo -> commit e29122d/sof-dev
- make defconfig
- then merge:
bash ./scripts/kconfig/merge_config.sh .config $KCONFIG_PATH/base-defconfig $KCONFIG_PATH/sof-defconfig $KCONFIG_PATH/hdaudio-codecs-defconfig
- and build: make bindeb-pkg -j32

Did you have any additional patches or steps? Or maybe additional thinks on linux machine to do - W/A or stop/start any services?

If yes please write your detailes steps dor building and configuring.

RanderWang, 2019-04-01:
@emilchudzik  this is another bug,  sof can't resume with pulse audio enabled.

lgirdwood, 2019-04-01:
@RanderWang can you confirm this issue is seen on other platforms like APL etc ? If so it's a 1.3 blocker.

ranj063, 2019-04-06:
> @emilchudzik this is another bug, sof can't resume with pulse audio enabled.

@RanderWang @emilchudzik https://github.com/thesofproject/linux/pull/795 fixes this problem

emilchudzik, 2019-04-08:
@ranj063 did you check it on the newest kernel? Currently the blokcer is: #767. 
Which commit of linux repo is working on your side with your patch?

RanderWang, 2019-04-08:
> @RanderWang can you confirm this issue is seen on other platforms like APL etc ? If so it's a 1.3 blocker.

As I know, there is no such issue on APL

emilchudzik, 2019-04-15:
Please check my comments:
https://github.com/thesofproject/sof/issues/1254#issuecomment-483200302

Jiangxinx, 2019-04-16:
This issue can also be reproduced on BYT rt5651 platform.

***Env***
sof master: 1b15fad8
kernel sof-dev: 2de29fa(compiled with PR #823 )
tplg: sof-byt-rt5651.tplg

emilchudzik, 2019-04-16:
Plese check my comments here: https://github.com/thesofproject/sof/issues/1263#issuecomment-483551695

xiulipan, 2019-04-16:
@Jiangxinx 
If this issue is still with WHL HDA. If not please close this issue and create a new issue for BYT.

tlauda, 2019-04-19:
@Jiangxinx Still reproducible?

mengdonglin, 2019-04-22:
@tlauda @Jiangxinx Rander is working on this . This is a kernel issue.

stevyan, 2019-04-22:
**Summary**
This issue cannot be reproduced on BYT rt5651 platform.

**Test Recipe**
date: 2019-04-22
sof master: 0c932613
kernel sof-dev: 90860434
tplg: sof-byt-rt5651.tplg

Jiangxinx, 2019-04-22:
This issue can not be reproduced on daily test.

***Environment***
Kernel Repo: https://github.com/RanderWang/linux/commits/cml-validation-20190422
Made from: RanderWang/linux@9086043 [thesofproject/linux: topic/sof-dev]
+RanderWang/linux@c105b1b
+RanderWang/linux@ffecddc8
+RanderWang/linux@3b917aa
+RanderWang/linux@6f928a3
+RanderWang/linux@a916d2d
+RanderWang/linux@694599b
+RanderWang/linux@a43379d
+RanderWang/linux@d36ed40
+RanderWang/linux@9b3bd71
+RanderWang/linux@e852311
+RanderWang/linux@8f30111
+RanderWang/linux@95cd082

Firmware Repo :https://github.com/bardliao/sof/tree/whl-daliy-20190422
Made from: bardliao@0c93261
+bardliao@35fbf79
+bardliao@3f91567

Topology: Same as above SOF repo

lgirdwood, 2019-04-23:
@Jiangxinx please close once patches merged upstream into master and sof-dev.

emilchudzik, 2019-04-29:
@Jiangxinx @mengdonglin please close. issue currently doesn't exists on: kernel:cml-validation-20190425 and FW: master/e4a9f86

RanderWang, 2019-04-30:
@emilchudzik  It is a tmp solution included in the cml-validation-20190425

wenqingfu, 2019-05-08:
will close after thesofproject/linux#895 is merged

closed 2019-05-29 by emilchudzik