thesofproject/sof #1179: [WHL] Noise in the few seconds since the start of playback on WHL-hda platform.

Jiangxinx, 2019-03-25:
Summary:
Aplay have noise in the few seconds since the start of playback on WHL-hda platform. 
Dmesg and Sof-logger have no obvious error trace.

Step:
1.aplay -Dhw:0,0 -r 48000 -c 2 -f s16_le Music/sine_48k_16bit_10s_stereo_l5_r2.wav -vv -i

Logger:
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      2          DMA            116780.104167    116780.101562 intel/cavs/hda-dma.c:480 	hda-dmac: 4 channel 0 -> get
    0      2          DMA            116863.541667        83.437500 intel/cavs/hda-dma.c:703 	hda-dmac: 4 channel 0 -> config
    0      2          DMA            116915.416667        51.875000 intel/cavs/hda-dma.c:549 	hda-dmac: 4 channel 0 -> start
    0      2          DMA            116918.906250         3.489583 intel/cavs/hda-dma.c:396 	hda-dmac: 4 channel 0 -> enable
    0      2          IPC            117227.552083       308.645844    src/ipc/handler.c:917 	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC            117230.729167         3.177083    src/ipc/handler.c:920 	ipc: pipe 1 comp 0 -> new (type 1)
    0      2         HOST            117236.041667         5.312500     src/audio/host.c:344 	host_new()
    0      2          DMA            117245.104167         9.062500 intel/cavs/hda-dma.c:839 	hda-dmac :5 -> probe
    0      2          DMA            117261.562500        16.458334        src/lib/dma.c:142 	dma_get() ID 5 sref = 1 busy channels 0
    0      2          IPC            117511.093750       249.531250    src/ipc/handler.c:917 	ipc: hdr 0x30010000 tx (84) > rx (28)
    0      2          IPC            117514.166667         3.072917    src/ipc/handler.c:920 	ipc: pipe 1 comp 1 -> new (type 5)
    0      2       VOLUME            117518.750000         4.583333   src/audio/volume.c:178 	volume_new()
    0      2          IPC            117757.708333       238.958328    src/ipc/handler.c:950 	ipc: pipe 1 buffer 2 -> new (0x300 bytes)
    0      2       BUFFER            117761.822917         4.114583   src/audio/buffer.c:53  	buffer_new()
    0      2          IPC            117928.177083       166.354172    src/ipc/handler.c:950 	ipc: pipe 1 buffer 3 -> new (0x300 bytes)
    0      2       BUFFER            117932.447917         4.270833   src/audio/buffer.c:53  	buffer_new()
    0      2          IPC            118096.510417       164.062500    src/ipc/handler.c:917 	ipc: hdr 0x30010000 tx (80) > rx (28)
    0      2          IPC            118099.843750         3.333333    src/ipc/handler.c:920 	ipc: pipe 1 comp 4 -> new (type 2)
    0      2          DAI            118104.791667         4.947917      src/audio/dai.c:197 	dai_new()
    0      2          DAI            118113.593750         8.802083        src/lib/dai.c:87  	dai_get(), d = 0xbe05c0b0, sref = 1
    0      2          DMA            118117.708333         4.114583 intel/cavs/hda-dma.c:839 	hda-dmac :7 -> probe
    0      2          DMA            118134.114583        16.406250        src/lib/dma.c:142 	dma_get() ID 7 sref = 1 busy channels 0
    0      2          IPC            118383.750000       249.635422    src/ipc/handler.c:978 	ipc: pipe 1 -> new
    0      2         PIPE            118390.677083         6.927083 src/audio/pipeline.c:71  	pipeline_new()
    0      2          IPC            118550.885417       160.208328    src/ipc/handler.c:660 	ipc: pm core mask 0x1 -> enable
    0      2          IPC            118667.708333       116.822914    src/ipc/handler.c:917 	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC            118670.833333         3.125000    src/ipc/handler.c:920 	ipc: pipe 2 comp 7 -> new (type 1)
    0      2         HOST            118676.927083         6.093750     src/audio/host.c:344 	host_new()
    0      2          DMA            118686.250000         9.322917        src/lib/dma.c:142 	dma_get() ID 4 sref = 2 busy channels 1
    0      2          IPC            118817.968750       131.718750    src/ipc/handler.c:917 	ipc: hdr 0x30010000 tx (84) > rx (28)
    0      2          IPC            118821.041667         3.072917    src/ipc/handler.c:920 	ipc: pipe 2 comp 8 -> new (type 5)
    0      2       VOLUME            118827.083333         6.041667   src/audio/volume.c:178 	volume_new()
    0      2          IPC            118947.604167       120.520836    src/ipc/handler.c:950 	ipc: pipe 2 buffer 9 -> new (0x300 bytes)
    0      2       BUFFER            118953.541667         5.937500   src/audio/buffer.c:53  	buffer_new()
    0      2          IPC            119092.083333       138.541672    src/ipc/handler.c:950 	ipc: pipe 2 buffer 10 -> new (0x300 bytes)
    0      2       BUFFER            119098.333333         6.250000   src/audio/buffer.c:53  	buffer_new()
    0      2          IPC            119248.072917       149.739578    src/ipc/handler.c:917 	ipc: hdr 0x30010000 tx (80) > rx (28)
    0      2          IPC            119251.197917         3.125000    src/ipc/handler.c:920 	ipc: pipe 2 comp 11 -> new (type 2)
    0      2          DAI            119257.864583         6.666667      src/audio/dai.c:197 	dai_new()
    0      2          DAI            119266.770833         8.906250        src/lib/dai.c:87  	dai_get(), d = 0xbe05c0fc, sref = 1
    0      2          DMA            119270.572917         3.802083 intel/cavs/hda-dma.c:839 	hda-dmac :6 -> probe
    0      2          DMA            119286.927083        16.354166        src/lib/dma.c:142 	dma_get() ID 6 sref = 1 busy channels 0
    0      2          IPC            119436.250000       149.322922    src/ipc/handler.c:978 	ipc: pipe 2 -> new
    0      2         PIPE            119446.562500        10.312500 src/audio/pipeline.c:71  	pipeline_new()
    0      2          IPC            119566.302083       119.739586    src/ipc/handler.c:660 	ipc: pm core mask 0x1 -> enable
    0      2          IPC            119680.104167       113.802086    src/ipc/handler.c:917 	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC            119683.229167         3.125000    src/ipc/handler.c:920 	ipc: pipe 3 comp 14 -> new (type 1)
    0      2         HOST            119691.197917         7.968750     src/audio/host.c:344 	host_new()
    0      2          DMA            119700.156250         8.958333        src/lib/dma.c:142 	dma_get() ID 5 sref = 2 busy channels 0
    0      2          IPC            119815.781250       115.625000    src/ipc/handler.c:917 	ipc: hdr 0x30010000 tx (84) > rx (28)
    0      2          IPC            119819.010417         3.229167    src/ipc/handler.c:920 	ipc: pipe 3 comp 15 -> new (type 5)
    0      2       VOLUME            119827.031250         8.020833   src/audio/volume.c:178 	volume_new()
    0      2          IPC            119950.989583       123.958336    src/ipc/handler.c:950 	ipc: pipe 3 buffer 16 -> new (0x300 bytes)
    0      2       BUFFER            119958.697917         7.708333   src/audio/buffer.c:53  	buffer_new()
    0      2          IPC            120097.708333       139.010422    src/ipc/handler.c:950 	ipc: pipe 3 buffer 17 -> new (0x300 bytes)
    0      2       BUFFER            120105.677083         7.968750   src/audio/buffer.c:53  	buffer_new()
    0      2          IPC            120280.885417       175.208328    src/ipc/handler.c:917 	ipc: hdr 0x30010000 tx (80) > rx (28)
    0      2          IPC            120284.010417         3.125000    src/ipc/handler.c:920 	ipc: pipe 3 comp 18 -> new (type 2)
    0      2          DAI            120292.656250         8.645833      src/audio/dai.c:197 	dai_new()
    0      2          DAI            120301.250000         8.593750        src/lib/dai.c:87  	dai_get(), d = 0xbe05c148, sref = 1
[... 266 lines cut ...]
    0      2          DMA           1678595.000000      1011.614563 intel/cavs/hda-dma.c:549 	hda-dmac: 7 channel 0 -> start
    0      2          DMA           1678598.802083         3.802083 intel/cavs/hda-dma.c:396 	hda-dmac: 7 channel 0 -> enable
    0      2          IPC          14274148.958333  12595550.000000    src/ipc/handler.c:464 	ipc: comp 0 -> trigger cmd 0x50000
    0      2         PIPE 1.5      14274152.864583         3.906250 src/audio/pipeline.c:526 	pipeline_trigger()
    0      2         HOST          14274156.406250         3.541667     src/audio/host.c:281 	host_trigger()
    0      2          DMA          14274159.843750         3.437500 intel/cavs/hda-dma.c:645 	hda-dmac: 5 channel 0 -> stop
[... cut ...]
    0      2         HOST          14658493.906250         3.333333     src/audio/host.c:681 	host_reset()
    0      2          DMA          14658497.239583         3.333333 intel/cavs/hda-dma.c:645 	hda-dmac: 5 channel 0 -> stop
    0      2       VOLUME          14658503.333333         6.093750   src/audio/volume.c:614 	volume_reset()
```

Env:
sof cml-003-drop-stable: 55ae67b
kernel sof-dev: e127788a
tplg: sof-hda-generic.tplg

Log file:
[dmesg.log](https://github.com/thesofproject/sof/files/3002694/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3002696/logger.log)

tlauda, 2019-03-25:
Can you play sine wave and show us how does the noise look like?
Short recording or screenshot from Audacity will be OK.

mengdonglin, 2019-03-28:
@tlauda Should we use https://github.com/thesofproject/sof/blob/cml-003-drop-stable/tools/topology/sof-hda-intel-common.m4 for cml-003? And since the timer driven scheduler is also merged by master branch, shall we revise all topology that involve HDA pipelines?

Now cml-003 is used to for both ChromeOS (WHL RVP with I2S codec ALC5682) and Ubuntu (WHL RVP with on board HD-A codec ALC700).

RanderWang, 2019-03-28:
sof-hda-intel-common.m4  is renamed to sof-hda-generic.4 on master.

Jiangxinx, 2019-03-28:
I have recorded this noise by using sina wave, and there are four time frames.
1 2.59s:
[image]

2 2.67s:
[image]

3 2.75s:
[image]

4 2.84s:
[image]

sina wave:
[image]

sina file:
[sine_noise.zip](https://github.com/thesofproject/sof/files/3016390/sine_noise.zip)

tlauda, 2019-03-28:
@Jiangxinx This whole sine wave looks strange and not correct, not only those 4 places. So the playback is OK at the beginning and just go like this after ~2 seconds? Do you have 100% reproduction rate?
@emilchudzik Do you see something similar on our side?

lgirdwood, 2019-03-28:
@bardliao could this be codec related i.e. PLL not locking ?

emilchudzik, 2019-03-28:
HDA tests on our python tests works well: FW cml-003-drop-stable: 55ae67b

On linux (kernel and FW from cml003 release) and topology:
sof-hda-intel-common.m4 -> /lib/firmware/intel/sof-tplg/sof-hda-generic.tplg
- loading is ok (no error in dmesg)

- playback using aplay return errors:
```
[   40.493927] sof-audio-pci 0000:00:1f.3: error: rx list empty but received 0x0
[   40.493938] sof-audio-pci 0000:00:1f.3: error: can't find message header 0x0
[   43.495860] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 timedout
[   43.495875] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -62
[   43.495917] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[   43.495925] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[   43.495934] sof-audio-pci 0000:00:1f.3: unknown ROM status value 05000001
[   43.495940] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   43.495949] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware after resume -62
[   43.495961] sof-audio-pci 0000:00:1f.3: error: volume get failed to resume -62
[   49.902963] sof-audio-pci 0000:00:1f.3: error: volume get failed to resume -22
[   66.646600] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[   66.646602] sof-audio-pci 0000:00:1f.3: error: pcm open failed to resume -22
[   66.646607] sof-audio-pci 0000:00:1f.3: ASoC: can't open component 0000:00:1f.3: -22
[   66.646610]  HDA Analog: ASoC: failed to start FE -22
[   76.682404] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[   76.682405] sof-audio-pci 0000:00:1f.3: error: pcm open failed to resume -22
[   76.682408] sof-audio-pci 0000:00:1f.3: ASoC: can't open component 0000:00:1f.3: -22
[   76.682411]  HDA Analog: ASoC: failed to start FE -22

```

emilchudzik, 2019-03-28:
Sorry, closed by mistake, just reopen

RanderWang, 2019-03-29:
> HDA tests on our python tests works well: FW cml-003-drop-stable: [55ae67b](https://github.com/thesofproject/sof/commit/55ae67b5c06a0349dcaab20958c299a090e12c1e)
>
> On linux (kernel and FW from cml003 [...]

please don't try it on cml-003 now. Michal told me this is not for HDA, tested HDA on master.

RanderWang, 2019-03-29:
@Jiangxinx  can you validate it on master ? I tested it and got no noise

Jiangxinx, 2019-03-29:
@RanderWang I retested this issue on my side by using master branch,  the noise still exist.
Step:
1.aplay -Dhw:0,0 -r 48000 -c 2 -f s16_le ~/Music/sine_48k_16bit_10s_stereo_l5_r2.wav -vv -i

Env:
sof master: 742f83ba
kernel sof-dev: e29122d6
tplg: sof-hda-generic.tplg

bardliao, 2019-03-29:
> @bardliao could this be codec related i.e. PLL not locking ?

@lgirdwood Yes, it could be. Can we use the same DUT and test it with legacy HDA driver first?

emilchudzik, 2019-03-29:
I have the same symptoms like in: https://github.com/thesofproject/sof/issues/1173

I can't play beacuase of this erros: [image]

linux Env:
FW: master/[a10e69d2](https://github.com/thesofproject/sof/commit/a10e69d2)
tplg: master/[a10e69d2](https://github.com/thesofproject/sof/commit/a10e69d2) -> sof-hda-generic.tplg 
kernel: sof-dev/[e29122d](https://github.com/thesofproject/linux/commit/e29122d)
HW: WHL (HD-A mode) with ALC700

On our python tests in windows HDA works without issues. (FW: master/[a10e69d2](https://github.com/thesofproject/sof/commit/a10e69d2))

emilchudzik, 2019-03-29:
I cannot reproduce both issues: #1173 and #1179.
Do you have any additional patches for kernel?
Or do you have any additional steps for your WHL linux or BIOS?

In this bug you have issue with strange noise, but on the same configuration on #1173 you have no audio.
Both issues I cannot reproduce due to unable to play at all.

emilchudzik, 2019-04-02:
I suggest close this issue due unable to reproduce from submitter steps.
In my opinion this is issue whioch blocked playback on WHLwith HDA: https://github.com/thesofproject/linux/issues/774

lgirdwood, 2019-04-02:
@Jiangxinx can you try and use @RanderWang asound.state file (see man alsactl for info). This will guarantee you have the same codec settings. @RanderWang can you try the same file from @Jiangxinx config and see if you can reproduce.

Jiangxinx, 2019-04-04:
I have checked with @RanderWang . His Env can also reproduce this issue.The reason why he didn’t hear the noise before was that the music file he used was quite in the first few seconds.

bardliao, 2019-04-04:
@Jiangxinx Do you mean the noise will only happen in the first few seconds? Sorry I didn't get why @RanderWang can't reproduce it with a sound file which is quiet in the first few seconds.

Jiangxinx, 2019-04-04:
@bardliao Yes, the noise will only happen in the first few seconds. Then aplay will be smooth.

lgirdwood, 2019-04-04:
@Jiangxinx @RanderWang, I want to check the HDA link/code is initializing correctly. Can you

1) In a terminal run arecord from HDA codec (arecord -f dat > /dev/null)
2) wait 2 seconds
3) run aplay at the same time in another terminal. Does the startup noise still occur ?

running arecord should enable the HDA link and power on HDA codec so if 3 works then it's likely we need some investigation around codec init.

stevyan, 2019-04-08:
[sine_48k_stereo_440Hz.zip](https://github.com/thesofproject/sof/files/3052511/sine_48k_stereo_440.zip)

stevyan, 2019-04-08:
@lgirdwood I've attempted the test steps you provided, but the startup noise still can be heard at the begin of 2~3s.

emilchudzik, 2019-04-15:
Please check my comments:
https://github.com/thesofproject/sof/issues/1254#issuecomment-483200302.

When pulseaudio is disabled - I have clear sound, there is no any noises.

**Environment:**
Platform: WHL RVP - HDA mode
**Kernel:** 
2de29fae6 [thesofproject/linux: topic/sof-dev]
+d04ccf494 [RanderWang/linux: cml-validation-20190415]
+bfb0729dd [RanderWang/linux: cml-validation-20190415]
+8a2b13a81 [RanderWang/linux: cml-validation-20190415]
+63ceb8f68 [RanderWang/linux: cml-validation-20190415]
+db89510d0 [RanderWang/linux: cml-validation-20190415]

**Firmware:** 00bbb113d [thesofproject/sof: master]
**Topology:** 00bbb11 [thesofproject/sof: master] -> sof-hda-generic.tplg

emilchudzik, 2019-04-16:
With below configuratin I have clear sound when pulseaudio is enabled or not.

**Environment:**
Platform: WHL RVP - HDA mode
**Kernel:** 
2de29fae6 [thesofproject/linux: topic/sof-dev]
+d04ccf494 [RanderWang/linux: cml-validation-20190415]
+bfb0729dd [RanderWang/linux: cml-validation-20190415]
+8a2b13a81 [RanderWang/linux: cml-validation-20190415]
+63ceb8f68 [RanderWang/linux: cml-validation-20190415]
+db89510d0 [RanderWang/linux: cml-validation-20190415]

**Firmware:** 1b15fad [thesofproject/sof: master] + https://github.com/thesofproject/sof/pull/1269/commits/624c18935589b4f87bc4a058959ef5d53ac26882
**Topology:** 1b15fad [thesofproject/sof: master] -> sof-hda-generic.tplg

lgirdwood, 2019-04-16:
@bkokoszx looks like https://github.com/thesofproject/sof/commit/624c18935589b4f87bc4a058959ef5d53ac26882 is fixing quite a few issues, can you add this information to the PR and **why** this fixes the issues.

tlauda, 2019-04-19:
@Jiangxinx Still reproducible?

lgirdwood, 2019-04-19:
@Jiangxinx if it's still open can you attach pulseaudio logs of playback and init and also run
```
aplay -f dat -v --test-position /dev/zero
```
This will provide position information of aplay and pulseaudio and we should see any issues with positions that would cause an issue in userspace.

Jiangxinx, 2019-04-22:
This issue can not be reprodused on daily test.

***Environment***
Kernel Repo: https://github.com/RanderWang/linux/commits/cml-validation-20190422
Made from: RanderWang/linux@9086043 [thesofproject/linux: topic/sof-dev]
+RanderWang/linux@c105b1b
+RanderWang/linux@ffecddc8
+RanderWang/linux@3b917aa
+RanderWang/linux@6f928a3
+RanderWang/linux@a916d2d
+RanderWang/linux@694599b
+RanderWang/linux@a43379d
+RanderWang/linux@d36ed40
+RanderWang/linux@9b3bd71
+RanderWang/linux@e852311
+RanderWang/linux@8f30111
+RanderWang/linux@95cd082

Firmware Repo :https://github.com/bardliao/sof/tree/whl-daliy-20190422
Made from: bardliao@0c93261
+bardliao@35fbf79
+bardliao@3f91567

Topology: Same as above SOF repo

lgirdwood, 2019-04-23:
@Jiangxinx ok great, I assume you will close when patches are merged ?

emilchudzik, 2019-04-29:
@Jiangxinx @mengdonglin please verify issue and close if there is no repro.

mengdonglin, 2019-05-06:
Closing this issue because FW fix has been merged. Will track driver PR status by #1173

closed 2019-05-06 by mengdonglin