thesofproject/sof #1185: ipc: support compound messages

lgirdwood, 2019-03-26:
Firmware should be able to support compound IPC whereby multiple IPC commands are send as one (using one doorbell IRQ). This allows a speed up in initialization and reduced latency switching use cases.

We need to categorize which types we can categorize and how we handle failure for each. I can think of :-

1) Topology construction. Whole topology is rewound on failure on any component. e.g. if we send 20 cmds and cmd 7 fails then we rewind and delete cmds 0 - 6.

2) kcontrol updates. For mixers and volumes, not sure is applicable to binary for size constraints. I would be tempted to keep going on any unlikely failure (since all volume/mute/enum values should be within bounds).

3) Stream trigger ops. e.g. for simultaneous stream ops. Any failure here would reset all streams. 

The tricky part is knowing when to enable compound mode for 2, we could speculatively enable it for any kcontrol update and only send when we timeout of further kcontrol IO. Enabling for 1 is simple enough and 3 could be based on linked PCMs in ALSA.

lgirdwood, 2019-04-18:
@juimonen is this merged now and can be closed ?

lgirdwood, 2020-08-18:
Not needed atm, so move out to v1.7

lgirdwood, 2021-01-12:
Still not needed - move to v1.9

kv2019i, 2023-02-15:
Parking on TBD for the moment, this is still a valid item.