thesofproject/sof #1190: PCM512x EQ pipeline playback failure

ranj063, 2019-03-26:
Unable to start playback with the EQ pipeline

kernel: sof-dev: e127788a6
FW: master: 9890d2b782

: SET_VALUE
[ 1529.683838] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[ 1529.683851] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[ 1529.683907] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[ 1529.697421] pcm512x i2c-104C5122:00: No SCLK, using BCLK: -2
[ 1529.700256] sof-audio-pci 0000:00:0e.0: pcm: open stream 0 dir 0
[ 1529.700266] sof-audio-pci 0000:00:0e.0: period min 192 max 16384 bytes
[ 1529.700271] sof-audio-pci 0000:00:0e.0: period count 2 max 16
[ 1529.700275] sof-audio-pci 0000:00:0e.0: buffer max 65536 bytes
[ 1529.830663] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[ 1529.830667] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[ 1529.830671] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[ 1529.830673] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[ 1529.830683] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[ 1529.830685] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[ 1529.830691] sof-audio-pci 0000:00:0e.0: pcm: hw params stream 0 dir 0
[ 1529.830695] sof-audio-pci 0000:00:0e.0: generating page table for 00000000a55c78cb size 0xff00 pages 16
[ 1529.830716] sof-audio-pci 0000:00:0e.0: period_bytes:0xfc0
[ 1529.830718] sof-audio-pci 0000:00:0e.0: periods:16
[ 1529.830728] sof-audio-pci 0000:00:0e.0: stream_tag 1
[ 1529.830735] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1529.830918] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[ 1529.830927] sof-audio-pci 0000:00:0e.0: error: hw params ipc failed for stream 1
[ 1529.830930] sof-audio-pci 0000:00:0e.0: ASoC: 0000:00:0e.0 hw params failed: -22
[ 1529.830933]  Port5: ASoC: hw_params FE failed -22
[ 1529.832824] sof-audio-pci 0000:00:0e.0: pcm: free stream 0 dir 0
[ 1529.832834] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1529.833018] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1529.834253] sof-audio-pci 0000:00:0e.0: pcm: close stream 0 dir 0
[ 1530.681520] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1530.681535] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1532.030625] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1532.030899] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1532.030925] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
[ 1532.032013] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
[ 1532.032028] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 3
[ 1532.032058] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000010 at  00000044

Sof-logger trace:
    0      2       VOLUME             48410.989583         4.010417   src/audio/volume.c:327 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 1
    0      2       VOLUME             48414.635417         3.645833   src/audio/volume.c:332 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 0, value = 65536
    0      2       VOLUME             48418.125000         3.489583   src/audio/volume.c:332 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 1, value = 65536
    0      2          IPC             48672.708333       254.583328    src/ipc/handler.c:643 	ipc: pm -> restore
    0      2          IPC            195559.479167    146886.765625    src/ipc/handler.c:239 	ipc: comp 0 -> params
    0      2         PIPE 1.9        195565.000000         5.520833 src/audio/pipeline.c:320 	pipeline_params()
    0      2         HOST            195569.531250         4.531250     src/audio/host.c:508 	host_params()
    0      2          DMA            195577.187500         7.656250 intel/cavs/hda-dma.c:480 	hda-dmac: 5 channel 0 -> get
    0      2          DMA            195581.718750         4.531250 intel/cavs/hda-dma.c:703 	hda-dmac: 5 channel 0 -> config
    0      2       EQ_IIR            195587.708333         5.989583   src/audio/eq_iir.c:570 	eq_iir_params()
    0      2       EQ_FIR            195592.187500         4.479167   src/audio/eq_fir.c:468 	eq_fir_params()
    0      1       EQ_FIR            195595.885417         3.697917   src/audio/eq_fir.c:483 	eq_fir_params() error: buffer_set_size() failed
    0      1         PIPE            195601.458333         5.572917 src/audio/pipeline.c:329 	pipeline_params() error: ret = -22, host->comp.id = 0
    0      1          IPC            195607.604167         6.145833    src/ipc/handler.c:316 	ipc: pipe 1 comp 0 params failed -22
    0      2         PIPE 1.9        195613.906250         6.302083 src/audio/pipeline.c:576 	pipeline_reset()
    0      2         HOST            195617.812500         3.906250     src/audio/host.c:704 	host_reset()
    0      2          DMA            195621.666667         3.854167 intel/cavs/hda-dma.c:645 	hda-dmac: 5 channel 0 -> stop
    0      2       EQ_IIR            195628.645833         6.979167   src/audio/eq_iir.c:870 	eq_iir_reset()
    0      2         COMP            195633.333333         4.687500 rc/audio/component.c:136 	comp_set_state(), state already set to 1
    0      2       EQ_FIR            195637.187500         3.854167   src/audio/eq_fir.c:779 	eq_fir_reset()
    0      2         COMP            195641.822917         4.635417 rc/audio/component.c:136 	comp_set_state(), state already set to 1
    0      2       VOLUME            195645.572917         3.750000   src/audio/volume.c:621 	volume_reset()
    0      2         COMP            195649.218750         3.645833 rc/audio/component.c:136 	comp_set_state(), state already set to 1
    0      2          DAI 1.8        195652.968750         3.750000      src/audio/dai.c:521 	dai_reset()
    0      2         COMP            195656.822917         3.854167 rc/audio/component.c:136 	comp_set_state(), state already set to 1
    0      2          IPC            197658.229167      2001.406250    src/ipc/handler.c:367 	ipc: comp 0 -> free
    0      2         PIPE 1.9        197663.229167         5.000000 src/audio/pipeline.c:576 	pipeline_reset()
    0      2         HOST            197667.187500         3.958333     src/audio/host.c:704 	host_reset()
    0      1          DMA            197670.781250         3.593750 intel/cavs/hda-dma.c:634 	hda-dmac: 5 invalid channel -1
    0      2       EQ_IIR            197677.916667         7.135417   src/audio/eq_iir.c:870 	eq_iir_reset()
    0      2         COMP            197682.500000         4.583333 rc/audio/component.c:136 	comp_set_state(), state already set to 1
    0      2       EQ_FIR            197686.145833         3.645833   src/audio/eq_fir.c:779 	eq_fir_reset()
    0      2         COMP            197690.468750         4.322917 rc/audio/component.c:136 	comp_set_state(), state already set to 1
    0      2       VOLUME            197694.114583         3.645833   src/audio/volume.c:621 	volume_reset()
    0      2         COMP            197697.760417         3.645833 rc/audio/component.c:136 	comp_set_state(), state already set to 1
    0      2          DAI 1.8        197701.406250         3.645833      src/audio/dai.c:521 	dai_reset()
    0      2         COMP            197705.156250         3.750000 rc/audio/component.c:136 	comp_set_state(), state already set to 1

ranj063, 2019-03-26:
@singalsu @juimonen do you also see this error?
Is there a temp workaround for this that can unblock me?

juimonen, 2019-03-27:
@ranj063 yes I see it. As @singalsu said in irc the problem is the memcpy_s in component new().
memcpy_s(&dev->comp, sizeof(dev->comp) 
should be 
memcpy_s(&dev->comp, sizeof(struct sof_ipc_comp_process). 
with that change the error is gone. 
I can make PR for this if @singalsu doesn't have it ready already...

singalsu, 2019-03-27:
The fix is in PR #1191.

PR #1191 (2019-03-29): EQ: Fix IPC data copy size in new() method in IIR and FIR
The used memcpy_s() has wrong size for destination. It causes the
prepare() method to fail later in component life cycle into sink buffer
resize due to zero config->periods_sink value. This patch fixes the
fail of pipeline instantantion with IIR and FIR EQ components.

this fixes #1190
closed 2019-03-29 by tlauda