thesofproject/sof #1200: Add to github templates for issues / enhancements /  RFCs / feature requests

lbetlej, 2019-03-29:
The requested project improvement is to add to the Github project interface templates for:
* Bug report
* Enhancement
* Feature request
* RFC / Proposal

Example project that has such support:
https://github.com/zephyrproject-rtos/zephyr/issues/new/choose
I suggest to use the templates as in the link above.
The procedure to add the templates descibed at https://help.github.com/en/articles/creating-issue-templates-for-your-repository requires administrator rights,

lgirdwood, 2019-04-01:
@lbetlej interestingly this adds the templates to the SOF repo ! this will mean it can then be updated by anyone. :)

@michalgrodzicki @mengdonglin I will add you guys as the default owners of bugs, features etc for firmware and driver respectively.

lgirdwood, 2019-04-01:
Added bugs and features for SOF FW, but feature, RFC and enhancement are all the same thing so should be filed unde feature request. 

@plbossart @ranj063 I'm not going to do this for kernel as it will add commits to your repo (master branch by default).  Any comments ?

lbetlej, 2019-04-02:
@lgirdwood Thank you for adding the templates for SOF repo. Actually I was looking to add them myself as a PR but I've found that for it may not work for multiple templates according to https://help.github.com/en/articles/manually-creating-a-single-issue-template-for-your-repository

lgirdwood, 2019-04-02:
@lbetlej let my try and update the file via a PR, it may be the above only applies for the initial creation.

lgirdwood, 2019-04-02:
@lbetlej @michalgrodzicki see https://github.com/thesofproject/sof/pull/1213

closed 2019-04-18 by lgirdwood