thesofproject/sof #1204: Tool test topology build failed

DarrenxWang, 2019-04-01:
m4: test-tone-playback.m4: 45: fatal error: Invalid parameters (8) to PIPELINE_ADD
m4: test-tone-playback.m4: 45: fatal error: Invalid parameters (8) to PIPELINE_ADD
m4: test-tone-playback.m4: 45: fatal error: Invalid parameters (8) to PIPELINE_ADD
test/topology/CMakeFiles/tests.dir/build.make:106: recipe for target 'test/topology/CMakeFiles/tests' failed
make[3]: *** [test/topology/CMakeFiles/tests] Error 123
CMakeFiles/Makefile2:1557: recipe for target 'test/topology/CMakeFiles/tests.dir/all' failed
make[2]: *** [test/topology/CMakeFiles/tests.dir/all] Error 2
CMakeFiles/Makefile2:1564: recipe for target 'test/topology/CMakeFiles/tests.dir/rule' failed
make[1]: *** [test/topology/CMakeFiles/tests.dir/rule] Error 2
Makefile:697: recipe for target 'tests' failed
make: *** [tests] Error 2

lgirdwood, 2019-04-01:
@xiulipan I dont this this make target is tested by default for CI.

jajanusz, 2019-04-01:
> @xiulipan I dont this this make target is tested by default for CI.

It's not. CI runs `scripts/build-tools.sh`, it would run `make tests` with `-t` argument. Should @xiulipan change to run it with tests?

xiulipan, 2019-04-02:
@lgirdwood 
For sof-ci, we have the `-t`. But without #1206 it won't exit with error.
For travis-ci we did not test `-t`. It will generate too many logs exclude the travis limitation, so we disabled it. If we would like to enable it again, we will need to try to redirect the logs of tools build.

xiulipan, 2019-04-02:
@ranj063 @singalsu @xun2z @lgirdwood 
I think we can discuss here about the test topology.

If we should renew it? or just delete it?

lgirdwood, 2019-04-02:
@xiulipan I think @singalsu wanted to keep a few of the test topologies for test bench. @singalsu can you name the tests you ant to keep and we can build them as part of test bench.

singalsu, 2019-04-02:
We have used in tests playback pipelines with host -> comp -> DAI, where comp has been volume, SRC, IIR, FIR. The PCM format to use can be stereo S32_LE and DAI parameters do not matter. Testbench can overwrite the PCM format to other that's specified in the command line.

lgirdwood, 2019-04-02:
@singalsu Sorry, not following - we are looking for the test files you need to keep from the following.
```
lrg@lofn:~/work/sof/sof$ ls -l tools/test/topology/
total 60
-rw-r--r-- 1 lrg lrg   702 Mar 11 15:38 CMakeLists.txt
-rw-rw-r-- 1 lrg lrg 12039 Jan 31 16:18 Makefile
-rw-rw-r-- 1 lrg lrg 11890 Jan 31 16:18 Makefile.in
-rw-r--r-- 1 lrg lrg  2859 Jan 31 09:18 test-all.m4
-rw-r--r-- 1 lrg lrg  2589 Jan 31 09:18 test-capture.m4
-rw-r--r-- 1 lrg lrg  1980 Jan 31 09:18 test-equalizer-playback-ssp.m4
-rw-r--r-- 1 lrg lrg  2191 Jan 31 09:18 test-playback.m4
-rw-r--r-- 1 lrg lrg  1971 Jan 31 09:18 test-tone-playback.m4
-rwxr-xr-x 1 lrg lrg 11379 Mar 11 15:38 tplg-build.sh
```
Which ones do you need as @xiulipan will delete the others.

singalsu, 2019-04-03:
@lgirdwood I need none of those (obsolete). I seem to need to create new test pipelines so I it's better to create them from scratch cleanly.

jajanusz, 2019-04-03:
> @lgirdwood I need none of those (obsolete). I seem to need to create new test pipelines so I it's better to create them from scratch cleanly.

Are these tests ran by ./testbench (+octave scripts) or on real device?

lgirdwood, 2019-04-03:
@jajanusz it sounds like they are not. @singalsu ok, but we will need examples of your working test bench topologies so that others can reuse as examples. Could you upstream some of your existing test bench pipelines and that will et @xiulipan delete these tests.

PR #1205 (2019-04-03): tools: test: topology: add argument to fix build error in test tplg
remove test-tone-playback in test to fix build errors in ci

This is to fix #1204
closed 2019-04-03 by lgirdwood