thesofproject/sof #1208: Hear the wrong frequency sound when aplay correct params on BYT-rt5651 platform.

Jiangxinx, 2019-04-01:
Summary:
Hear the wrong frequency sound when aplay correct params on BYT-rt5651 platform.
Dmesg and sof-logger have no obvious error trace.

Step:
1.aplay -Dhw:0,0 -c2 -r48000 -fS16_LE Music/sine_48k_16bit_10s_stereo_l5_r2.wav -vv -i

Output:
```
Playing WAVE 'Music/sine_48k_16bit_10s_stereo_l5_r2.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-bytcr-rt5651' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16320
  period_size  : 4032
  period_time  : 84000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4032
  period_event : 0
  start_threshold  : 16320
  stop_threshold   : 16320
  silence_threshold: 0
  silence_size : 0
  boundary     : 9187343239835811840
  appl_ptr     : 0
  hw_ptr       : 0
#########################+                         | 48%
```

Env:
sof master: 052942c8
kernel sof-dev: f304b5c2
tplg: sof-byt-rt5651.tplg

Recorded sine file
[Recorded_sine.zip](https://github.com/thesofproject/sof/files/3028706/Recorded_sine.zip)
:
[image]

Original sine file:
[Original_sine.zip](https://github.com/thesofproject/sof/files/3028716/Original_sine.zip)

tlauda, 2019-04-01:
@Jiangxinx Please always add logs. There is no point in creating an issue without dmesg and FW logs. What is the repro ratio?

lgirdwood, 2019-04-01:
@Jiangxinx L/R look similarly corrupted. @bardliao could this be codec misconfiguration ?

bardliao, 2019-04-01:
@lgirdwood Yes, codec can copy the data from one channel to the other channel or mix L/R together and output to both L/R channels. @Jiangxinx Could you use [LR_48k.zip](https://github.com/thesofproject/sof/files/3029791/LR_48k.zip) to see if the L/R channel are output separated or not? 
Regarding the frequency issue, could you measure the physical frequency of LRCK, BCLK, and MCLK?
Also, could you add ``#define DEBUG`` in rt5651.c and attach the dmesg output here?

singalsu, 2019-04-02:
@Jiangxinx When doing error recordings please use low enough recording gain so that recording is not clipped to full scale as this is. Also mp3 encoding is not a recommended format. It's a lossy coding that does not preserve the waveform shape. Please use flac or wav format.

But I think the mp3 is representative enough to say that the effect that is heard is not wrong frequency (e.g. pitch shift perceived) but quite bad corruption. If the used format would be lossless e.g. buffer discontinuity errors would be visible. But mp3 format blurs such details.

Jiangxinx, 2019-04-02:
@bardliao I tried LR_48K using the daily version ofter enable DEBUG in rt5651.c, L/R channel are not separated and can hear the wrong frequency sound when aplay.

Dmesg after enable DEBUG:
```
[  819.097328] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 12096 DAI position 1766016
[  819.128297] sof-audio-acpi 80860F28:01: ipc rx: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.128320] sof-audio-acpi 80860F28:01: posn mailbox: posn offset is 0x144a00
[  819.128326] sof-audio-acpi 80860F28:01: posn : host 0x1e00 dai 0x69cd80 wall 0x14aa2261
[  819.128338] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 1920 DAI position 1733472
[  819.128355] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.128628] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 1920 DAI position 1733472
[  819.180699] sof-audio-acpi 80860F28:01: ipc rx: 0x600a000d: GLB_STREAM_MSG: POSITION
[  819.180728] sof-audio-acpi 80860F28:01: posn mailbox: posn offset is 0x144a4c
[  819.180738] sof-audio-acpi 80860F28:01: posn : host 0x0 dai 0x6c4800 wall 0x1526181e
[  819.180758] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 0 DAI position 1774080
[  819.180790] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a000d: GLB_STREAM_MSG: POSITION
[  819.181292] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 0 DAI position 1774080
[  819.212716] sof-audio-acpi 80860F28:01: ipc rx: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.212747] sof-audio-acpi 80860F28:01: posn mailbox: posn offset is 0x144a00
[  819.212760] sof-audio-acpi 80860F28:01: posn : host 0x5d00 dai 0x6a4b80 wall 0x14c2be99
[  819.212783] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 5952 DAI position 1741536
[  819.212814] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.213004] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 5952 DAI position 1741536
[  819.264706] sof-audio-acpi 80860F28:01: ipc rx: 0x600a000d: GLB_STREAM_MSG: POSITION
[  819.264737] sof-audio-acpi 80860F28:01: posn mailbox: posn offset is 0x144a4c
[  819.264750] sof-audio-acpi 80860F28:01: posn : host 0x3f00 dai 0x6cc600 wall 0x153eb420
[  819.264772] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 4032 DAI position 1782144
[  819.264807] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a000d: GLB_STREAM_MSG: POSITION
[  819.265268] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 4032 DAI position 1782144
[  819.296745] sof-audio-acpi 80860F28:01: ipc rx: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.296776] sof-audio-acpi 80860F28:01: posn mailbox: posn offset is 0x144a00
[  819.296788] sof-audio-acpi 80860F28:01: posn : host 0x9c00 dai 0x6ac980 wall 0x14db5a92
[  819.296811] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 9984 DAI position 1749600
[  819.296843] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.297028] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 9984 DAI position 1749600
[  819.305442] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[  819.305454] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  819.348768] sof-audio-acpi 80860F28:01: ipc rx: 0x600a000d: GLB_STREAM_MSG: POSITION
[  819.348797] sof-audio-acpi 80860F28:01: posn mailbox: posn offset is 0x144a4c
[  819.348807] sof-audio-acpi 80860F28:01: posn : host 0x7e00 dai 0x6d4400 wall 0x15575021
[  819.348827] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 8064 DAI position 1790208
[  819.348858] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a000d: GLB_STREAM_MSG: POSITION
[  819.349457] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 8064 DAI position 1790208
[  819.380777] sof-audio-acpi 80860F28:01: ipc rx: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.380808] sof-audio-acpi 80860F28:01: posn mailbox: posn offset is 0x144a00
[  819.380820] sof-audio-acpi 80860F28:01: posn : host 0xdb00 dai 0x6b4780 wall 0x14f3f69c
[  819.380843] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 14016 DAI position 1757664
[  819.380876] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.381057] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 14016 DAI position 1757664
[  819.432775] sof-audio-acpi 80860F28:01: ipc rx: 0x600a000d: GLB_STREAM_MSG: POSITION
[  819.432806] sof-audio-acpi 80860F28:01: posn mailbox: posn offset is 0x144a4c
[  819.432818] sof-audio-acpi 80860F28:01: posn : host 0xbd00 dai 0x6dc200 wall 0x156fec19
[  819.432841] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 12096 DAI position 1798272
[  819.432877] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a000d: GLB_STREAM_MSG: POSITION
[  819.433269] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 12096 DAI position 1798272
[  819.464809] sof-audio-acpi 80860F28:01: ipc rx: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.464840] sof-audio-acpi 80860F28:01: posn mailbox: posn offset is 0x144a00
[  819.464854] sof-audio-acpi 80860F28:01: posn : host 0x1b00 dai 0x6bc580 wall 0x150c9298
[  819.464863] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 1728 DAI position 1765728
[  819.464875] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.465006] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 1728 DAI position 1765728
[... cut ...]
[  819.800575] sof-audio-acpi 80860F28:01: posn : host 0x1800 dai 0x6dbd80 wall 0x156f0270
[  819.800583] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 1536 DAI position 1797984
[  819.800590] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a0001: GLB_STREAM_MSG: POSITION
[  819.800820] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 0 DMA position 1536 DAI position 1797984
```

Log file:
[dmesg.log](https://github.com/thesofproject/sof/files/3033042/dmesg.log)

LRCK:
[image]

BCLK:
[image]

MCLK:
[image]

bardliao, 2019-04-02:
@Jiangxinx All LRCK/BCLK/MCLK looks the same? And the frequency is very low. Could you double check it?

Jiangxinx, 2019-04-02:
@bardliao Retested the aplay function after removed  ccflags-y += -DDEBUG, and attach dmesg info below.By the way,i have checked the LRCK/BCLK/MCLK again, and got the same results.

Git diff:

dmesg:
```
[   13.215996] sof-audio-acpi 80860F28:01: BYT-CR not detected
[   13.216113] sof-audio-acpi 80860F28:01: LPE PHY base at 0x90a00000 size 0x200000
[   13.216150] sof-audio-acpi 80860F28:01: LPE VADDR 00000000902383c0
[   13.216153] sof-audio-acpi 80860F28:01: IMR base at 0x20000000 size 0x100000
[   13.216166] sof-audio-acpi 80860F28:01: IMR VADDR 00000000a56d2176
[   13.216170] sof-audio-acpi 80860F28:01: using IRQ 29
[   13.252718] sof-audio-acpi 80860F28:01: ipc: DSP is ready 0x70000000 offset 0x144000
[   13.252744] sof-audio-acpi 80860F28:01: Firmware info: version 1:1:0-c4b1a
[   13.252747] sof-audio-acpi 80860F28:01: Firmware: ABI 3:2:0 Kernel ABI 3:3:0
[   13.252812] sof-audio-acpi 80860F28:01:  mailbox upstream 0x144000 - size 0x400
[   13.252815] sof-audio-acpi 80860F28:01:  mailbox downstream 0x144400 - size 0x400
[   13.252818] sof-audio-acpi 80860F28:01:  stream region 0x144a00 - size 0x200
[   13.255113] sof-audio-acpi 80860F28:01: firmware boot complete
[   13.478268] bytcr_rt5651 bytcr_rt5651: quirk IN1_MAP enabled
[   13.478271] bytcr_rt5651 bytcr_rt5651: quirk MCLK_EN enabled
[   13.478285] bytcr_rt5651 bytcr_rt5651: info: override FE DAI link SSP2-Codec
[   13.505373] sof-audio-acpi 80860F28:01: warning: widget type 1 name codec_out0 not handled
[   13.506213] sof-audio-acpi 80860F28:01: warning: widget type 7 name ssp2 Rx not handled
[   13.506219] sof-audio-acpi 80860F28:01: warning: widget type 1 name codec_out1 not handled
[   13.506745] sof-audio-acpi 80860F28:01: warning: widget type 7 name ssp2 Tx not handled
[   13.506750] sof-audio-acpi 80860F28:01: warning: widget type 0 name codec_in0 not handled
[   13.507418] sof-audio-acpi 80860F28:01: warning: widget type 0 name codec_in1 not handled
[   13.508610] sof-audio-acpi 80860F28:01: ASoC: Parent card not yet available, widget card binding deferred
[   13.509833] rt5651 i2c-10EC5651:00: Warning: Invalid over-current-threshold-microamp value: 0, defaulting to 2000uA
[   13.510410] bytcr_rt5651 bytcr_rt5651: rt5651-aif1 <-> ssp2-port mapping ok
[   13.510497] bytcr_rt5651 bytcr_rt5651: snd-soc-dummy-dai <-> Low Latency 0 mapping ok
[   13.513548] bytcr_rt5651 bytcr_rt5651: snd-soc-dummy-dai <-> Media Playback 1 mapping ok
[   13.514542] rt5651 i2c-10EC5651:00: bypass=0 m=3 n=30 k=3
[   13.515081] rt5651 i2c-10EC5651:00: Sysclk is 24576000Hz and clock id is 1
[   14.375271] systemd-udevd (1478) used greatest stack depth: 12408 bytes left
[   19.125108] systemd-journald[1427]: File /var/log/journal/925896dd495840eca8c6be1ff455ace4/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   21.032651] igb 0000:03:00.0 enp3s0: igb: enp3s0 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
[   21.033204] IPv6: ADDRCONF(NETDEV_CHANGE): enp3s0: link becomes ready
[   38.851963] rt5651 i2c-10EC5651:00: bclk is 3072000Hz and lrck is 48000Hz
[   38.851967] rt5651 i2c-10EC5651:00: bclk_ms is 1 and pre_div is 1 for iis 0
[   89.006707] random: crng init done
[   89.006720] random: 7 urandom warning(s) missed due to ratelimiting
[   94.856708] sof-audio-acpi 80860F28:01: error: ipc timed out for 0x60050000 size 12
[   94.856793] sof-audio-acpi 80860F28:01: error: unexpected fault 0x600a0001 trace 0x60050000
[   94.856799] sof-audio-acpi 80860F28:01: error: waking up any trace sleepers
[   94.856803]  Low Latency: ASoC: trigger FE failed -110
[  100.889513] rfkill: input handler disabled
[  163.139376] rt5651 i2c-10EC5651:00: bclk is 3072000Hz and lrck is 48000Hz
[  163.139381] rt5651 i2c-10EC5651:00: bclk_ms is 1 and pre_div is 1 for iis 0
[  190.296721] sof-audio-acpi 80860F28:01: error: ipc timed out for 0x60050000 size 12
[  190.296816] sof-audio-acpi 80860F28:01: error: unexpected fault 0x600a0001 trace 0x60050000
[  190.296824] sof-audio-acpi 80860F28:01: error: waking up any trace sleepers
[  190.296834]  Low Latency: ASoC: trigger FE failed -110
```

Log file:
[dmesg.log](https://github.com/thesofproject/sof/files/3033198/dmesg.log)

singalsu, 2019-04-02:
@Jiangxinx Are the oscilloscope pictures above just mains interference, 50 Hz?

lgirdwood, 2019-04-02:
> @Jiangxinx Are the oscilloscope pictures above just mains interference, 50 Hz?

Yep, and given LRCLK, BCLK and MCLK are all 50Hz means @Jiangxinx your scope is not correctly connected to the signasl and grounded. Please also make sure the playback is enabled and running  too (since scopes will "autodetect" the mains if nothing else is propagating).

juimonen, 2019-04-02:
@lgirdwood don't now about scope grounding, but I just got mb&rt5651 playing first time and the audio is complete bogus. it is like really low frequency brittle or anyway something very difficult to describe. I can switch between hp and speaker so I think my alsamixer setup is at least somehow correct.

bardliao, 2019-04-02:
@lgirdwood 
```
[   94.856708] sof-audio-acpi 80860F28:01: error: ipc timed out for 0x60050000 size 12
[   94.856793] sof-audio-acpi 80860F28:01: error: unexpected fault 0x600a0001 trace 0x60050000
[   94.856799] sof-audio-acpi 80860F28:01: error: waking up any trace sleepers
[   94.856803]  Low Latency: ASoC: trigger FE failed -110
```
Does it mean there are something wrong on FW?

lgirdwood, 2019-04-02:
@bardliao yes, also something wrong with FW here too or driver here too.

juimonen, 2019-04-03:
@Jiangxinx did you update SW in between the tests? I don't get why removing DDEBUG would cause error. I mean your first log doesn't contain error and I guess the streaming should not even start if you get this...?

juimonen, 2019-04-03:
@Jiangxinx @bardliao I have the same error, but it's coming always _after_ streaming when "stop" is triggered. For e.g. I have a experimental pm setup where I'm loading topology always from scratch  when streaming starts. So in my opinion this error should not be the cause of corrupted sound...

bardliao, 2019-04-03:
> @Jiangxinx did you update SW in between the tests? I don't get why removing DDEBUG would cause error. I mean your first log doesn't contain error and I guess the streaming should not even start if you get this...?

@juimonen I think we didn't see the error in the first log is that there are too many logs and it exceeds the buffer of dmesg. So we can only see the recent log.

juimonen, 2019-04-03:
@bardliao yes understood it also later... would it be possible for you to check we are setting the codec parameters ok? Otherwise I can't think of anything else than the dma to dsp is different in this platform and/or we are copying some total garbage to codec.

juimonen, 2019-04-03:
@bardliao sorry to confuse you, I think there's no issue with the codec. I had myself some issues with the build system (I think I was building some garbage firmware) see my comments in: https://github.com/thesofproject/sof/issues/1215

bardliao, 2019-04-04:
@juimonen No problem. Hope it can be fixed soon :)

wenqingfu, 2019-04-04:
@juimonen , is this a duplicate of #1215?

juimonen, 2019-04-04:
@wenqingfu not 100% sure but really looks like it.

wenqingfu, 2019-04-04:
mark as duplicate of #1215 . @tlauda @juimonen please move covnversation to #1215. @Jiangxinx please provide testing support to #1215 and reopen this bug if you can still reproduce it after #1215 is fixed.

closed 2019-04-04 by wenqingfu