thesofproject/sof #1212: [FEATURE] IPC 2.0

lgirdwood, 2019-04-02:
Current IPC survives MINOR and PATCH semantic ABI version updates but needs to be fully future proof. We need a fully future proof IPC where structure additions and changes are non fatal to every structure and substructure.

plbossart, 2019-05-30:
@lgirdwood I have zero cycles to work on this until Q4.

lgirdwood, 2019-05-30:
IPC 2.0 should have the following requirements.
1) Support IPC deprecation
2) Support IPC growth
3) Support IPC discoverability.
4) Support multiple classes of features (not just audio).

Requirements can be met if :

1) IPC "ID space" is split into several classes like audio, debug, sensors, etc
2) Each IPC class has a set of supported IPC commands (with GUID).
3) Each IPC command is tightly coupled to a C structure.
4) Supported IPC classes and IPC command GUIDs are sent to the host using the FW ready message.
5) All IPC C structures use the same small header.

All the above means that a host drivers receiving a list of IPC classes and IPC command GUIDs at FW boot time can map commands to the correct IPC function that deals with that GUID. i..e the IPC audio class and IPC audio command GUIDs will be parsed by audio driver, likewise for sensor driver etc.

lgirdwood, 2019-05-30:
@plbossart I know, it's to keep you in the loop.