thesofproject/sof #1215: [BUG] Baytrail/CherryTrail: samples are not played in correct order

plbossart, 2019-04-02:
**Describe the bug**

When playing samples that are incremented by one from a sample to the other, the output shows discontinuities of 96 samples, see 
[baytrail.xlsx](https://github.com/thesofproject/sof/files/3035671/baytrail.xlsx)
. With regular sine waves this has the net effect of adding noise.

This clearly points to bad buffer/pointer management or scheduling issues.

**To Reproduce**

Configure Minnowboard, use kernel topic/sof-dev and firmware v1.3-rc
connect LogicPro on SSP pins (bclk, fs, dac)
unzip and play file
[cnt_1024_left.pcm.gz](https://github.com/thesofproject/sof/files/3035664/cnt_1024_left.pcm.gz)
aplay -Dhw:1,0 -c2 -r48000 -f dat cnt_1024_left.pcm
grab results with LogicPro and check the decoded values (should be similar to excel sheet provided above)

**Expected behavior**

samples are expected to be played in order, with wrap-around at 1024

**Impact**

showstopper

**Environment**
1) Branch name and commit hash of 3 repositories: sof (firmware), linux (kernel driver) and soft (tools & topology).

kernel f3adfd66c97c730a9db99dafabab7a94f6bcf552
firmware eac4a732324c058bf44690ad7df3c7475f833b2e
2) Name of the topology file

sof-byt-nocodec.tplg (or any byt topology, doesn't matter)

3) Name of the platform(s) on which the bug is observed.

All Baytrail and cherrytrail platforms. The measurements are only seen when the connector is exposed, e.g. on Minnowboard.

4) Reproducibility Rate. 100%

lgirdwood, 2019-04-03:
@mengdonglin @RanderWang I'm sure this is dw-dma DMA buffer R/W pointer related. The DMA on BYT does not support HW LLI mode and LLI must be programmed via SW. There have also been some recent commits to the DMA driver that should be bisected. 

The issue here is the DMA HW read period (that is be sent to SSP) is being written by the FW pipeline at the same time. i.e. we are reading and writing the same period at the same time. Some trace should show this. This is obviously working in HW LLI mode for non BYT platforms.

tlauda, 2019-04-03:
@ranj063 @xiulipan @lgirdwood  What has changed since issue #1107 has been fixed? Are you saying there is another regression? The only problem with noise I was aware of was problem with SRC #1161.

juimonen, 2019-04-03:
@tlauda @lgirdwood I did some bisecting and for me the problems start way back:
so this is minnowboard + rt5651 (and rt5651 topology, sof master  cbd3e07031)
1) 22e3cf935bd71c7d42a39f0c4389c78a25837ddc works for me I can play audio and its clean.
2) 9a6a38a609c2e1cb67715075c74005d48b665578 starts problems, this commit doesn't work at all
3) 407be960c0b9591b2b43f7943d7be08c7efb271b onward until HEAD I can play but the sound is metallic.

lgirdwood, 2019-04-03:
@tlauda not sure but CI was reporting false success.
@juimonen I've ran into this before hence I know the root cause. Can you check the DMA and it's source component R/W pointers as they are both using the same period for IO as explained above.

xiulipan, 2019-04-04:
@tlauda @lgirdwood @juimonen 
Confirmed with the today's daily build binary. With pause/resume the noise can be randomly recover.
It should be like what @lgirdwood have said

> I'm sure this is dw-dma DMA buffer R/W pointer related. The DMA on BYT does not support HW LLI mode and LLI must be programmed via SW. There have also been some recent commits to the DMA driver that should be bisected.
> 
> The issue here is that the DMA HW read period (data that is be sent to SSP) is being written by the FW pipeline at the same time. i.e. we are reading and writing the same period at the same time. Some trace should show this. This is obviously working in HW LLI mode for non BYT platforms.

And we have fixed similar issue once. I think this maybe caused by the recent change to the pipeline.

But as https://github.com/thesofproject/sof/commit/9a6a38a609c2e1cb67715075c74005d48b665578 is something related to the mixer component. I will try to use some pass through pipeline to check if it is related to the mixer component first.

juimonen, 2019-04-04:
@xiulipan 9a6a38a is the first in series of pipeline changes for variable frame size and I don't think the commits work individually. So I had to go up until 407be96 to get the pipeline even working.

ranj063, 2019-04-04:
@juimonen to rule out the mixer as a suspect, can you try a simple host->vol->dai pipeline on BYT?

xiulipan, 2019-04-04:
@juimonen thanks for the hint. It is a set of patches.
@ranj063 
I will do that too.

juimonen, 2019-04-04:
@xiulipan @tomek @ranj063 @lgirdwood  I got simplified topology (without mixer) from Ranjani and it plays cleanly with master. The basic rt5651 topology with mixer has metallic sound with master. I have some unreliability in my setup so sometimes I get just low level rumble. Anyway it is quite clear that there's some issues with mixer component (or the whole pipeline including it).

tlauda, 2019-04-04:
@juimonen What is the stream format vs. dai format? Can you get me verbose FW logs? It looks like some kind of mismatch in frame_fmt between mixer and volume components.

juimonen, 2019-04-04:
@tlauda can someone please tell me how to enable TRACEV?
I'm running ./scripts/docker-run.sh ./scripts/xtensa-build-all.sh -l byt, so how to enable it?

tlauda, 2019-04-04:
@juimonen make menuconfig and select verbose traces.

juimonen, 2019-04-04:
@tlauda doesn't work with docker so the "CI way"... that's the only setup I have atm. Where is the menuconfig inventing this? can I add it by hand to some file. I cna't add it to .config because they get always regenerated.

juimonen, 2019-04-04:
@tlauda ok got it working by editing src/lib/Kconfig

here's the log from mixer pipeline with metallic sound:

[log_mixer.txt](https://github.com/thesofproject/sof/files/3042516/log_mixer.txt)

here's the log with simple pipeline with clean sound:

[log_simple.txt](https://github.com/thesofproject/sof/files/3042520/log_simple.txt)

xiulipan, 2019-04-04:
@juimonen You can do it with docker. You can just run what you would like with docker. Docker used map for the device.
@tlauda @ranj063 
I also tried with sof-cht-rt5682.tplg that is a simple `host->vol->dai` pipeline.
But I found if I pause/resume the playback, the noise will random come.

With the sof-byt-rt5651.tplg that is `host->vol->mixer->volume->dai`, the noise will come at beginning. But with some pause/resume it will recover.

lgirdwood, 2019-04-04:
@xiulipan @juimonen to restate, the issue is there is a race starting or synchronizing the DW-DMA HW wrt the rest of the pipeline.

Consider the SSP DMA buffer B with two periods p0, p1. On failure, the pipeline is writing to P0 whilst the DMA is simultaneously readying from p0. Normal operation is that pipeline writes to p0 whilst DMA reads p1.

The ordering PCM data from @plbossart is a big hint here as the DMA engine will prefetch p0 or p1 (and copy to the internal DMA FIFO) in multiples of DMA burst size. This will show a mixture of old and new data aligned on burst size multiples.

It is also worth checking the host DMA side too given the scheduling updates.

@juimonen can you resend logs of good and bad using the **same** simple pipeline topology. Just an excerpt is needed showing copying of about 20 periods.

juimonen, 2019-04-04:
@lgirdwood here's only bad log. I just can't get the clean thing out of this. Don't know if it is the excessive logging or something, but I booted the device 10 times, always bad.
[log-simple-bad.txt](https://github.com/thesofproject/sof/files/3043234/log-simple-bad.txt)

tlauda, 2019-04-04:
@juimonen So now also the simple topology doesn't work without noise?

juimonen, 2019-04-04:
@tlauda yeah, I can sometimes play simple topology like couple of seconds, but then begins the noise (its like wind noise on microphone) and the music is gone. Don't know if the dma pointers get out of sync etc. I have to go way back to 22e3cf935bd71c7d42a39f0c4389c78a25837ddc to get reliable playback

juimonen, 2019-04-04:
@tlauda there is something really strange with the system. I'm not sure how I get my device in to this low rumble mode, but for e.g. now I can play simple topology clean with master. I'm sorry about this confusion but I'm also myself confused, is it that our CI scripts are not making clean builds when I jump around in git history or is my device borked... really unreliable results

ranj063, 2019-04-04:
@tlauda @juimonen I can confirm that with the commit cbbc6afd3 the audio is clean on my minnowboard with a **no mixer topology**. but I did see an underrun once during 2 minutes of playback but audio recovered and continued to play without any issues. 

With a mixer based topology, the sound is metallic.

juimonen, 2019-04-04:
@ranj063 if you try what @xiulipan said today in bug scrub: try aplay -i and use space to pause/resume, you can get the mixer topology also to play clean sometimes. I think this is something what @lgirdwood described so host or dai dma is getting confused. Mixer topology is just bigger so it gets to "metallic" more often.

lgirdwood, 2019-04-04:
@juimonen thanks for the log, but it needs verbose trace enabled to see the R/W pointer or they may need manually added to trace. It would be good to see the timing and R/W pointer at trigger start and then when the pipeline is next scheduled (as this could be scheduled too soon).

juimonen, 2019-04-08:
I've been trying to log the dai write vs. dma read pointers and at least to me it seems that dma read pointer is always behind the dai writer (so not at least not overlapping with the current write period). I'm checking this in dw_dma_copy where dai has written a period and is updating pointers for dma (sar register value is not inside current_ptr + bytes).

I've also tried to fiddle around cache_writeback because the error sounds like buffer corruption, but without luck.

If someone has time I could really use extra eye balls or good advice...

tlauda, 2019-04-10:
Fixed by #1247.

mengdonglin, 2019-04-15:
@Jiangxinx Please help to check if this issue is fixed by  #1247 on Baytrail.

lgirdwood, 2019-04-17:
@mengdonglin @tlauda according to @stevyan this is still an issue for playback but not capture on BYT.

tlauda, 2019-04-17:
@lgirdwood Does @stevyan use the right topology? Note that only Dhw:0,0 works for now, Dhw:0,1 is media pipe and still needs topology changes.

lgirdwood, 2019-04-17:
@tlauda not sure, but I would be happy to do an v3.1-rc2 if Dhw:0,0 is working. @stevyan can you describe analog playback/capture test you are using.

wenqingfu, 2019-04-17:
@juimonen  how about testing on your side? Is everything ok with @tlauda 's patch?

juimonen, 2019-04-17:
@wenqingfu yes for me the hw0:0 device is working with @tlauda topology.

juimonen, 2019-04-17:
@tlauda have not been following fw PR's closely, but do you have the PR also for the pause/resume fix? I think that would be very good for rc2...

tlauda, 2019-04-17:
@juimonen It's by default fixed by #1273.

juimonen, 2019-04-17:
@tlauda, cool and thanks!

juimonen, 2019-04-18:
works for me with byt-mb + rt5651!

closed 2019-04-18 by juimonen