thesofproject/sof #1222: Heap debug : Firmware build fail  [BUG]

rsalma, 2019-04-04:
Summary : 
Firmware binaries build failed after enabling the Heap debug.

Screenshot : 
[image]

Test env:
sof master: cbd3e07031bca60865ef6c6fd5f3b0c0d5fe296b

abonislawski, 2019-04-04:
Nice catch, looks like this functions simply does not exists and even not exists in commit which introduced this functionality.

Fixed in PR #1223

lgirdwood, 2019-04-04:
@xiulipan any reason why this slipped through CI ? I guess we are only building a few Kconfigs ?
@mengdonglin we should be able to use 0day again now thet we use Kconfig and have run build randconfig.

jajanusz, 2019-04-04:
> @xiulipan any reason why this slipped through CI ? I guess we are only building a few Kconfigs ?
> @mengdonglin we should be able to use 0day again now thet we use Kconfig and run build randconfig.

We build defconfigs. Having 'randconfig' would result in non-determistic CI and I think we don't want that. If we want to test things like this we should just have some config with all debug options enabled for example. Or just add something like `allnoconfig` / `allyesconfig` / `alldefconfig` (kernel-like targets).

xiulipan, 2019-04-04:
@lgirdwood 
This should be some long existed bug. I was once try to test with HEAP_DEBUG but the feature seems to be totally not working. I think we may need some clean up about some debug option first then to enable it in CI. BTW: I did not think the DEBUG macros now is working.

And for each PR, we could not run all settings as it will take long time. To track all features we have, we first need to make sure it worked with manually test first and then we can try to use some scripts for the test.

I agreed with @jajanusz that we may not be able to run with random config to build FW. The FW may need some basic requirement?

lgirdwood, 2019-04-04:
@jajanusz @xiulipan rand config is not meant for deterministic CI, but for async background building (like 0day does for Linux) that will report any errors early before they ship in releases. 

@jajanusz building ```allnoconfig / allyesconfig / alldefconfig``` would be good for CI providing we have the spare cycles. I dont want to slow CI any further atm ;)

@xiulipan please report bugs when you discover them. No one can fix it if they dont know about it. It did work in the past.

closed 2019-04-04 by abonislawski