thesofproject/sof #1234: Why don't we have TRACE_DMA_FREE for trace?

ranj063, 2019-04-06:
Just like the pcm streams have a pcm params and pcm free?
Why does trace only have the TRACE_DMA_PARAMS? Do we need a free for trace?

plbossart, 2019-04-09:
The premise for PCM streams is that they can be used with different configurations, and you use the free/hw_params sequence to pass the new config information. For the trace things are quite static atm and I don't see a need to ever change any parameters. It's on and remains on. If we started going into dynamic traces activated on request then yes most likely we'd want to free.

lgirdwood, 2019-04-15:
I suppose this is something @mmaka1 eclipse plugin would eventually be able to control via debugFS and should probably be a feature.

ranj063, 2019-04-17:
Thanks @plbossart @lgirdwood. Maybe I'll leave this request for when the eclipse plugin in available to implement.