thesofproject/sof #1235: [BUG] sof_add_ld_script called too early

jajanusz, 2019-04-08:
**Describe the bug**
Gathering of includes here: https://github.com/thesofproject/sof/blob/dc452b2771bfbcc01c49c4cbce547c1efadaa48b/src/arch/xtensa/CMakeLists.txt#L102

Should be delayed to not depend on order of cmake subdirs.

**To Reproduce**
Add includes in platform that linker script will depend on and it will fail, because they are not yet added while creating target for linker script.

lgirdwood, 2019-04-18:
@jajanusz any update ?

jajanusz, 2019-04-18:
It's kinda architectural flow.
IMHO if you'd like to have it done nicely you'd have to have `platform` folder inside `arch`, because for example I think that platforms are specifications or something more generic - arch, I guess you are not going to have 1 apollolake for xtensa and 1 for arm. This way you'd be able to handle it there. I saw it even in kernel that for example you have samsung platforms-specifc stuff in arch/samsung/someplatform. With current structure we have to stay with workaround that depends on order and just processes platform CMakeLists before arch CMakeLists ...

jajanusz, 2019-05-16:
Closing, because it's not a major issue and solving it would require major refactor (not worth), so I'd have that in mind and refactor it when there will be opportunity.

closed 2019-05-16 by jajanusz