thesofproject/sof #1238: [BUG] Cmake/rimage: cannot change default key used by rimage

plbossart, 2019-04-08:
**Describe the bug**

To work on specific Intel boards, I need to change the private key used for signature. hardcoding the new path doesn't work with CMakeLists.txt, the default is always selected.  

Ideally we'd want an environment variable to be able to override this default to use an out-of-tree key instead of maintaining CMake changes.

**To Reproduce**

````diff
diff --git a/src/arch/xtensa/CMakeLists.txt b/src/arch/xtensa/CMakeLists.txt
index 6a43aeb3..2c5374b9 100644
--- a/src/arch/xtensa/CMakeLists.txt
+++ b/src/arch/xtensa/CMakeLists.txt
@@ -295,7 +295,7 @@ if(MEU_PATH)
                set(MEU_OFFSET 1088)
        endif()
 
-       set(otc_private_key ${PROJECT_SOURCE_DIR}/rimage/keys/otc_private_key.pem)
+       set(otc_private_key ${PROJECT_SOURCE_DIR}/rimage/keys/debug/private_key.pem)
 
        add_custom_target(
                run_rimage
````
**Expected behavior**

The new key is used for signature

**Impact**

Can't figure out how to sign the stupid firmware.

**Environment**

v1.3-rc1

xiulipan, 2019-04-09:
@plbossart 
That is interesting. As far as I know if you would like to change the key used. The best way is to change the cmd of the rimage call but not something inside the CMake. But that would be a good feature to let rimage to accept costume key with some ENV or CMake option.

I will check that.

jajanusz, 2019-04-09:
Ofc ability to change key should be added long time ago, I'm suprised it took so long for someone to complain about it. 

@xiulipan adds it in #1239, however we are not going to support anything via environment variables, because it's too easy to make a mess with them

jajanusz, 2019-04-09:
@plbossart BTW I guess your changes didn't work because you changed key for build with MEU when you just build FW without MEU and use only rimage. I cannot be sure beacause there are no reproduction steps, just *proposed solution*(?).

wenqingfu, 2019-04-16:
fixed by #1248?

jajanusz, 2019-04-16:
> fixed by #1248?

Yes, thanks, for noticing

closed 2019-04-16 by jajanusz