thesofproject/sof #1241: [S3][Regression] volume pipeline copy failure after 4 suspend/resume cycles on APL UP^2 PCM512X platform.

Jiangxinx, 2019-04-09:
**Summary**
Fail to resume aplay after suspend/resume 4 times on APL-UP2-pcm512x.
Dmesg have no abvious error trace.

**To Reproduce**
1.aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero& sudo ~/test/stress-s3.sh 50 1 1

**Expected behavior**
resume aplay

**Screenshots or console output**
```
== start time: 2019-04-09 16:44:25 ==
---- test 1 of 50 ----
System will suspend after 1 seconds ...
system will resume after 3 seconds ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Apr  9 08:44:31 2019
Suspended. Trying resume. Done.
---- test 2 of 50 ----
System will suspend after 3 seconds ...
system will resume after 4 seconds ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Apr  9 08:44:44 2019
Suspended. Trying resume. Done.
---- test 3 of 50 ----
System will suspend after 2 seconds ...
system will resume after 5 seconds ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Apr  9 08:44:57 2019
Suspended. Trying resume. Done.
---- test 4 of 50 ----
System will suspend after 5 seconds ...
system will resume after 4 seconds ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Apr  9 08:45:12 2019
Suspended. Trying resume. Done.
underrun!!! (at least 0.015 ms long)
aplay: pcm_write:2011: write error: Input/output error
aplay doesn't exist!
== end time: 2019-04-09 16:45:16 ==
[1]+  Exit 1                  aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero
```

**Environment**
sof master: 18d116b2
kernel sof-dev: b5ece71d
tplg: sof-apl-pcm512x.tplg

**Sof-logger**
```
    0      1       VOLUME           1697509.635417       824.375000   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1697679.687500       170.052078 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       1697685.625000         5.937500 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      1       VOLUME           1698509.739583       824.114563   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1698678.125000       168.385422 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       1698684.062500         5.937500 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      1       VOLUME           1699509.739583       825.677063   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1699678.541667       168.802078 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       1699684.531250         5.989583 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      1       VOLUME           1700509.739583       825.208313   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1700678.229167       168.489578 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       1700684.166667         5.937500 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      1       VOLUME           1701509.739583       825.572937   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1701676.250000       166.510422 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       1701682.187500         5.937500 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      1       VOLUME           1702552.968750       870.781250   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1702723.125000       170.156250 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       1702729.270833         6.145833 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      1       VOLUME           1702866.093750       136.822922   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1703035.885417       169.791672 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       1703041.979167         6.093750 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      1       VOLUME           1703509.739583       467.760406   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1703678.958333       169.218750 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       1703685.052083         6.093750 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      1       VOLUME           1704509.687500       824.635437   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1704679.218750       169.531250 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       1704685.312500         6.093750 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      1       VOLUME           1705509.791667       824.479187   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1705678.906250       169.114578 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       1705685.000000         6.093750 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      1       VOLUME           1706509.635417       824.635437   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      1         PIPE           1706679.375000       169.739578 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
```

**Log file**
[dmesg.log](https://github.com/thesofproject/sof/files/3058034/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3058040/logger.log)

Jiangxinx, 2019-04-10:
Update the daily report: this issue still exist,however,dmesg is null,sof-logger have no abvious error trace on today's version.

***To Reproduce***
1.aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero& sudo ~/test/stress-s3.sh 50 1 1

***Screenshots or console output***
```
[1] 2352
+ readonly LOG_DIR=logs
+ LOG_DIR=logs
+ readonly RANDOM_INIT_VALUE=1
+ RANDOM_INIT_VALUE=1
+ readonly RANDOM_RANGE=5
+ RANDOM_RANGE=5
+ readonly S3_SLEEP=2
+ S3_SLEEP=2
+ set +x
removed 'logs-last/test_2.log'
removed 'logs-last/test_1.log'
removed directory 'logs-last'
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
== start time: 2019-04-10 16:47:48 ==
---- test 1 of 50 ----
System will suspend after 5 seconds ...
system will resume after 5 seconds ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Wed Apr 10 08:47:59 2019
Suspended. Trying resume. Failed. Restarting stream. Done.
---- test 2 of 50 ----
```

***sof-logger***
```
    0      2       VOLUME             43962.447917         7.083333   src/audio/volume.c:434 	volume_cmd()
    0      2       VOLUME             43965.312500         2.864583   src/audio/volume.c:327 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 13
    0      2       VOLUME             43970.260417         4.947917   src/audio/volume.c:332 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 0, value = 2072
    0      2       VOLUME             43972.916667         2.656250   src/audio/volume.c:332 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 1, value = 2072
    0      2          IPC             44050.833333        77.916664    src/ipc/handler.c:865 	ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC             44053.593750         2.760417    src/ipc/handler.c:867 	ipc: comp 7 -> cmd 0
    0      2       VOLUME             44058.645833         5.052083   src/audio/volume.c:434 	volume_cmd()
    0      2       VOLUME             44061.197917         2.552083   src/audio/volume.c:327 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 7
    0      2       VOLUME             44063.802083         2.604167   src/audio/volume.c:332 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 0, value = 20723
    0      2       VOLUME             44066.354167         2.552083   src/audio/volume.c:332 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 1, value = 20723
    0      2          IPC             44134.479167        68.125000    src/ipc/handler.c:865 	ipc: hdr 0x50010000 tx (108) > rx (92)
    0      2          IPC             44137.135417         2.656250    src/ipc/handler.c:867 	ipc: comp 1 -> cmd 0
    0      2       VOLUME             44140.520833         3.385417   src/audio/volume.c:434 	volume_cmd()
    0      2       VOLUME             44143.229167         2.708333   src/audio/volume.c:327 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, cdata->comp_id = 1
    0      2       VOLUME             44145.677083         2.447917   src/audio/volume.c:332 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 0, value = 32844
    0      2       VOLUME             44148.281250         2.604167   src/audio/volume.c:332 	volume_ctrl_set_cmd(), SOF_CTRL_CMD_VOLUME, channel = 1, value = 32844
    0      2          IPC             44216.666667        68.385414    src/ipc/handler.c:643 	ipc: pm -> restore
    0      2          IPC             78683.229167     34466.562500    src/ipc/handler.c:239 	ipc: comp 0 -> params
    0      2         PIPE 1.5         78687.447917         4.218750 src/audio/pipeline.c:320 	pipeline_params()
    0      2         HOST             78691.041667         3.593750     src/audio/host.c:508 	host_params()
    0      2          DMA             78697.552083         6.510417 intel/cavs/hda-dma.c:480 	hda-dmac: 5 channel 0 -> get
    0      2          DMA             78701.250000         3.697917 intel/cavs/hda-dma.c:703 	hda-dmac: 5 channel 0 -> config
    0      2       VOLUME             78706.510417         5.260417   src/audio/volume.c:245 	volume_params()
    0      2          DAI 1.4         78710.364583         3.854167      src/audio/dai.c:419 	dai_params()
    0      2         PIPE 1.5         78717.083333         6.718750 src/audio/pipeline.c:358 	pipeline_prepare()
    0      2         HOST             78720.260417         3.177083     src/audio/host.c:640 	host_prepare()
    0      2       VOLUME             78733.020833        12.760417   src/audio/volume.c:535 	volume_prepare()
    0      2          DAI 1.4         78755.260417        22.239584      src/audio/dai.c:476 	dai_prepare()
    0      2          IPC             79104.166667       348.906250    src/ipc/handler.c:464 	ipc: comp 0 -> trigger cmd 0x40000
    0      2         PIPE 1.5         79107.760417         3.593750 src/audio/pipeline.c:533 	pipeline_trigger()
    0      2         HOST             79111.197917         3.437500     src/audio/host.c:281 	host_trigger()
    0      2          DMA             79114.583333         3.385417 intel/cavs/hda-dma.c:549 	hda-dmac: 5 channel 0 -> start
    0      2          DMA             79117.500000         2.916667 intel/cavs/hda-dma.c:396 	hda-dmac: 5 channel 0 -> enable
    0      2       VOLUME             79153.437500        35.937500   src/audio/volume.c:454 	volume_trigger()
    0      2          DAI 1.4         79156.822917         3.385417      src/audio/dai.c:544 	dai_comp_trigger(), command = 1
    0      2          DAI 1.4         79159.947917         3.125000      src/audio/dai.c:555 	dai_comp_trigger(), START
    0      2          DMA             79291.406250       131.458328 src/drivers/dw/dma.c:1384	dw_dma_interrupt_register()
    0      2          SSP             79305.104167        13.697917 ers/intel/cavs/ssp.c:812 	ssp_trigger() cmd 1
```

***Environment***
sof master: ed6ac05f
kernel sof-dev: f0ce2c52
tplg: sof-apl-pcm512x.tplg

***Log file***
[logger.log](https://github.com/thesofproject/sof/files/3062928/logger.log)

ranj063, 2019-04-11:
@Jiangxinx looking at the latest log, I dont see an error. Can you please explain what the problem is?

Jiangxinx, 2019-04-16:
@ranj063 This issue still exist even ignore "Suspended. Trying resume. Failed. Restarting stream. Done".After  suspend/resume, can not aplay,dmesg and sof-logger have no abvious error trace.

***To Reproduce***
1.aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero& sudo ~/test/stress-s3.sh 50 1 1
```
---- test 48 of 50 ----
System will suspend after 3 seconds ...
system will resume after 3 seconds ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Apr 16 08:00:03 2019
---- test 49 of 50 ----
System will suspend after 2 seconds ...
system will resume after 2 seconds ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Apr 16 08:00:13 2019
---- test 50 of 50 ----
System will suspend after 4 seconds ...
system will resume after 3 seconds ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Apr 16 08:00:26 2019
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
aplay: main:788: audio open error: Device or resource busy
Failed to aplay source!
== end time: 2019-04-16 16:00:30 ==
```

2.aplay -Dhw:0,0 -r 48000 -c2 -fs16_le /dev/zero
```
aplay: main:788: audio open error: Device or resource busy
```

***Environment***
sof master: 1b15fad8
kernel sof-dev: 2de29fa(compiled with PR #823 )
tplg: sof-apl-pcm512x.tplg

lgirdwood, 2019-04-16:
@Jiangxinx the busy error could imply that there is still a user with the sound device open. Can you confirm is pulseaudio is used here ?

https://askubuntu.com/questions/338382/pulseaudio-not-working-home-directory-not-accessible-permission-denied

Can you also try and see which application has the audio device open as this could be a pulseaudio issue.
```
lrg@lofn:~/work/sof/sof$ fuser /dev/snd/*
/dev/snd/controlC0:   2828
/dev/snd/controlC1:   2828
/dev/snd/controlC2:   2828
/dev/snd/pcmC2D0p:    2828m
lrg@lofn:~/work/sof/sof$ ps -ax | grep 2828
 2828 ?        Ssl  218:24 /usr/bin/pulseaudio --daemonize=no
```

Jiangxinx, 2019-04-17:
@lgirdwood It’s true that an application is taking up audio device,and this issue can not be reproduced after kill the application.

***Environment***
sof master: 99e9f437
kernel sof-dev: 25088516(compiled with PR #823 )
tplg: sof-apl-pcm512x.tplg

lgirdwood, 2019-04-17:
@Jiangxinx thanks, but I need to know if it's pulseaudio or aplay that's has the device open ? if it's PA then it's likely a PA bug (which can be dissected when driver upstreaming is done).

Jiangxinx, 2019-04-18:
@lgirdwood It is aplay that taking up audio device.I have no idea to comfirm it is another issue.
***Step***
aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero& sudo ./stress-s3.sh 50 1 1

***Output***
```
test@cd-apl-up2-pcm512x-02:~/test$ aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero
aplay: main:788: audio open error: Device or resource busy
```
```
test@cd-apl-up2-pcm512x-02:~/test$ fuser /dev/snd/*
/dev/snd/pcmC0D0p:    2234m
test@cd-apl-up2-pcm512x-02:~/test$ ps -ax | grep 2234
 2234 pts/0    S      0:00 aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero
 4504 pts/0    R+     0:00 grep --color=auto 2234
test@cd-apl-up2-pcm512x-02:~/test$ sudo kill 2234
Aborted by signal Terminated...
test@cd-apl-up2-pcm512x-02:~/test$ aplay -Dhw:0,0 -fS16_LE -c2 -r48000 /dev/zero
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
```

***Environment***
sof master: 99e9f43
kernel sof-dev: 90860434
tplg: sof-apl-pcm512x.tplg

lgirdwood, 2019-04-18:
@Jiangxinx can you log this as a kernel bug, link to this one and close this issue.

@plbossart @ranj063 @kv2019i this looks like a kernel race where aplay will hang on to the PCM (and keep it BUSY), it could be that PM works when the aplay PCM is in playing state but may not when the PCM is in hw_free(), hw_params() or prepare() state ? This may prevent aplay from close()ing the PCM device.

The log below shows driver stopping the stream, but do we check the PCM state prior to STOP ?

```
[   58.574535] Freezing remaining freezable tasks ... (elapsed 0.000 seconds) done.
[   58.575502] printk: Suspending console(s) (use no_console_suspend to debug)
[   58.590994] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 5
[   58.591009] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   58.591426] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   58.594467] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   58.594824] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   58.594850] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
[   58.594855] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1000303 successful
[   58.594860] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 3
[   58.594883] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000044
[   58.594887] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000010 at  00000044
[   58.595695] r8169 0000:02:00.0 enp2s0: Link is Down
[   58.768699] ACPI: Preparing to enter system sleep state S3

...snip...

[   59.031894] Restarting tasks ... done.
[   59.040936] video LNXVIDEO:00: Restoring backlight state
[   59.040950] PM: suspend exit
[   59.047158] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 6
[   59.047164] sof-audio-pci 0000:00:0e.0: PCM: resume stream 0 dir 0 DMA position 0
[   59.047166] sof-audio-pci 0000:00:0e.0: pcm: hw params stream 0 dir 0
[   59.047187] sof-audio-pci 0000:00:0e.0: period_bytes:0x3fc0
[   59.047189] sof-audio-pci 0000:00:0e.0: periods:4
[   59.047198] sof-audio-pci 0000:00:0e.0: stream_tag 1
[   59.047205] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   59.047358] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   59.047360] sof-audio-pci 0000:00:0e.0: pcm: stream dir 0, posn mailbox offset is 0xc1000
[   59.047366] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   59.047569] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   59.047622] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 0
[   59.047628] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   59.047857] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   59.059982] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 1
[   59.059997] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   59.060181] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[   59.069045] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   59.069055] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   59.080160] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   59.080174] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   59.090162] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   59.090194] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   59.100156] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   59.100182] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG

...snip...

[   60.740155] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   60.750133] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   60.750147] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   60.760131] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   60.760145] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   60.770136] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   60.770151] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   60.776875] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 0
[   60.776895] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   60.777248] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   60.777300] sof-audio-pci 0000:00:0e.0: pcm: free stream 0 dir 0
[   60.777307] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   60.777487] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   60.779018] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   60.779032] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   60.792661] sof-audio-pci 0000:00:0e.0: pcm: close stream 0 dir 0
[   61.051879] r8169 0000:02:00.0 enp2s0: Link is Up - 100Mbps/Full - flow control rx/tx
[   61.278895] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   61.278904] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   63.019792] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[... cut ...]
[   63.020173] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
[   63.020179] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1000303 successful
[   63.020185] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 3
[   63.020208] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000044
[   63.020214] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000010 at  00000044
[  317.735741] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
```

Jiangxinx, 2019-04-19:
Close this as duplicated with https://github.com/thesofproject/linux/issues/848#issue-435097382

closed 2019-04-19 by Jiangxinx