thesofproject/sof #1250: [BUG][BYT]Can not aplay on BYT-MB-NOCODEC platform.

Jiangxinx, 2019-04-10:
**Describe the bug**
Fail to aplay on BYT-MB-NOCODEC .
Sof-logger have no abvious error trace.

**To Reproduce**
aplay -Dhw:0,0 -fs16_le -c2 -r 48000 /dev/zero

**Environment**
sof master: 18d116b
kernel sof-dev: b5ece71d
tplg: sof-byt-nocodec.tplg

**Screenshots or console output**
```
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 85000
PERIOD_SIZE: 4080
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 341000
BUFFER_SIZE: 16368
BUFFER_BYTES: 65472
TICK_TIME: 0
```

***Dmesg***
```
[   17.358189] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   17.358248] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   17.358338] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   17.538938] systemd-udevd (1456) used greatest stack depth: 12048 bytes left
[   17.798813] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[   17.798825] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   18.882436] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   18.884577] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   18.884721] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   18.886512] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   18.886598] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   18.888300] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   18.888382] sof-audio-acpi 80860F28:01: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   18.888488] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   19.299254] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[   19.299269] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   19.799424] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[   19.799442] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   22.985954] systemd-journald[1396]: File /var/log/journal/ddf9fc7e80cb45a58f798fb83423d895/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   24.520520] igb 0000:03:00.0 enp3s0: igb: enp3s0 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
[   24.521228] IPv6: ADDRCONF(NETDEV_CHANGE): enp3s0: link becomes ready
[   58.709869] sof-audio-acpi 80860F28:01: pcm: open stream 0 dir 0
[   58.709876] sof-audio-acpi 80860F28:01: period min 192 max 16384 bytes
[   58.709880] sof-audio-acpi 80860F28:01: period count 2 max 16
[   58.709883] sof-audio-acpi 80860F28:01: buffer max 65536 bytes
[   58.710287] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[   58.710292] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[   58.710298] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[   58.710301] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[   58.710305] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[   58.710309] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[   58.710315] sof-audio-acpi 80860F28:01: pcm: hw params stream 0 dir 0
[   58.710322] sof-audio-acpi 80860F28:01: generating page table for 000000003430876b size 0xffc0 pages 16
[   58.710326] sof-audio-acpi 80860F28:01: stream_tag 0
[   58.710337] sof-audio-acpi 80860F28:01: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   58.710713] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   58.710719] sof-audio-acpi 80860F28:01: pcm: stream dir 0, posn mailbox offset is 1329664
[   58.710766] sof-audio-acpi 80860F28:01: ASoC: platform prepare error: -22
[   58.710769]  NoCodec-2: ASoC: backend prepare failed -22
[   58.711608] sof-audio-acpi 80860F28:01: pcm: free stream 0 dir 0
[   58.711616] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   58.712510] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   58.712596] sof-audio-acpi 80860F28:01: pcm: close stream 0 dir 0
[   58.810477] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[   58.810499] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

***Log file***
[dmesg.log](https://github.com/thesofproject/sof/files/3063244/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3063245/logger.log)

tlauda, 2019-04-10:
@Jiangxinx Not FW issue, works with older kernel. Please create the issue in the linux repo.

xiulipan, 2019-04-11:
@tlauda @plbossart @ranj063 
Reopen this issue in SOF since it maybe related with tplg

xiulipan, 2019-04-11:
The issue happens because we would like to find some pcm from id. but got some mismatch in sof-byt-nocodec.tplg

juimonen, 2019-04-11:
@tlauda what kernel sha1 this is working? I can bisect...

xiulipan, 2019-04-11:
@juimonen Can be fixed with https://github.com/thesofproject/linux/pull/816

But still need confirm that no tplg change is needed.

xiulipan, 2019-04-11:
close via https://github.com/thesofproject/linux/pull/816 is merged and confirmed by @ranj063 no spcm for BE.

closed 2019-04-11 by xiulipan