thesofproject/sof #1251: [BUG][WHL HDA] incorrect sound when playback

xun2z, 2019-04-10:
### Description
The playback is not right with the clock frequency on both WHL RVP and laptop.

### Cause
commit: [cavs: ssp: generic ssp mclk configuration for cavs ](https://github.com/thesofproject/sof/commit/f5e78430ff3dbbe8bcceca5f0f4f1fa86bbd78b8) is the cause. 
Playback is OK when this commit is reverted.

### To Reproduce
**kernel**
(since recent regression, I prefer to use this branch): [rander/cml-validation-branch](https://github.com/RanderWang/linux/tree/cml-validation-branch)
[regression #1: linux/issues/767 ](https://github.com/thesofproject/linux/issues/767) (this can be fixed by former branch)
[regression #2: linux/issues/774 ](https://github.com/thesofproject/linux/issues/774) (this still exists, I do a work-around to delete the /usr/bin/pulseaudio)

**firmware**
master newest commit: ed6ac05f6b884b5aa5597bf739fc97527bdab909

### Impact
this blocks our release on whl.

bkokoszx, 2019-04-10:
I believe that setting XTAL clock as a default clock on CNL will fix the problem (#1246). I will try also investigate what is a problem with PLL_FIXED (96 MHz) clock - I'm a little confused, because master/slave ssp with this clock passed.

mengdonglin, 2019-04-10:
@xun2z Please share test steps. And please share kmesg and sof trace if there is any error.

bkokoszx, 2019-04-12:
@mengdonglin @xun2z 
@zrombel has checked this issue with #1246 patch on our python tests and regression does not exist.

@xun2z 
Could you confirm that applying #1246 fixes this issue?

closed 2019-04-12 by mmaka1

xun2z, 2019-04-15:
confirm fixed on our HDA device

PR #1246 (2019-04-12): dmic: fix dmic initial ramp regression
Set XTAL_OSCILLATOR clock as a default clock on CNL.
Setting PLL_FIXED clock as a default clock causes very long DMIC initial ramp.

[image]

Also fixes #1251