thesofproject/sof #1254: [BUG][CML] topology load failed sometimes when pulseaudio active. [2/10]

stevyan, 2019-04-11:
**Describe the bug**
topology load failed sometimes when pulseaudio active;

**To Reproduce**
aplay -l

**Expected behavior**
list all the playback pcm pipelines;

**Actual behavior**
no sound card found

**Environment**
Kernel: 55f5a2ec (rander/cml-validation-branch)
Firmware: ed6ac05 (with reverted commit f5e78430f)
Topology: 0efa03d8(rander/cml-validation-branch)
Reproducibility Rate: 2/10

**dmesg log**
```
[    9.540286] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    9.844356] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[    9.844455] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[    9.844744] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[    9.844821] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[    9.844889] sof-audio-pci 0000:00:1f.3: error: failed to set DAI config for direction:0 of HDA dai 2
[    9.844972] sof-audio-pci 0000:00:1f.3: error: failed to process hda dai link Digital Playback and Capture
[    9.845060] sof-audio-pci 0000:00:1f.3: ASoC: physical link loading failed
[    9.845203] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[    9.845209] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845212] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.845220] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845225] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845228] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.845235] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845240] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845243] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.845251] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845256] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845259] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.845267] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845285] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845289] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.845297] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845302] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845305] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.845314] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845319] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845322] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.845330] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845335] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845338] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.845346] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845351] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.845354] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.845360] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -110
[    9.845438] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[    9.845506] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[    9.845655] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -22
[    9.845790] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22
[   11.984003] systemd-udevd (1711) used greatest stack depth: 11240 bytes left
[   37.964964] iwlwifi 0000:00:14.3: BIOS contains WGDS but no WRDS
[   39.624948] ax88179_178a 2-2:1.0 enx000ec6a1e622: ax88179 - Link status is: 1
[   39.632078] IPv6: ADDRCONF(NETDEV_CHANGE): enx000ec6a1e622: link becomes ready
[   52.948792] rfkill: input handler disabled
```
[dmesg_topology_load_failed.log](https://github.com/thesofproject/sof/files/3067870/module_load_failed.log)

tlauda, 2019-04-11:
@stevyan It's IPC timeout, so FW logs are needed.

stevyan, 2019-04-12:
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      2          DMA             98479.739583     98479.742188 intel/cavs/hda-dma.c:480 	hda-dmac: 4 channel 0 -> get
    0      2          DMA             98645.729167       165.989578 intel/cavs/hda-dma.c:703 	hda-dmac: 4 channel 0 -> config
    0      2          DMA             98691.354167        45.625000 intel/cavs/hda-dma.c:549 	hda-dmac: 4 channel 0 -> start
    0      2          DMA             98694.843750         3.489583 intel/cavs/hda-dma.c:396 	hda-dmac: 4 channel 0 -> enable
    0      2          IPC             99383.229167       688.385437    src/ipc/handler.c:926 	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC             99386.614583         3.385417    src/ipc/handler.c:929 	ipc: pipe 1 comp 0 -> new (type 1)
    0      2         HOST             99391.770833         5.156250     src/audio/host.c:348 	host_new()
    0      2          DMA             99407.083333        15.312500 intel/cavs/hda-dma.c:840 	hda-dmac :5 -> probe
    0      2          DMA             99423.958333        16.875000        src/lib/dma.c:142 	dma_get() ID 5 sref = 1 busy channels 0
    0      2          IPC             99543.489583       119.531250    src/ipc/handler.c:926 	ipc: hdr 0x30010000 tx (84) > rx (28)
    0      2          IPC             99546.718750         3.229167    src/ipc/handler.c:929 	ipc: pipe 1 comp 1 -> new (type 5)
    0      2       VOLUME             99551.197917         4.479167   src/audio/volume.c:178 	volume_new()
    0      2          IPC             99661.666667       110.468750    src/ipc/handler.c:959 	ipc: pipe 1 buffer 2 -> new (0x300 bytes)
    0      2       BUFFER             99665.937500         4.270833   src/audio/buffer.c:54  	buffer_new()
    0      2          IPC             99802.968750       137.031250    src/ipc/handler.c:959 	ipc: pipe 1 buffer 3 -> new (0x300 bytes)
    0      2       BUFFER             99807.239583         4.270833   src/audio/buffer.c:54  	buffer_new()
    0      2          IPC             99959.218750       151.979172    src/ipc/handler.c:926 	ipc: hdr 0x30010000 tx (80) > rx (28)
    0      2          IPC             99962.604167         3.385417    src/ipc/handler.c:929 	ipc: pipe 1 comp 4 -> new (type 2)
    0      2          DAI             99967.916667         5.312500      src/audio/dai.c:203 	dai_new()
    0      2          DAI             99981.666667        13.750000        src/lib/dai.c:87  	dai_get(), d = 0xbe05d0a0, sref = 1
    0      2          DMA             99985.729167         4.062500 intel/cavs/hda-dma.c:840 	hda-dmac :7 -> probe
    0      2          DMA            100002.187500        16.458334        src/lib/dma.c:142 	dma_get() ID 7 sref = 1 busy channels 0
    0      2          IPC            100107.500000       105.312500    src/ipc/handler.c:987 	ipc: pipe 1 -> new
    0      2         PIPE            100114.479167         6.979167 src/audio/pipeline.c:72  	pipeline_new()
    0      2          IPC            100560.625000       446.145844    src/ipc/handler.c:660 	ipc: pm core mask 0x1 -> enable
    0      2          IPC            100946.302083       385.677094    src/ipc/handler.c:926 	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC            100949.531250         3.229167    src/ipc/handler.c:929 	ipc: pipe 2 comp 7 -> new (type 1)
    0      2         HOST            100955.520833         5.989583     src/audio/host.c:348 	host_new()
    0      2          DMA            100970.729167        15.208333        src/lib/dma.c:142 	dma_get() ID 4 sref = 2 busy channels 1
    0      2          IPC            101717.291667       746.562500    src/ipc/handler.c:926 	ipc: hdr 0x30010000 tx (84) > rx (28)
    0      2          IPC            101720.625000         3.333333    src/ipc/handler.c:929 	ipc: pipe 2 comp 8 -> new (type 5)
    0      2       VOLUME            101726.718750         6.093750   src/audio/volume.c:178 	volume_new()
    0      2          IPC            102074.010417       347.291656    src/ipc/handler.c:959 	ipc: pipe 2 buffer 9 -> new (0x300 bytes)
    0      2       BUFFER            171994.739583     69920.726562   src/audio/buffer.c:54  	buffer_new()

tlauda, 2019-04-12:
@stevyan Does this issue reproduce with topology from master without any extra patches?

emilchudzik, 2019-04-15:
In my env I have properly loaded topology each time, but issue with pulse audio exists:

**Environment:**
Platform: WHL RVP - HDA mode
**Kernel:** 
2de29fae6 [thesofproject/linux: topic/sof-dev]
+d04ccf494 [RanderWang/linux: cml-validation-20190415]
+bfb0729dd [RanderWang/linux: cml-validation-20190415]
+8a2b13a81 [RanderWang/linux: cml-validation-20190415]
+63ceb8f68 [RanderWang/linux: cml-validation-20190415]
+db89510d0 [RanderWang/linux: cml-validation-20190415]

**Firmware:** 00bbb113d [thesofproject/sof: master]
**Topology:** 00bbb11 [thesofproject/sof: master] -> sof-hda-generic.tplg

When I **disabled pulseaudio**, I have only 1/10 issue while second playback start:
[image]

currently after reboot i have no issue, I can play each time.

Whan I have **enabled pulsuaudio** I have many erros with timeout. 
Logs here: https://github.com/thesofproject/sof/issues/1263#issuecomment-483196594

keyonjie, 2019-04-15:
Today I setup this and can duplicate the issue also, I also managed to create simultaneous 2 playback streams to duplicate the ipc timeout issue, will do advanced debugging tomorrow.

emilchudzik, 2019-04-16:
I have no issue with #PR 1269 applied on the current sof master branch without reverting any others commits. Without this PR I had an erros even during boot. Currently simple aplay and arecord works, also in loop.

**Environment:**
Platform: WHL RVP - HDA mode
**Kernel:** 
2de29fae6 [thesofproject/linux: topic/sof-dev]
+d04ccf494 [RanderWang/linux: cml-validation-20190415]
+bfb0729dd [RanderWang/linux: cml-validation-20190415]
+8a2b13a81 [RanderWang/linux: cml-validation-20190415]
+63ceb8f68 [RanderWang/linux: cml-validation-20190415]
+db89510d0 [RanderWang/linux: cml-validation-20190415]

**Firmware:** 1b15fad [thesofproject/sof: master] + https://github.com/thesofproject/sof/pull/1269/commits/624c18935589b4f87bc4a058959ef5d53ac26882
**Topology:** 1b15fad [thesofproject/sof: master] -> sof-hda-generic.tplg

emilchudzik, 2019-04-19:
I just checked WHL RVP with HDA mode. 
10/10 playback works, there is no issue when pulseaudio is enabled with loading topology also there is no IPC timeouts with environment from this comments: https://github.com/thesofproject/sof/issues/1282#issue-435017798

keyonjie, 2019-04-20:
let's validate if https://github.com/thesofproject/linux/pull/850 can benefit or even fix this issue.

ClarexZhou, 2019-04-22:
Error still occurred after 44 times system reboot to check ipc timeout.
Total fail 15 times after reboot 44 times.
3 times:
```
[    9.125524] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[    9.125528] sof-audio-pci 0000:00:1f.3: tplg: 1 hw_configs found, default id: 1!
[    9.125544] sof-audio-pci 0000:00:1f.3: link dma channel 2 for playback
[    9.125556] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    9.427434] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[    9.427539] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[    9.427901] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[    9.427978] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[    9.428052] sof-audio-pci 0000:00:1f.3: error: failed to set DAI config for direction:0 of HDA dai 4
[    9.428135] sof-audio-pci 0000:00:1f.3: error: failed to process hda dai link iDisp1
[    9.428208] sof-audio-pci 0000:00:1f.3: ASoC: physical link loading failed
[    9.428396] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[    9.428406] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[    9.428413] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.428427] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428435] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428440] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.428449] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428460] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428464] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.428475] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428486] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428490] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.428502] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428513] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428517] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.428529] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428540] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428544] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.428556] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428567] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428571] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.428582] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428593] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428597] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.428608] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428619] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[    9.428623] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[    9.428630] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -110
[    9.428716] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[    9.431133] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[    9.433619] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: failed to instantiate card -22
[    9.436184] skl_hda_dsp_generic: probe of skl_hda_dsp_generic failed with error -22
[    9.772730] ax88179_178
```
2 times:
```
[   19.812271] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   19.812292] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   19.812507] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   19.812519] sof-audio-pci 0000:00:1f.3: ipc tx: 0x50010000: GLB_COMP_MSG: SET_VALUE: SET_VALUE
[   19.812899] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[   19.812922] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   20.117346] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x50010000 size 108
[   20.117360] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   20.117576] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   20.117580] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   20.117584] sof-audio-pci 0000:00:1f.3: error: set/get ctrl ipc comp 1
[   20.117588] sof-audio-pci 0000:00:1f.3: error: failed kcontrol value set for widget: 1
[   20.117591] sof-audio-pci 0000:00:1f.3: error: restoring kcontrols after resume
[   20.117595] sof-audio-pci 0000:00:1f.3: error: failed to restore pipeline after resume -110
[   20.117601] sof-audio-pci 0000:00:1f.3: error: volume put failed to resume -110
[   20.117736] sof-audio-pci 0000:00:1f.3: error: volume put failed to resume -22
[   20.117776] sof-audio-pci 0000:00:1f.3: error: volume put failed to resume -22
[   20.117811] sof-audio-pci 0000:00:1f.3: error: volume put failed to resume -22
[   20.117860] sof-audio-pci 0000:00:1f.3: error: volume put failed to resume -22
[... 312 lines cut ...]
Made from:
RanderWang/linux@2de29fa

RanderWang/linux@d04ccf4
RanderWang/linux@bfb0729
RanderWang/linux@8a2b13a
RanderWang/linux@63ceb8f
RanderWang/linux@db89510
RanderWang/linux@066e6f4
RanderWang/linux@bfb3fcb
RanderWang/linux@edc012c
RanderWang/linux@a9f0431
RanderWang/linux@d5113b0
RanderWang/linux@aeccfd5
RanderWang/linux@22999bd
RanderWang/linux@c827ffb
Sof Repo: https://github.com/bardliao/sof/tree/whl-daliy-20190419
Made from: bardliao@ 4c6f6f7

bardliao/sof@90a6efd
bardliao/sof@5ef0713
bardliao/sof@50bd96d
bardliao/sof@fdf1eda
bardliao/sof@f135dd9
bardliao/sof@e8d5622

ClarexZhou, 2019-04-24:
arecord start fail for IPC timeout occurred on CML laptop.

Test Recipe 

Kernel repo | https://github.com/RanderWang/linux/tree/cml-validation-20190424
  Made from:
  2e94569 [thesofproject/linux: topic/sof-dev]
+ 85a86d9 [RanderWang/linux: cml-validation-20190424]
+ b51284b[RanderWang/linux: cml-validation-20190424]
+ 8f41f15 [RanderWang/linux: cml-validation-20190424]
+ 202216f [RanderWang/linux: cml-validation-20190424]
+ ce16096 [RanderWang/linux: cml-validation-20190424]
+ 4459c97 [RanderWang/linux: cml-validation-20190424]
+ 4f4bfb1 [RanderWang/linux: cml-validation-20190424]
+ 0bc8e92 [RanderWang/linux: cml-validation-20190424]
+ 4de6c77 [RanderWang/linux: cml-validation-20190424]
+ 7ec78da [RanderWang/linux: cml-validation-20190424]
+ 30d5691 [RanderWang/linux: cml-validation-20190424]
+ 98d6916 [RanderWang/linux: cml-validation-20190424]
  
Firmware repo | https://github.com/RanderWang/linux/tree/cml-validation-20190424
  Made from:
  0c932613  [thesofproject/sof: master]
+  e3ade00 [RanderWang/sof: cml-linux-20190424]
+  08d40fc [RanderWang/sof: cml-linux-20190424]
```
[arcordStartFailForIPCtimeout_logger.log](https://github.com/thesofproject/sof/files/3111287/arcordStartFailForIPCtimeout_logger.log)
[arecordStartFailForIPCtimeout_Dmesg.log](https://github.com/thesofproject/sof/files/3111288/arecordStartFailForIPCtimeout_Dmesg.log)

[  842.637589] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  842.637598] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  842.637693] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30200000: GLB_TPLG_MSG: BUFFER_NEW
[  842.637705] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  842.637822] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[  842.637832] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[  842.637931] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[  842.637940] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[  842.637946] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[  842.942620] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40070000 size 12
[  842.942638] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  842.942867] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  842.942870] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  842.942878] sof-audio-pci 0000:00:1f.3: error: core enable ipc failure
[  842.942885] sof-audio-pci 0000:00:1f.3: error: failed to load widget type 32 with ID: 12
[  842.942889] sof-audio-pci 0000:00:1f.3: error: failed to restore pipeline after resume -110
[  842.942911] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 1
[  842.942914] sof-audio-pci 0000:00:1f.3: error: pcm open failed to resume -22
[  842.942918] sof-audio-pci 0000:00:1f.3: ASoC: can't open component 0000:00:1f.3: -22
```

bkokoszx, 2019-04-24:
@ClarexZhou 
Did you build topology from:
0c93261 [thesofproject/sof: master]?

I've just checked this issue by building FW, kernel and topology from branches you mentioned above: https://github.com/thesofproject/sof/issues/1254#issuecomment-486122235
and I could not reproduce issue. On the other side I have following dmesg issues:
```
[    7.872315] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    7.872316] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 1
[    7.872317] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[    7.872318]  HDA Analog: ASoC: hw_params FE failed -22
```

With @emilchudzik we checked this issue with kernel:
https://github.com/RanderWang/linux/tree/cml-validation-20190418
and didn't have reproduction. Could you confirm that?

ClarexZhou, 2019-04-25:
My mistake. update firmware info as below. And that IPC timeout only occurred one time during testing.

Firmware repo | https://github.com/RanderWang/sof/tree/cml-linux-20190424
  Made from:
  e4c1651  [thesofproject/sof: master]
+ e3ade00 [RanderWang/sof: cml-linux-20190424]
+ 08d40fc [RanderWang/sof: cml-linux-20190424]

Topology: | https://github.com/RanderWang/sof/tree/cml-linux-20190424 | Same as above SOF repo
  file:  tools/topology/sof-hda-generic.m4,  using sof-hda-generic-2ch.tplg

keyonjie, 2019-04-25:
@ClarexZhou as it is not easy to reproduce this kind of ipc time out issues now, can you please help capture and paste sof-logger error logs also for each fails, that is very valuable for debugging.

ClarexZhou, 2019-04-25:
Error occurred on 20190425 build after aplay.
```
  323.599475] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 0 dir 0
[  323.599481] sof-audio-pci 0000:00:1f.3: generating page table for 00000000e8377743 size 0xffc0 pages 16
[  323.599529] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  323.599531] sof-audio-pci 0000:00:1f.3: periods:4
[  323.599566] sof-audio-pci 0000:00:1f.3: stream_tag 1
[  323.599580] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  323.599856] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  323.599865] sof-audio-pci 0000:00:1f.3: pcm: stream dir 0, posn mailbox offset is 790528
[  330.620513] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[  330.620535] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  330.926930] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60040000 size 12
[  330.926954] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  330.927280] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  330.927285] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  330.927295]  HDA Analog: ASoC: trigger FE failed -110
[  330.927309] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  330.927331] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  330.927464] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  330.927478] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  330.927518] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  330.927573] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  330.927608] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  331.414796] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  331.414824] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  331.914757] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  331.914783] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

[ipcTimeOutAfterAplay_dmesg.log](https://github.com/thesofproject/sof/files/3115346/ipcTimeOutAfterAplay_dmesg.log)
[ipcTimeOutAfterAplay_logger.log](https://github.com/thesofproject/sof/files/3115347/ipcTimeOutAfterAplay_logger.log)

**Test Recipe**
Kernel repo | https://github.com/RanderWang/linux/tree/cml-validation-20190425

  Made from:
   f7e9d4a [thesofproject/linux: topic/sof-dev]
+  8a8e045 [RanderWang/linux: cml-validation-20190425]
+  62eaa40 [RanderWang/linux: cml-validation-20190425]
+  bf461fd [RanderWang/linux: cml-validation-20190425]
+  49bb879 [RanderWang/linux: cml-validation-20190425]
+  6af82dc [RanderWang/linux: cml-validation-20190425]
+  e54f292 [RanderWang/linux: cml-validation-20190425]
+  e026573 [RanderWang/linux: cml-validation-20190425]
+  3bdeae3 [RanderWang/linux: cml-validation-20190425]
+  cea87d5 [RanderWang/linux: cml-validation-20190425]
  
Firmware repo | https://github.com/RanderWang/sof/tree/cml-linux-20190425
   Made from:
   395a34e  [thesofproject/sof: master]
+  279c995b [RanderWang/sof: cml-linux-20190425]
+  8598799 [RanderWang/sof: cml-linux-20190425]
    
Topology: | https://github.com/RanderWang/sof/tree/cml-linux-20190425
   file:  tools/topology/sof-hda-generic.m4,  sof-hda-generic-2ch.tplg
    
Kconfig: | https://sof-ci-storage.sh.intel.com/daily/20190425/linux-2e945691-1/config-daily-hda-20190425

tlauda, 2019-04-25:
@ClarexZhou Please reverify this trigger start timeout with #1289.

keyonjie, 2019-04-25:
thanks @tlauda , the PR looks making sense to me. the difficulty here we don't have constant pattern to duplicate the same issue.

ClarexZhou, 2019-04-26:
Trigger start Time out issue is not observed with #1289 after about 60 times system reboot on CML laptop.  Keep monitor.
Only below timeout occurred once.  No error message in sof-logger
```
[   16.227693] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[   16.227700] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   16.227804] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   16.227807] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[   16.227825] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[   16.227827] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   16.227938] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   16.227974] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[   35.508916] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[   35.508968] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   35.543749] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   35.846825] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[   35.846845] sof-audio-pci 0000:00:1f.3: unknown ROM status value ffffffff
[   35.847118] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xffffffff trace 0xffffffff
[   35.847123] sof-audio-pci 0000:00:1f.3: error: ctx_save ipc error during suspend -110
[   35.847137] sof-audio-pci 0000:00:1f.3: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -110)
[0425+1289dmesg_failed_48.log](https://github.com/thesofproject/sof/files/3120095/0425%2B1289dmesg_failed_48.log)

```

keyonjie, 2019-04-26:
Thanks @ClarexZhou this is another type of error we hit: CTX_SAVE(suspend) IPC. so it looks to be quite random, different kind of IPCs can be timed out at different cases.

emilchudzik, 2019-04-26:
My suggestion is to create uniqe issue for each IPC timeout, with detailed test flow/step to reproduce/detailed description of environment.
If it will be generic issue it will be hard to debug and reproduce and it will become a sack for every IPC timeout.

ranj063, 2019-04-26:
@keqiaozhang @ClarexZhou have you tested this with the fixes applied for #1263?

ClarexZhou, 2019-04-28:
ipc time out still occurred with  #1263 fix on CML laptop

Log are disabled on dmesg on current build. No error log on sof-logger.
[logger.log](https://github.com/thesofproject/sof/files/3124642/logger.log)

Kernel repo | https://github.com/bardliao/linux/tree/sof-dev-rebase-reduce-v3
-- | --
  |  
Firmware repo | https://github.com/thesofproject/sof/commits/master
  | be1331a  [thesofproject/sof: master]
  |  
Topology: | https://github.com/thesofproject/sof/commits/master
  | file:  tools/topology/sof-hda-generic.m4,  sof-hda-generic-2ch.tplg

lgirdwood, 2019-04-28:
@ClarexZhou please enable dmesg log and try again. Please also state if fixes have improved failure rate from initial state.

keyonjie, 2019-04-28:
ack to @lgirdwood , without dmesg log, we can't analyze anything from driver side.

btw, for the topology parsing and pipe_connect or enable_core IPC time out issue, I created a RFC driver PR, can you help check if it helps? https://github.com/thesofproject/linux/pull/888

ranj063, 2019-04-28:
@keyonjie if the timeout is always with the CTX_SAVE ipc, we can simply ignore the timeout in the driver andjust power down the DSP in any case. We used to have this earlier but removed it recently when we fixed the power down sequence in GLK.

@mmaka1 could you please double check the power down sequence for CNL to make sure we didnt miss anything?

ClarexZhou, 2019-04-29:
Add 2 more ipc timed out logs on CML laptop:
1. ipc timetout during multi pipeline test
no error sof logger. 
[ipctimeout_multiPipeline_dmesg.log](https://github.com/thesofproject/sof/files/3125732/ipctimeout_multiPipeline_dmesg.log)

```
[    3.463462] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[    3.478008] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[    4.490375] systemd-udevd (1664) used greatest stack depth: 11504 bytes left
[    5.022408] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[    5.031547] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[    6.729078] ax88179_178a 1-3:1.0 enx000ec6a1e569: ax88179 - Link status is: 1
[    6.738230] IPv6: ADDRCONF(NETDEV_CHANGE): enx000ec6a1e569: link becomes ready
[    6.770838] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    6.771022] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    6.771238] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    6.771452] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    6.771634] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    6.771822] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    6.771997] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    6.772209] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    6.772436] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    9.275017] rfkill: input handler disabled
[   28.084128] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   28.167629] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   40.161181] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   40.161200] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   40.161386] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   40.161391] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   40.161403]  HDMI1: ASoC: trigger FE failed -110
[   40.465442] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   40.465469] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   40.465765] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   40.465769] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   40.769435] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   40.769456] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   40.769686] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   40.769691] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   40.769708]  HDA Analog: ASoC: trigger FE failed -110
[   41.073325] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   41.073344] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   41.073440] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   41.073444] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
```
2. ipc timeoud when start speaker-test: speaker-test -Dhw:0,0 -r48000 -f1000 -c2 -t wav -l 10
no error sof logger info  
[speaker-test-logger.log](https://github.com/thesofproject/sof/files/3125734/speaker-test-logger.log)
[speaker-test-dmesg.log](https://github.com/thesofproject/sof/files/3125735/speaker-test-dmesg.log)

```
[    5.520215] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    6.943460] ax88179_178a 1-3:1.0 enx000ec6a1e569: ax88179 - Link status is: 1
[    6.951870] IPv6: ADDRCONF(NETDEV_CHANGE): enx000ec6a1e569: link becomes ready
[    7.105157] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    7.105439] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    7.105705] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    7.105951] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    7.106161] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    7.106372] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    7.106651] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    7.106914] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    7.107132] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[   10.657116] rfkill: input handler disabled
[   28.427115] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   28.510675] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   56.479136] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   56.569053] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   74.732600] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   74.816129] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  101.152722] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[... 11 lines cut ...]
[  320.283306] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  320.372488] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  348.519904] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[  348.520200] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xffffffff trace 0xffffffff
[... cut ...]
[  549.219708] sof-audio-pci 0000:00:1f.3: error: pcm open failed to resume -22
[  549.219720] sof-audio-pci 0000:00:1f.3: ASoC: can't open component 0000:00:1f.3: -22
[  549.219728]  HDA Analog: ASoC: failed to start FE -22
```

Kernel repo | https://github.com/bardliao/linux/tree/sof-dev-rebase-reduce-v4
-- | --
  |  
Firmware repo | https://github.com/thesofproject/sof/commits/master
  | be1331a  [thesofproject/sof: master]
  |  
Topology: | https://github.com/thesofproject/sof/commits/master
  | file:  tools/topology/sof-hda-generic.m4,  sof-hda-generic-2ch.tplg

ClarexZhou, 2019-04-29:
Try reboot 100 times on CML laptop. Only twice ipc timeout.
[dmesg_failed_56.log](https://github.com/thesofproject/sof/files/3125755/dmesg_failed_56.log)
[logger_failed_56.log](https://github.com/thesofproject/sof/files/3125756/logger_failed_56.log)
[logger-t_failed_56.log](https://github.com/thesofproject/sof/files/3125757/logger-t_failed_56.log)

```
[    2.467997] iwlwifi 0000:00:14.3: BIOS contains WGDS but no WRDS
[    3.264407] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[    3.280020] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[    4.887228] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[    4.902907] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[    5.099145] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    5.188759] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    5.761912] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    5.762197] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    5.762422] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    5.762711] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    5.763008] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    5.763254] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    5.763490] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    5.763810] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    5.764111] sof-audio-pci 0000:00:1f.3: error: volume get failed to idle -16
[    7.855730] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[    7.855749] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[    7.855978] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[    7.855982] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[    8.078423] usb 1-6: new high-speed USB device number 5 using xhci_hcd
[    8.227325] usb 1-6: New USB device found, idVendor=0bda, idProduct=5538, bcdDevice=88.07
[    8.227326] usb 1-6: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[    8.227327] usb 1-6: Product: Integrated_Webcam_HD
[    8.227328] usb 1-6: Manufacturer: CKFHH66O4634000251A0
[    8.227328] usb 1-6: SerialNumber: 200901010001
[    8.343423] usb 1-10: new full-speed USB device number 6 using xhci_hcd
[    8.471350] usb 1-10: New USB device found, idVendor=8087, idProduct=0aaa, bcdDevice= 0.02
[    8.471351] usb 1-10: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    8.592633] rfkill: input handler disabled
[    8.813054] ax88179_178a 1-3:1.0 eth0: register 'ax88179_178a' at usb-0000:00:14.0-3, ASIX AX88179 USB 3.0 Gigabit Ethernet, 00:0e:c6:a1:e5:69
[    8.813099] usbcore: registered new interface driver ax88179_178a
[    8.816923] ax88179_178a 1-3:1.0 enx000ec6a1e569: renamed from eth0
[    9.160236] systemd-udevd (1676) used greatest stack depth: 11632 bytes left
[   12.247791] ax88179_178a 1-3:1.0 enx000ec6a1e569: ax88179 - Link status is: 1
[   12.255779] IPv6: ADDRCONF(NETDEV_CHANGE): enx000ec6a1e569: link becomes ready
[   13.175424] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   13.175432] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   13.175649] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   13.175651] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   13.479445] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   13.479457] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   13.479681] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   13.479682] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   13.479685]  HDA Analog: ASoC: trigger FE failed -110
[   13.783486] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   13.783499] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   13.783746] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   13.783748] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   16.183867] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[   16.183884] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   16.184153] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   16.184154] sof-audio-pci 0000:00:1f.3: error: ctx_save ipc error during suspend -110
[   16.184159] sof-audio-pci 0000:00:1f.3: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -110)
```
failed sof-logger:
```
    0      1         PIPE 1.5      10102700.260417         6.770833 src/audio/pipeline.c:844 	pipeline_xrun_recover()
    0      1          DMA          10104145.000000      1444.739624 intel/cavs/hda-dma.c:279 	hda-dmac: 5 wait for buffer full timeout
    0      1         HOST          10104151.406250         6.406250     src/audio/host.c:792 	host_copy() error: dma_copy() failed, ret = 4294967234
    0      1         PIPE          10104157.656250         6.250000 src/audio/pipeline.c:736 	pipeline_copy() error: ret = -62, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5      10104164.114583         6.458333 src/audio/pipeline.c:844 	pipeline_xrun_recover()
[... cut ...]
[   33.432269] sof-audio-pci 0000:00:1f.3: error: ctx_save ipc error during suspend -110
[   33.432283] sof-audio-pci 0000:00:1f.3: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -110)

```

Kernel repo | https://github.com/bardliao/linux/tree/sof-dev-rebase-reduce-v3
-- | --
  |  
Firmware repo | https://github.com/thesofproject/sof/commits/master
  | be1331a  [thesofproject/sof: master]
  |  
Topology: | https://github.com/thesofproject/sof/commits/master
  | file:  tools/topology/sof-hda-generic.m4,  sof-hda-generic-2ch.tplg
```

keyonjie, 2019-04-29:
it's tricky that we still seeing this error "hda-dmac: 5 wait for buffer full timeout" in sof-logger.
@tlauda can anybody from FW team help on this?

emilchudzik, 2019-04-29:
My suggestion is to create uniqe issue for each IPC timeout, with detailed test flow/step to reproduce/detailed description of environment.
If it will be generic issue it will be hard to debug and reproduce and it will become a sack for every IPC timeout.

@mengdonglin please verify if issue still exists and close if doesn't.

keyonjie, 2019-04-30:
we might need to figure out https://github.com/thesofproject/linux/issues/724 for debugging this kind of issue with linux driver.
@wenqingfu Michael, can anyone helps on that during PRC team having 4 days' holiday here.

ClarexZhou, 2019-05-08:
0x40010000: GLB_PM_MSG: CTX_SAVE  IPC timeout occurred once after stop aplay when Pulseaudio is off. UI freeze then timetout.

```
[ 1338.699643] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1338.699828] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1338.699840] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[ 1338.699980] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[ 1338.699991] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1338.700105] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[ 1338.700216] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[ 1339.368582] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1339.368615] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1342.055963] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1342.360546] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[ 1342.360576] sof-audio-pci 0000:00:1f.3: unknown ROM status value ffffffff
[ 1342.360855] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xffffffff trace 0xffffffff
[ 1342.360863] sof-audio-pci 0000:00:1f.3: error: ctx_save ipc error during suspend -110
[ 1342.360881] sof-audio-pci 0000:00:1f.3: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -110)
[ 1382.242672] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[ 1382.242678] sof-audio-pci 0000:00:1f.3: error: pcm open failed to resume -22
[ 1382.242687] sof-audio-pci 0000:00:1f.3: ASoC: can't open component 0000:00:1f.3: -22
[ 1382.242693]  HDA Analog: ASoC: failed to start FE -22
[ 1450.570690] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 0
[ 1450.570697] sof-audio-pci 0000:00:1f.3: error: pcm open failed to resume -22
[ 1450.570707] sof-audio-pci 0000:00:1f.3: ASoC: can't open component 0000:00:1f.3: -22
[ 1450.570715]  HDA Analog: ASoC: failed to start FE -22

```
**Environment**
Rander UCM https://github.com/RanderWang/intel-UCM   
Kernel repo：c99938c https://github.com/thesofproject/linux/tree/release/sof-v5.0 
Firmware repo: 47fcac2  https://github.com/thesofproject/sof/commits/master |  
Topology: https://github.com/thesofproject/sof/commits/master . Same as above SOF repo
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg

[ipctimeout_Poff_dmesg.log](https://github.com/thesofproject/sof/files/3156262/ipctimeout_Poff_dmesg.log)
[ipctimeout_Poff_logger.log](https://github.com/thesofproject/sof/files/3156263/ipctimeout_Poff_logger.log)

ClarexZhou, 2019-05-08:
0x30030000: GLB_TPLG_MSG: COMP_CONNECT  ipc timed out occurred one time  after plug headset.
No error sof logger
Dmesg:
```
[ 1583.936726] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[ 1583.936861] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30100000: GLB_TPLG_MSG: PIPE_NEW
[ 1583.936866] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[ 1583.936870] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[ 1583.936946] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[ 1583.936951] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[ 1583.937070] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[ 1583.937074] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[ 1583.937165] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[ 1583.937170] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[ 1583.937274] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[ 1583.937278] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
[ 1584.241624] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x30030000 size 16
[ 1584.241638] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[ 1584.241867] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 1584.241872] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[ 1584.241878] sof-audio-pci 0000:00:1f.3: error: failed to load route sink  control none source 
[ 1584.241881] sof-audio-pci 0000:00:1f.3: error: failed to restore pipeline after resume -110
[ 1584.241900] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 5:0
[ 1584.241909] sof-audio-pci 0000:00:1f.3: pcm: open stream 3 dir 0
[ 1584.241911] sof-audio-pci 0000:00:1f.3: error: pcm open failed to resume -22
[ 1584.241915] sof-audio-pci 0000:00:1f.3: ASoC: can't open component 0000:00:1f.3: -22
[ 1584.241919]  HDMI1: ASoC: failed to start FE -22
[ 1584.438605] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1584.438637] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 1588.739566] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 5:0
[ 1588.739570] sof-audio-pci 0000:00:1f.3: pcm: open stream 3 dir 0
[ 1588.739571] sof-audio-pci 0000:00:1f.3: error: pcm open failed to resume -22
[ 1588.739574] sof-audio-pci 0000:00:1f.3: ASoC: can't open component 0000:00:1f.3: -22
[ 1588.739575]  HDMI1: ASoC: failed to start FE -22

```
**Environment**
Rander UCM https://github.com/RanderWang/intel-UCM   
Kernel repo：c99938c https://github.com/thesofproject/linux/tree/release/sof-v5.0 
Firmware repo: 05d4e19  https://github.com/thesofproject/sof/commits/master |  
Topology: https://github.com/thesofproject/sof/commits/master . Same as above SOF repo
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg

[ipcTimeOutAfterPlug.log](https://github.com/thesofproject/sof/files/3156274/ipcTimeOutAfterPlug.log)
[ipcTimeOutAfterPlugLogger.log](https://github.com/thesofproject/sof/files/3156275/ipcTimeOutAfterPlugLogger.log)

wenqingfu, 2019-05-08:
@ClarexZhou I will close this issue. Please report the last two findings as stand alone issue to track. Thanks!

closed 2019-05-08 by wenqingfu

ClarexZhou, 2019-05-09:
@wenqingfu Report new bugs as below:
1. [BUG][CML] ipc timed out for '0x30030000: GLB_TPLG_MSG: COMP_CONNECT' #1395
2. [BUG][CML]IPC timed out for '0x60040000: GLB_STREAM_MSG: TRIG_START' #1396
3. [Bug][CML] ipc timed out for '0x40010000: GLB_PM_MSG: CTX_SAVE'  https://github.com/thesofproject/linux/issues/919

mengdonglin, 2019-05-09:
Other IPC time out:
https://github.com/thesofproject/sof/issues/1342
https://github.com/thesofproject/linux/issues/808
https://github.com/thesofproject/linux/issues/876
https://github.com/thesofproject/linux/issues/873