thesofproject/sof #1256: [BUG][CML] aplay would be interrputed by 'write error: Input/output error' when DUT wakeup from S3

stevyan, 2019-04-11:
**Describe the bug**
aplay would be interrputed by 'write error: Input/output error' when DUT wakeup from S3

**To Reproduce**
1. aplay -Dhw:0,0 -f dat /dev/zero &
2. sudo rtcwake -m mem -s 1

**Actual behavior**
Device wakeup successful, but the aplay process will be interrupted with i/o error.

**Environment**
Kernel: 55f5a2ec (rander/cml-validation-branch)
Firmware: ed6ac05 (with reverted commit f5e7843)
Topology: 0efa03d(rander/cml-validation-branch)

**sof-logger log**
```
    0      2          DMA           2837234.843750         3.489583 intel/cavs/hda-dma.c:549 	hda-dmac: 5 channel 0 -> start
    0      2          DMA           2837238.385417         3.541667 intel/cavs/hda-dma.c:396 	hda-dmac: 5 channel 0 -> enable
    0      2       VOLUME           2837295.312500        56.927082   src/audio/volume.c:454 	volume_trigger()
    0      2          DAI 1.4       2837298.697917         3.385417      src/audio/dai.c:544 	dai_comp_trigger(), command = 1
    0      2          DAI 1.4       2837301.979167         3.281250      src/audio/dai.c:555 	dai_comp_trigger(), START
    0      1       VOLUME           2838419.375000      1117.395874   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      2         COMP           2838424.375000         5.000000 of/audio/component.h:708 	comp_overrun(), ((dev->comp.id << 16) | sink->free) = 65536, ((min_bytes << 16) | copy_bytes) = 0
    0      2         PIPE 1.5       2838427.447917         3.072917 src/audio/pipeline.c:533 	pipeline_trigger()
    0      2         HOST           2838430.781250         3.333333     src/audio/host.c:281 	host_trigger()
    0      2          DMA           2838434.218750         3.437500 intel/cavs/hda-dma.c:645 	hda-dmac: 5 channel 0 -> stop
    0      2       VOLUME           2838437.864583         3.645833   src/audio/volume.c:454 	volume_trigger()
    0      2          DAI 1.4       2838441.458333         3.593750      src/audio/dai.c:544 	dai_comp_trigger(), command = 8
    0      2          DAI 1.4       2838444.635417         3.177083      src/audio/dai.c:602 	dai_comp_trigger(), XRUN
    0      2          DAI 1.4       2838447.500000         2.864583      src/audio/dai.c:608 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA           2838450.625000         3.125000 intel/cavs/hda-dma.c:645 	hda-dmac: 7 channel 0 -> stop
    0      1         PIPE           2838457.604167         6.979167 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       2838463.906250         6.302083 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      2         PIPE 1.5       2838468.593750         4.687500 src/audio/pipeline.c:358 	pipeline_prepare()
    0      2         HOST           2838471.770833         3.177083     src/audio/host.c:640 	host_prepare()
    0      2         COMP           2838474.791667         3.020833 rc/audio/component.c:136 	comp_set_state(), state already set to 3
    0      2         PIPE 1.5       2838478.229167         3.437500 src/audio/pipeline.c:533 	pipeline_trigger()
    0      2         HOST           2838481.562500         3.333333     src/audio/host.c:281 	host_trigger()
    0      2          DMA           2838485.052083         3.489583 intel/cavs/hda-dma.c:549 	hda-dmac: 5 channel 0 -> start
    0      2          DMA           2838488.489583         3.437500 intel/cavs/hda-dma.c:396 	hda-dmac: 5 channel 0 -> enable
    0      2       VOLUME           2838545.572917        57.083332   src/audio/volume.c:454 	volume_trigger()
    0      2          DAI 1.4       2838548.958333         3.385417      src/audio/dai.c:544 	dai_comp_trigger(), command = 1
    0      2          DAI 1.4       2838552.135417         3.177083      src/audio/dai.c:555 	dai_comp_trigger(), START
    0      1       VOLUME           2839669.479167      1117.343750   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      2         COMP           2839674.531250         5.052083 of/audio/component.h:708 	comp_overrun(), ((dev->comp.id << 16) | sink->free) = 65536, ((min_bytes << 16) | copy_bytes) = 0
    0      2         PIPE 1.5       2839677.500000         2.968750 src/audio/pipeline.c:533 	pipeline_trigger()
    0      2         HOST           2839680.833333         3.333333     src/audio/host.c:281 	host_trigger()
    0      2          DMA           2839684.270833         3.437500 intel/cavs/hda-dma.c:645 	hda-dmac: 5 channel 0 -> stop
    0      2       VOLUME           2839687.864583         3.593750   src/audio/volume.c:454 	volume_trigger()
    0      2          DAI 1.4       2839691.354167         3.489583      src/audio/dai.c:544 	dai_comp_trigger(), command = 8
    0      2          DAI 1.4       2839694.531250         3.177083      src/audio/dai.c:602 	dai_comp_trigger(), XRUN
    0      2          DAI 1.4       2839697.500000         2.968750      src/audio/dai.c:608 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA           2839700.625000         3.125000 intel/cavs/hda-dma.c:645 	hda-dmac: 7 channel 0 -> stop
    0      1         PIPE           2839707.395833         6.770833 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       2839713.854167         6.458333 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      2         PIPE 1.5       2839718.489583         4.635417 src/audio/pipeline.c:358 	pipeline_prepare()
    0      2         HOST           2839721.666667         3.177083     src/audio/host.c:640 	host_prepare()
    0      2         COMP           2839724.687500         3.020833 rc/audio/component.c:136 	comp_set_state(), state already set to 3
    0      2         PIPE 1.5       2839728.072917         3.385417 src/audio/pipeline.c:533 	pipeline_trigger()
    0      2         HOST           2839731.302083         3.229167     src/audio/host.c:281 	host_trigger()
    0      2          DMA           2839734.791667         3.489583 intel/cavs/hda-dma.c:549 	hda-dmac: 5 channel 0 -> start
    0      2          DMA           2839738.333333         3.541667 intel/cavs/hda-dma.c:396 	hda-dmac: 5 channel 0 -> enable
    0      2       VOLUME           2839795.364583        57.031250   src/audio/volume.c:454 	volume_trigger()
    0      2          DAI 1.4       2839798.750000         3.385417      src/audio/dai.c:544 	dai_comp_trigger(), command = 1
    0      2          DAI 1.4       2839802.083333         3.333333      src/audio/dai.c:555 	dai_comp_trigger(), START
    0      1       VOLUME           2840919.635417      1117.552124   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
    0      2         COMP           2840924.635417         5.000000 of/audio/component.h:708 	comp_overrun(), ((dev->comp.id << 16) | sink->free) = 65536, ((min_bytes << 16) | copy_bytes) = 0
    0      2         PIPE 1.5       2840927.760417         3.125000 src/audio/pipeline.c:533 	pipeline_trigger()
    0      2         HOST           2840931.093750         3.333333     src/audio/host.c:281 	host_trigger()
[... 10 lines cut ...]
    0      2         COMP           2840975.104167         3.020833 rc/audio/component.c:136 	comp_set_state(), state already set to 3
    0      2         PIPE 1.5       2840978.541667         3.437500 src/audio/pipeline.c:533 	pipeline_trigger()
    0      2         HOST           2840981.875000         3.333333     src/audio/host.c:281 	host_trigger()
    0      2          DMA           2840985.364583         3.489583 intel/cavs/hda-dma.c:549 	hda-dmac: 5 channel 0 -> start
    0      2          DMA           2840988.750000         3.385417 intel/cavs/hda-dma.c:396 	hda-dmac: 5 channel 0 -> enable
    0      2       VOLUME           2841046.250000        57.500000   src/audio/volume.c:454 	volume_trigger()
    0      2          DAI 1.4       2841049.635417         3.385417      src/audio/dai.c:544 	dai_comp_trigger(), command = 1
    0      2          DAI 1.4       2841052.812500         3.177083      src/audio/dai.c:555 	dai_comp_trigger(), START
    0      1       VOLUME           2842169.270833      1116.458374   src/audio/volume.c:494 	volume_copy() error: sink component buffer has not enough free bytes for copy
[... cut ...]
    0      1         PIPE           2842207.187500         6.770833 src/audio/pipeline.c:674 	pipeline_copy() error: ret = -5, start->comp.id = 4, dir = 1
    0      1         PIPE 1.5       2842213.645833         6.458333 src/audio/pipeline.c:782 	pipeline_xrun_recover()
    0      2         PIPE 1.5       2842218.385417         4.739583 src/audio/pipeline.c:358 	pipeline_prepare()
```
[dmesg_io_error_after_s3.log](https://github.com/thesofproject/sof/files/3067951/io_error_after_s3.log)

[sof-logger_io_error_after_s3.log](https://github.com/thesofproject/sof/files/3067957/sof-logger_io_error_after_s3.log)

tlauda, 2019-04-11:
@stevyan It looks like duplicate of #1152. Please verify and close. Next time please check for existing issues before creating the new one.

libinyang, 2019-04-11:
Is this HDA audio or I2S audio? If HDA audio it should be the same as #1152

stevyan, 2019-04-12:
It is HDA audio, duplicate to #1152

closed 2019-04-12 by stevyan