thesofproject/sof #1261: CML: Repeatedly switching from speaker-headset-speaker... results in DSP Panic

naveen-manohar, 2019-04-12:
During Speaker playback, repeatedly switching to headset and back to speaker, within 10 iterations hit following dsp panic error.

Test Steps:
On v0.2 recipe

Play audio in speaker
Plug 3.5mm headset jack to observe audio routing to hs
now in UI system tray, change the end-point node selection to speaker and back to headset repeatedly..
Observation: Within 10 iteration of step#3 DSP panic is hit and no audio
LOG:
[ 968.507135] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[ 968.507150] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[ 968.507436] sof-audio-pci 0000:00:1f.3: error: runtime exception
[ 968.507441] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[ 968.507446] sof-audio-pci 0000:00:1f.3: error: panic happen at :0
[ 968.507451] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[ 968.507456] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[14_headset_to-speaker_frequent_switch-dsp-panic.txt](https://github.com/thesofproject/sof/files/3074122/14_headset_to-speaker_frequent_switch-dsp-panic.txt)

wenqingfu, 2019-04-12:
@naveen-manohar appreciated if you can follow the template next time :ok_hand:

naveen-manohar, 2019-04-16:
Apologize, yes shall follow the template for all future bugs.
Can you please help add the issue to right owner in firmware team. Thanks

lgirdwood, 2019-04-16:
@dabekjakub @michalgrodzicki fyi

michalgrodzicki, 2019-04-18:
@akloniex - could you take a look?

akloniex, 2019-04-24:
@naveen-manohar 
I would need few things:
1. What is "v0.2 recipe"?
2. Do you have core dump saved / can you collect one? (/sys/kernel/debug/exception file, AFAIR)
3. Could you provide sof-logger output?

These would be very helpful to have in order to root-cause the issue.

naveen-manohar, 2019-04-26:
Apologize for the delay, was working on to switch to a recipe where logger was enabled.
With below recipe, I wasn't able to hit the issue. Have requested validation team to stress test the scenario & update.

**Test Recipe:**
SOF FW source
URL: https://github.com/thesofproject/sof/tree/cml-003-drop-stable 
Commit: 4731a1ad66ce78920714e7b7c17da34c0e24a523
SOF Drivers source
URL: https://github.com/thesofproject/linux/tree/topic/sof-v4.19-rebase
Commit: d52233f7
Plus commit 9d033568 and 9ffddce8 which need to be cherry picked from branch topic/sof-dev to the release branch topic/sof-v4.19-rebase.`

lgirdwood, 2019-04-26:
thanks @naveen-manohar, please close this if validation cant reproduce.

naveen-manohar, 2019-04-29:
Validation team's stress script dint hit the issue with above recipe, hence closing it. Thanks

closed 2019-04-29 by naveen-manohar