thesofproject/sof #1270: [BUG]trace error: invalid channel number of hda-dmac

keqiaozhang, 2019-04-16:
**Describe the bug**
This error happens after boot OS. But what's interesting is that the dmic recording works.

**To Reproduce**
Boot the system and check logger trace:
$ sudo ./sof-logger -l sof-apl.ldc -t
$ sudo ./sof-logger -l sof-apl.ldc 

**Expected behavior**
No trace errors in boot stage.

**Impact**
Currently, no functional effects.

**Environment**
Kernel: origin/topic/sof-dev-2de29f
Firmware: origin/sof-master-1b15fa
tplg name：sof-glk-da7219.tplg
GLK (Ubuntu OS).
Reproducibility Rate：100%

**Trace**
logger -t:
```
    0      2         PIPE 6.39      6019746.927083        13.750000 src/audio/pipeline.c:583    pipeline_reset()
    0      2         HOST           6019750.781250         3.854167     src/audio/host.c:710    host_reset()
    0      1          DMA           6019754.114583         3.333333 intel/cavs/hda-dma.c:616    hda-dmac: 5 invalid channel -1
    0      2       VOLUME           6019761.614583         7.500000   src/audio/volume.c:621    volume_reset()
    0      2         COMP           6019765.000000         3.385417 rc/audio/component.c:136    comp_set_state(), state already set to 1
```
logger:
```
    0      1          DMA           5899639.531250   5899639.500000 intel/cavs/hda-dma.c:616    hda-dmac: 5 invalid channel -1
    0      1          DMA           5899842.187500       202.656250 intel/cavs/hda-dma.c:616    hda-dmac: 5 invalid channel -1
    0      1          DMA           5899999.531250       157.343750 intel/cavs/hda-dma.c:616    hda-dmac: 5 invalid channel -1
........
```
[logger.log](https://github.com/thesofproject/sof/files/3082873/logger.log)
[logger-t.log](https://github.com/thesofproject/sof/files/3082874/logger-t.log)

tlauda, 2019-04-19:
@keqiaozhang Wrong logs attached? I don't see such errors.

keqiaozhang, 2019-04-19:
@tlauda 
I just confirmed this issue but failed to reproduce it again. (It may caused by some debug patches I applied at that time, but I cannot recall that now).  I also checked with the latest firmware/driver and also cannot reproduce it.  let's close this bug.

closed 2019-04-19 by keqiaozhang