thesofproject/sof #1277: [CML]DMIC captured volume is too low and no corresponding volume controller

stevyan, 2019-04-17:
**Describe the bug**
DMIC captured volume is too low, hard to hear even play via max volume.

**To Reproduce**
1. Record via DMIC Dhw:0,6 or Dhw:0,7. eg. arecord -Dhw:0,6 -r 48000 -c 2 -f s16_le cap.wav -vv -i 
2. Play captured file.

**Expected behavior**
Volume is too hard to hear

**Environment**
Kernel: rander/cml-validation-20190417(topic/sof-dev, 2de29fae6+6 patch)
Firmware: bard/whl-daily-20190417(top of master 99e9f4370+3patch)
Topology: bard/whl-daily-20190417(top of master 99e9f4370+3patch)
Reproducibility Rate: 5/5

ClarexZhou, 2019-04-17:
[cap_32.zip](https://github.com/thesofproject/sof/files/3089259/cap_32.zip)

ClarexZhou, 2019-04-18:
update Environment:
Kernel Repo: https://github.com/RanderWang/linux/commits/cml-validation-20190417
Made from: https://github.com/RanderWang/linux/commit/2de29fae6a2dadda3eb084023be2167be85115e1 [thesofproject/linux: topic/sof-dev]
+https://github.com/RanderWang/linux/commit/d04ccf494f25624cadb8f55fdc73e79cd49d8bc1
+https://github.com/RanderWang/linux/commit/bfb0729ddca5773c5c63bacec0f701a3d157ec84
+https://github.com/RanderWang/linux/commit/8a2b13a81b1c8599a25762ddd460a4dbe3e94ace
+https://github.com/RanderWang/linux/commit/63ceb8f6813daec7027c9449a162d14842a0386c
+https://github.com/RanderWang/linux/commit/db89510d0b0660396c94b19f0a7e9e3f88c29e05
+https://github.com/RanderWang/linux/commit/066e6f4d85e4ee16b0d84acbebdd7e57e449e116

Firmware Repo :https://github.com/bardliao/sof/tree/whl-daliy-20190417
Made from: https://github.com/bardliao/sof/commit/99e9f4370efbfbfb19323a069146231af65aef98
+https://github.com/bardliao/sof/commit/2a36f1d214a36289ec847dbec1e514905890f9c8
+https://github.com/bardliao/sof/commit/28b1adfa1612e5d2fe195cdd8676b81e5787dda5
+https://github.com/bardliao/sof/commit/e3aebebe7a24e1e6ca6283ef961bc8f9e292516f

Topology: Same as above SOF repo

ClarexZhou, 2019-04-18:
Compare the capture of 2 Dell Mantis laptop and GLK Chrombook side by side. Almost in same level. Close bug. 
[glk-mantis-dmic_cap_volume_Compare.zip](https://github.com/thesofproject/sof/files/3092752/glk-mantis-dmic_cap_volume_Compare.zip)

stevyan, 2019-04-18:
closed the issue with Clare test result.

closed 2019-04-18 by stevyan