thesofproject/sof #1278: [CML]No volume controller for DMIC16/32

ClarexZhou, 2019-04-17:
**Describe the bug**
No volume controller for DMIC16/32 

**To Reproduce**
1. Record via DMIC Dhw:0,6 or Dhw:0,7. eg. arecord -Dhw:0,6 -r 48000 -c 2 -f s16_le cap.wav -vv -i
2. alsamixer, press F4 to check Capture items.

**Expected behavior**
Should support volume controller for DMIC16/32 

**Environment**
Kernel: rander/cml-validation-20190417(topic/sof-dev, 2de29fae6+6 patch)
Firmware: bard/whl-daily-20190417(top of master 99e9f43+3patch)
Topology: bard/whl-daily-20190417(top of master 99e9f43+3patch)
Reproducibility Rate: 5/5

emilchudzik, 2019-04-17:
What do you mean by "3patch"?

ClarexZhou, 2019-04-18:
update Environment:
Kernel Repo: https://github.com/RanderWang/linux/commits/cml-validation-20190417
Made from: https://github.com/RanderWang/linux/commit/2de29fae6a2dadda3eb084023be2167be85115e1 [thesofproject/linux: topic/sof-dev]
+https://github.com/RanderWang/linux/commit/d04ccf494f25624cadb8f55fdc73e79cd49d8bc1
+https://github.com/RanderWang/linux/commit/bfb0729ddca5773c5c63bacec0f701a3d157ec84
+https://github.com/RanderWang/linux/commit/8a2b13a81b1c8599a25762ddd460a4dbe3e94ace
+https://github.com/RanderWang/linux/commit/63ceb8f6813daec7027c9449a162d14842a0386c
+https://github.com/RanderWang/linux/commit/db89510d0b0660396c94b19f0a7e9e3f88c29e05
+https://github.com/RanderWang/linux/commit/066e6f4d85e4ee16b0d84acbebdd7e57e449e116

Firmware Repo :https://github.com/bardliao/sof/tree/whl-daliy-20190417
Made from: https://github.com/bardliao/sof/commit/99e9f4370efbfbfb19323a069146231af65aef98
+https://github.com/bardliao/sof/commit/2a36f1d214a36289ec847dbec1e514905890f9c8
+https://github.com/bardliao/sof/commit/28b1adfa1612e5d2fe195cdd8676b81e5787dda5
+https://github.com/bardliao/sof/commit/e3aebebe7a24e1e6ca6283ef961bc8f9e292516f

Topology: Same as above SOF repo

ClarexZhou, 2019-04-22:
Now DMIC 32 volume can be adjusted via PGA 10.0 on CML laptop.
But still no way to adjust volume for DMIC16
[image]

Environment
Kernel Repo: https://github.com/RanderWang/linux/commits/cml-validation-20190422
Made from: RanderWang/linux@9086043 [thesofproject/linux: topic/sof-dev]
+RanderWang/linux@c105b1b
+RanderWang/linux@ffecddc
+RanderWang/linux@3b917aa
+RanderWang/linux@6f928a3
+RanderWang/linux@a916d2d
+RanderWang/linux@694599b
+RanderWang/linux@a43379d
+RanderWang/linux@d36ed40
+RanderWang/linux@9b3bd71
+RanderWang/linux@e852311
+RanderWang/linux@8f30111
+RanderWang/linux@95cd082

Firmware Repo :https://github.com/bardliao/sof/tree/whl-daliy-20190422
Made from: bardliao/linux@0c93261
+ bardliao@35fbf79
+ bardliao@3f91567

Topology: Same as above SOF repo

mengdonglin, 2019-04-22:
@libinyang @singalsu Please evaluate if we still need PR https://github.com/thesofproject/sof/pull/1271 for the 16khz DMIC, which applies passthough pipeline for this DMIC and so no volume control.

libinyang, 2019-04-23:
#1293 is submitted for this issue

emilchudzik, 2019-04-23:
@ClarexZhou lack of volume control is not a firmware issue, please rise driver issue on linux repo.

singalsu, 2019-04-23:
@emilchudzik  Why linux/driver issue, would this be a topology issue?

ClarexZhou, 2019-04-23:
Error isn't seen on CML 20190423 test recipe.
Firmware repo	https://github.com/RanderWang/sof/tree/cml-linux-20190423
Made from:
0c93261 [thesofproject/sof: master]
+	adf4db3 [RanderWang/sof: cml-linux-20190423]
+	1c8cbd1 [RanderWang/sof: cml-linux-20190423]

Kernel repo	https://github.com/thesofproject/linux/tree/release/sof-v5.0
Made from:
9086043 [thesofproject/linux: topic/sof-dev]
+575038e
+	53cfc0d
+	39d0823
+	1b65a44
+	98554b9
+	1a44fac
+	6e4d133
+	2a816f7
+	569f460
+	f273121
+	8cbbf45
+	8de4c6
+	f891831
+	e5ced5b
+	0ad31d4
+	f5f69bd
+	5df2c93
+	8594d32
+	c48713b
Topology: https://github.com/RanderWang/sof/tree/cml-linux-20190423
file: tools/topology/sof-hda-generic.m4, sof-hda-generic-2ch.tplg

jajanusz, 2019-04-23:
@singalsu Emil meant that it's not a firmware issue, but yes, it can be topology issue, so we may keep it **open** because topologies are in this repo, just someone from driver team has to modify them.

emilchudzik, 2019-04-23:
> Why linux/driver issue, would this be a topology issue?

Using alsamixer volume control starts driver to send suitable IPC to firmware. 
Lack of volume control issue should be checked by driver/kernel guys. I'm not sure if it is topology issue.

jajanusz, 2019-04-23:
I see you made 2 PRs related to this issue. I guess this one is the right one: https://github.com/thesofproject/sof/pull/1293

libinyang, 2019-04-23:
Yes. #1293 aims to fix this issue.

> I see you made 2 PRs related to this issue. I guess this one is the right one: #1293

closed 2019-04-26 by tlauda