thesofproject/sof #1279: [FEATURE] IPC fuzzer.

lgirdwood, 2019-04-18:
**Is your feature request related to a problem? Please describe.**
IPC mechanism needs to be fuzzed to ensure continuity of service.

**Describe the solution you'd like**
Userspace IPC fuzzer that can load and "manufacture" IPC to send to FW running on Qemu

lbetlej, 2019-04-18:
Is the goal to implement a custom user space fuzzer from scratch or to provide an easy connection for existing 3rd party fuzzers into SOF stack ?

ranj063, 2019-04-18:
> Is the goal to implement a custom user space fuzzer from scratch or to provide an easy connection for existing 3rd party fuzzers into SOF stack ?

@lbetlej we're starting off with a very primitive fuzzer from scratch. But this will probably need to evolve to use a fuzzing library if suitable later on.

lgirdwood, 2019-06-07:
@ranj063 I assume we will be good to merge our current fuzzer after IPC is done ?

lgirdwood, 2019-08-30:
@ranj063 anything still pending here ? I'm assuming we are all done for initail fuzzer ?

closed 2019-11-12 by lgirdwood

ranj063, 2019-11-12:
> @ranj063 anything still pending here ? I'm assuming we are all done for initail fuzzer ?

@lgirdwood yes we're done for the time being. I will add APL support next