thesofproject/sof #1282: [CML]Serious audio distortion/drop when aplay via Dhw:0,0 and Dhw:0,3(HDMI) at same time

ClarexZhou, 2019-04-19:
**Describe the bug**
Serious audio distortion/drop when aplay via Dhw:0,0 and Dhw:0,3(HDMI) at same time

**To Reproduce**
aplay -Dhw:0,0 -f s16_le -r 48000 -c 2  wav/48California_Dreaming.wav & aplay -Dhw:0,3 -f s16_le -r 48000 -c 2  wav/48California_Dreaming.wav

**Expected behavior**
Both audio output should be good, without interaction

**Environment**
Kernel Repo:
https://github.com/RanderWang/linux/commits/cml-validation-20190418
Made from:
https://github.com/RanderWang/linux/commit/2de29fae6a2dadda3eb084023be2167be85115e1
+https://github.com/RanderWang/linux/commit/d04ccf494f25624cadb8f55fdc73e79cd49d8bc1
+https://github.com/RanderWang/linux/commit/bfb0729ddca5773c5c63bacec0f701a3d157ec84
+https://github.com/RanderWang/linux/commit/8a2b13a81b1c8599a25762ddd460a4dbe3e94ace
+https://github.com/RanderWang/linux/commit/63ceb8f6813daec7027c9449a162d14842a0386c
+https://github.com/RanderWang/linux/commit/db89510d0b0660396c94b19f0a7e9e3f88c29e05
+https://github.com/RanderWang/linux/commit/066e6f4d85e4ee16b0d84acbebdd7e57e449e116
+https://github.com/RanderWang/linux/commit/bfb3fcb243b604e7f2814121e83e788943c439ca
+https://github.com/RanderWang/linux/commit/edc012c29972b0d6557b35c0c0ba5879c0fdab1c
+https://github.com/RanderWang/linux/commit/e0f16c6465c3e76f48151322e99f53abde0e146d
+https://github.com/RanderWang/linux/commit/852490e26322432b6e734bddce32c3799b1ab1ca
+https://github.com/RanderWang/linux/commit/2f1a4b6093de40b8101302c9f3de4e8a97449b7d
Sof Repo: https://github.com/bardliao/sof/tree/whl-daliy-20190418
Made from: https://github.com/bardliao/sof/commit/99e9f4370efbfbfb19323a069146231af65aef98
+https://github.com/bardliao/sof/commit/b56f07c8f3bb5131e66bbc383bfc2fc751ab813c
+https://github.com/bardliao/sof/commit/cc3846dad177d97dd62f943aa5448e4f3baa68f6
+https://github.com/bardliao/sof/commit/ec1ac7123f02f2631ae7d9d7a86f3cf202c38e7d
+https://github.com/bardliao/sof/commit/8b4cc887afb52fc2c003328eec5e4c384712c78b

tplg: Same as above SOF repo

tlauda, 2019-04-19:
@ClarexZhou Are those FW logs complete? I can see there is only one pipeline started and stopped. No traces as for the second pipeline. Maybe the second one is in legacy mode or something?
@RanderWang @bardliao Could you take a look?

keyonjie, 2019-04-19:
@ClarexZhou this is a regression issue on cml-validation driver side, please help verify today's recipe, if fixed, please help close it.

ClarexZhou, 2019-04-19:
Error not seen on Today's recipe.  Close.
Kernel Repo:
https://github.com/RanderWang/linux/commits/cml-validation-20190419
Made from:
RanderWang/linux@2de29fa
+ RanderWang/linux@d04ccf4
+ RanderWang/linux@bfb0729
+ RanderWang/linux@8a2b13a
+ RanderWang/linux@63ceb8f
+ RanderWang/linux@db89510
+ RanderWang/linux@066e6f4
+ RanderWang/linux@bfb3fcb
+ RanderWang/linux@edc012c
+ RanderWang/linux@a9f0431
+ RanderWang/linux@d5113b0
+ RanderWang/linux@aeccfd5
+ RanderWang/linux@22999bd
+ RanderWang/linux@c827ffb

Sof Repo: https://github.com/bardliao/sof/tree/whl-daliy-20190419
Made from: bardliao@ 4c6f6f7
+ bardliao@90a6efd 
+ bardliao@5ef0713
+ bardliao@50bd96d
+ bardliao@fdf1eda
+ bardliao@f135dd9
+ bardliao@e8d5622

tplg: Same as above SOF repo

closed 2019-04-19 by ClarexZhou